mbentley/docker-omada-controller #4: eap discovery not working using docker run example

stefan-svandijk-nl, 2019-04-24:
The docker run example doesn't forward the necessary ports to enable eap discovery. I have got it working with:

docker run -d   --name omada-controller   --restart unless-stopped   -p 8088:8088   -p 8043:8043  -p 29810:29810/udp -p 29811:29811 -p 29812:29812 -p 29813:29813  -v omada-data:/opt/tplink/EAPController/data   -v omada-work:/opt/tplink/EAPController/work   -v omada-logs:/opt/tplink/EAPController/logs   mbentley/omada-controller

Please fix the read.me
Testen on Synology DS918+

scotthernandez, 2019-05-01:
I suspect the issue is that the network the container is running on is not the same as the AP so it can't find/discover the AP. 

I too have the same issue and will see if I can find out how to change the container network (to a bridge of the host) or look at the config for the omada-controller to see if there is a scan/discover (sub)net to configure.

stefan-svandijk-nl, 2019-05-01:
> I suspect the issue is that the network the container is running on is not the same as the AP so it can't find/discover the AP.
>
> I too have the same issue and will see if I can find out how to [...]

Please note the extra port forwarding fixes the issue in my case. Docker indeed creates a 172.17.0.0/16 bridged network and the port forwarding fixes the issue. See also tp-link faq's on Omada controller.

Running on the container on the host network should also fix the issue, but would be less secure.

scotthernandez, 2019-05-02:
Thanks, yeah, I found the listed ports in the FAQ too and added them in the end; then I could use the sep. program to associate the AP+Omada controller (host ip) in order to adopt the AP in the Omada webui.

Unfortunately that resulted in wiping the existing config in the AP, so I lost wifi along the way, ooopss -- really poor software from TP-Link, all in all.

s0uls, 2019-05-22:
Same issue found, had to either set up as bridge or add all ports on creation to get it working on QNAP nas docker

RaveGun, 2019-07-03:
I tried to run another instance of the official omada controller on another machine as a setup for a migration. But according to the TP Link documentation is not possible to run the two instances in the same LAN.

Will this docker instance run as a secondary instance?
Maybe this migration stuff is not even ment to work the way I imagine. I have no idea what I am taking about. At least if I am breaking the WiFi I have the two machines directly connected (no WiFi).

eekdood, 2019-07-19:
Glad I found this issue. Solved by adding additional ports as listed in TP-Link FAQ at https://www.tp-link.com/us/support/faq/865/

My working compose file (segment):
`omada-controller:
    container_name: omada-controller
    restart: unless-stopped
    ports:
      - '8088:8088'
      - '8043:8043'
      - '27001:27001/udp'
      - '27002:27002'
      - '27017:27017'
      - '29810:29810/udp'
      - '29811:29811'
      - '29812:29812'
      - '29813:29813'
    volumes:
      - 'omada-data:/opt/tplink/EAPController/data'
      - 'omada-work:/opt/tplink/EAPController/work'
      - 'omada-logs:/opt/tplink/EAPController/logs'
    image: mbentley/omada-controller`

eekdood, 2019-07-19:
Thanks for this. Solved by adding additional ports.

My working docker-compose.yml:

>  omada-controller:
    container_name: omada-controller
    restart: unless-stopped
    ports:
      - '8088:8088'
      - '8043:8043'
      - '27001:27001/udp'
      - '27002:27002'
      - '27017:27017'
      - '29810:29810/udp'
      - '29811:29811'
      - '29812:29812'
      - '29813:29813'
    volumes:
      - 'omada-data:/opt/tplink/EAPController/data'
      - 'omada-work:/opt/tplink/EAPController/work'
      - 'omada-logs:/opt/tplink/EAPController/logs'
    image: mbentley/omada-controller

anklimov, 2019-08-25:
Thanks, I have same issue, additional port forwarding helps
What about including ports to upstream?

mbiebl, 2019-12-06:
I didn't need expose those ports
```
- '27001:27001/udp'
- '27002:27002'
- '27017:27017'
```

Nevertheless, EAP autodiscovery seems to have worked and I was able to find and bind my APs.
Do you by chance know, what those ports are needed for?

mbiebl, 2019-12-22:
To answer my own question, this is nicely documented in https://www.tp-link.com/us/support/faq/865/

```
Q4: Which ports does EAP Controller use?

A: 
8043 (TCP) for https, 
8088 (TCP) for http, 
27001 (UDP) for controller discovery, 
27002 (TCP) for controller searching, 
27217(TCP) for mongo DB server (This used to be 27017(TCP) before EAP Controller 3.0.0), 
29810 (UDP) for EAP discovery, 
29811 (TCP) for EAP management, 
29812 (TCP) for EAP adoption, 
29813 (TCP) for EAP upgrading. 

Make sure these ports are not occupied by other applications when you run EAP controller.
```

closed 2020-01-29 by mbentley

jiimint, 2023-09-27:
Hallo
I install docker omada controller but its not working adopting.
I have to to tell you that my EAP belongs to another subnet than my docker omada controller.
Is there any way to adopting with success?
Thank you

J4CE2, 2023-09-27:
Check this.

https://www.tp-link.com/us/support/faq/2713/#:~:text=EAP%20Discovery%20Utility%20is%20a,0.0%2F24%2C%20192.168.

commit 97c67194 (2020-01-29): Export ports required for EAP management

https://www.tp-link.com/us/support/faq/865/

Q4: Which ports does EAP Controller use?
...
29810 (UDP) for EAP discovery,
29811 (TCP) for EAP management,
29812 (TCP) for EAP adoption,
29813 (TCP) for EAP upgrading.

Fixes #4