mbentley/docker-omada-controller #6: Service fails to start

markcarroll, 2019-07-31:
I am getting the following error when trying to run this on a Synology NAS:

```
Exception in thread "Thread-0" java.lang.IllegalArgumentException: URI is not hierarchical | stdout
    at java.io.File.<init>(File.java:418)
    at com.tp_link.eap.util.g.a.a(SourceFile:47)
    at com.tp_link.eap.configuration.b.a(SourceFile:58)
    at com.tp_link.eap.util.o.c(SourceFile:443) | stdout
    at com.tp_link.eap.util.o.b(SourceFile:434)
    at com.tp_link.eap.start.f.run(SourceFile:35)
```

Folders have been set up and permissions have been set to everyone RW.

Ideas?

mbiebl, 2020-01-28:
I can reproduce this and this still seems to happen with 3.2.6.
I made sure data is properly owned by uid 508, but the controller software fails to create the db subdirectory.
We should at least document this issue in the README, I think.
An alternative would be to have a start script which properly creates those directories 

@mbentley wdyt?

mbentley, 2020-01-28:
Yeah I would lean toward a note in the readme. I would add an entrypoint to change ownership or at least make it an option, if the container was starting as root.  I’d rather set the USER parameter in the Dockerfile and drop any root access in the resulting image.

timdonovanuk, 2020-03-09:
Facing this too, although I don't see any specific error message:

```
INFO: Time zone set to 'Etc/UTC'
INFO: Starting Omada Controller as user omada
SLF4J: com.tp_link.eap.configuration.b
SLF4J: The following set of substitute loggers may have been accessed
SLF4J: during the initialization phase. Logging calls during this
SLF4J: phase were not honored. However, subsequent logging calls to these
SLF4J: loggers will work as normally expected.
SLF4J: See also http://www.slf4j.org/codes.html#substituteLogger
SLF4J: com.tp_link.eap.util.o
startup...
Mar 09, 2020 4:19:17 PM org.hibernate.validator.internal.util.Version <clinit>
INFO: HV000001: Hibernate Validator 4.3.1.Final
Failed to start omada controller, going to exit
```

I have manually created /opt/tplink/EAPController/data,work,logs and /data/db and chown 508:508 to all of these.

mbentley, 2020-03-09:
I’m thinking maybe I should see if I can do some pre-flight checks to make sure the user can write to all of the necessary folders and see if I can proactively bring these issues up in the logs and refuse to start.

timdonovanuk, 2020-03-09:
Actually I just solved this, thanks! I found in the comments of another docker image (forked from yours) that you need 4GB of free space to start the Omada controller, as it builds a db this large on first run. Didn't expect that (or would expect at the very least for omada to output some useful info about this!). Cheers for the docker image btw :)

mbiebl, 2020-03-09:
monodb will create 3 journal files, each 1GB in size (plus the actual database).
You can use the `--smallfiles` option, this will use journal files that are 128MB in size.
See my comments at https://github.com/mbentley/docker-omada-controller/pull/9#issuecomment-584244825

I think for the amount of data that is written by the omada controller software, 128MB of journal should be plenty.

timdonovanuk, 2020-03-10:
Not sure (due to it all being dockerised) - where does one apply the `--smallfiles` option? Many thanks!

mbiebl, 2020-03-10:
Inside the docker container in `/opt/tplink/EAPController/properties/mongodb.properties` append it to `eap.mongod.args=`

timdonovanuk, 2020-03-13:
I don't have a properties directory, or a mongodb.properties. I have data, logs and work directories, as per the container instructions.

timdonovanuk, 2020-03-13:
Ah..its within the running container. I ssh'ed in but the container doesn't have vim, vi, nano or pico available to edit the file. My unix foo is not good enough.

Also won't this change get blown away every time the docker container gets updated?

@mbentley would it be possible to expose this property file as part of the persistent volumes? Or make it an environmental parameter?

timdonovanuk, 2020-03-13:
Okay, had to get creative here, but for anyone else looking for a solution I think the following should work inside the container

`sed -i 's/nohttpinterface/nohttpinterface --smallfiles/g' /opt/tplink/EAPController/properties/mongodb.properties`

mbentley, 2020-03-13:
Yes, it would be possible to do something.  The easiest would be to copy the file out of the container  using `docker cp`, make the updates, and make it a persistent bind mount in addition to the other persistent directories.  I was looking at doing something allow a flag to add that setting using a similar method but I have not had a chance to do so.

timdonovanuk, 2020-03-13:
I can confirm my command works. I actually ran into this problem again where the omada-controller wouldn't start as there was under 3.4GB left on the system. Smallfiles fixed it! Thanks.

mbiebl, 2020-03-13:
> Also won't this change get blown away every time the docker container gets updated?

It will. You can use the approach though as suggested by @mbentley to bind-mount the properties file into the container. This will survive container updates.

mbentley, 2020-03-23:
@timdonovanuk - I have a more permanent fix for you that I added in https://github.com/mbentley/docker-omada-controller/pull/26  Let me know if that works.  Looks like it should.

commit 0353898a (2020-03-23): Merge pull request #26 from mbentley/small-files

Added SMALL_FILES env var; fixes #6
commit 141640a1 (2020-03-23): Added SMALL_FILES env var; fixes #6
closed 2020-03-23 by mbentley