mbentley/docker-omada-controller #10: errors during start and stop

mbiebl, 2019-11-09:
hi,

thanks for providing these docker images.

During runtime of the container, I get the following error messages:
```
Exception in thread "Thread-0" java.lang.IllegalArgumentException: URI is not hierarchical
        at java.io.File.<init>(File.java:418)
        at com.tp_link.eap.util.g.a.a(SourceFile:47)
        at com.tp_link.eap.configuration.b.a(SourceFile:58)
        at com.tp_link.eap.util.o.c(SourceFile:443)
        at com.tp_link.eap.util.o.b(SourceFile:434)
        at com.tp_link.eap.start.f.run(SourceFile:35)
startup...
Nov 09, 2019 4:26:56 AM org.hibernate.validator.internal.util.Version <clinit>
INFO: HV000001: Hibernate Validator 4.3.1.Final
Omada Controller started
ShutdownHook: service stopped.
Exception in thread "Thread-0" java.lang.IllegalArgumentException: URI is not hierarchical
        at java.io.File.<init>(File.java:418)
        at com.tp_link.eap.util.g.a.a(SourceFile:47)
        at com.tp_link.eap.configuration.b.a(SourceFile:58)
        at com.tp_link.eap.util.o.c(SourceFile:443)
        at com.tp_link.eap.util.o.b(SourceFile:434)
        at com.tp_link.eap.start.f.run(SourceFile:35)
startup...
Nov 09, 2019 1:05:55 PM org.hibernate.validator.internal.util.Version <clinit>
INFO: HV000001: Hibernate Validator 4.3.1.Final
Omada Controller started

ShutdownHook: service stopped.
```

KYLE-HILL, 2019-11-10:
See if this fixes your issue. https://github.com/mbentley/docker-omada-controller/issues/6

mbiebl, 2019-11-10:
This error only happens when I start the docker container for the first time. After a restart the error is gone.

fenix-silver, 2020-01-02:
I'm seeing a similar issue after updating to Unraid 6.8.  Couldn't get the container to start at all.  Even tried creating a new instance.  Downgraded back to 6.7.2 and it started right up.

mbiebl, 2020-01-28:
I no longer see his error with 3.2.6.

mbentley, 2020-01-29:
@mbiebl - think this is good to close as being fixed in 3.2.6 then?  I haven't paid attention to the logs.

mbiebl, 2020-01-29:
I think so. At least I haven't seen this error anymore after upgrading to 3.2.6.

closed 2020-01-30 by mbentley