mbentley/docker-omada-controller #12: Please update to latest version 3.2.4

mbiebl, 2019-12-01:
According to https://www.tp-link.com/en/support/download/eap-controller/#Controller_Software
the latest version is 3.2.4, released on 2019-11-08.

Would be great to have an updated docker image.

Thanks!

titilambert, 2019-12-22:
Fixed by #14

mbiebl, 2020-01-14:
Is there anything we can do to move this along? 3.2.6 has just been released (currently Win only, but Linux will likely follow soonish)

titilambert, 2020-01-14:
If nothing move shortly, I might fork the project and automate the build of the image when a new version is released ...

mbiebl, 2020-01-22:
3.2.6 has been released, so updating the bug title accordingly:
https://static.tp-link.com/2020/202001/20200116/Omada_Controller_v3.2.6_linux_x64.tar.gz

mbiebl, 2020-01-27:
@titilambert I went ahead and did just that.
https://hub.docker.com/r/mbiebl/omada-controller
https://github.com/mbiebl/docker-omada-controller
in case you are interested.

@mbentley happy to submit a PR if you find the changes in that fork interesting

mbentley, 2020-01-28:
Very nice!  A PR would be awesome.  I did add 3.2.6 support just now.   Do you do anything in specific to get notified of new releases?  I just happen to check the page for updates every once and a while.  I can't figure out a good way to check for new releases.

Some comments on the diff between our branches (awesome work, by the way):

* I use `macvlan` so I didn't know about the additional port requirements.  Thanks for finding that!
* Time zone is always nice to have the same across the host and containers.
* I honestly don't know why I had `libcap-dev` in my first image.  It might have been a dependency from an older version when I was first trying it; not sure as it was in my first commit.
* I usually add `DEBIAN_FRONTEND="noninteractive"` specifically in front of a given command if I care to use it; it's just cosmetic

mbiebl, 2020-01-28:
@mbentley  Hi, it just happened that I bought a couple of tp-link EAP245v3 last month. Which is why I got interested in this topic. I encountered a couple of issues with the controller software which I reported to the tp-link forum. They asked me to try the latest version to reproduce the issues.
So no, I don't have a mechanism to get automatically notified about new releases.

I'll clean up my fork and submit as PR.  Do you prefer separate PRs or one large one?

mbiebl, 2020-01-28:
Fwiw, I've never done anything docker related so far. This is just a result of me reading up docs over the weekend and creating my first docker image and pushing it to https://hub.docker.com/.
I do have have some extensive Debian knowledge though.
So bear with me, if I did something stupid regarding docker.

mbentley, 2020-01-28:
No problem; doing good so far :)  Feel free to just make it all in a single PR.

mbiebl, 2020-01-28:
@mbentley submitted as #15.

Closing this issue, as the update to 3.2.6 has happened in the master branch.

closed 2020-01-28 by mbiebl

mbiebl, 2020-02-01:
> @titilambert I went ahead and did just that.
> https://hub.docker.com/r/mbiebl/omada-controller
> https://github.com/mbiebl/docker-omada-controller
> in case you are interested.

Fwiw, I've deleted that dockerhub repository again as all my changes are now properly integrated into @mbentley 's docker image. Thanks again @mbentley