mbentley/docker-omada-controller #20: typo on --publish 29180 UDP port

proap, 2020-02-09:
The option should be 
-p 29180:29180/udp \

otherwise the controller won't see the AP.

mbentley, 2020-02-09:
Hmm, I'm using `macvan` so I don't have any ports exposed manually but I also do not see anything related to 29180 in my netstat output:

```
# netstat -anp
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:8088            0.0.0.0:*               LISTEN      -
tcp        0      0 127.0.0.11:38373        0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:8043            0.0.0.0:*               LISTEN      -
tcp        0      0 127.0.0.1:27217         0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:29811           0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:29812           0.0.0.0:*               LISTEN      -
tcp        0      0 192.168.2.160:29811     192.168.0.2:51466       ESTABLISHED -
tcp        0      0 127.0.0.1:27217         127.0.0.1:39944         ESTABLISHED -
tcp        0      0 192.168.2.160:29811     192.168.0.3:41620       ESTABLISHED -
tcp        0      0 127.0.0.1:27217         127.0.0.1:39768         ESTABLISHED -
tcp        0      0 127.0.0.1:27217         127.0.0.1:39628         ESTABLISHED -
tcp        0      0 127.0.0.1:39628         127.0.0.1:27217         ESTABLISHED -
tcp        0      0 127.0.0.1:39768         127.0.0.1:27217         ESTABLISHED -
tcp        0      0 127.0.0.1:39842         127.0.0.1:27217         ESTABLISHED -
tcp        0      0 127.0.0.1:39944         127.0.0.1:27217         ESTABLISHED -
tcp        0      0 192.168.2.160:44414     3.224.113.94:443        ESTABLISHED -
tcp        0      0 127.0.0.1:27217         127.0.0.1:39842         ESTABLISHED -
udp        0      0 127.0.0.11:44801        0.0.0.0:*                           -
udp        0      0 0.0.0.0:29810           0.0.0.0:*                           -
udp        0      0 0.0.0.0:27001           0.0.0.0:*                           -
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node   PID/Program name     Path
unix  2      [ ACC ]     STREAM     LISTENING     295669227 -                    /tmp/mongodb-27217.sock
unix  2      [ ]         STREAM     CONNECTED     295686510 -
unix  2      [ ]         STREAM     CONNECTED     295668983 -
```

mbentley, 2020-02-09:
Also, the [documentation](https://www.tp-link.com/us/support/faq/865/) lists the following:

> 8043 (TCP) for https, 8088 (TCP) for http, 27001 (UDP) for controller discovery, 27002 (TCP) for controller searching, 27217(TCP) for mongo DB server (This used to be 27017(TCP) before EAP Controller 3.0.0), 29810 (UDP) for EAP discovery, 29811 (TCP) for EAP management, 29812 (TCP) for EAP adoption, 29813 (TCP) for EAP upgrading

To make them easier to read:

```
8043 (TCP) for https
8088 (TCP) for http
27002 (TCP) for controller searching
27217 (TCP) for mongo DB server (This used to be 27017(TCP) before EAP Controller 3.0.0)
29811 (TCP) for EAP management
29812 (TCP) for EAP adoption
29813 (TCP) for EAP upgrading

27001 (UDP) for controller discovery
29810 (UDP) for EAP discovery
```

All of those are exposed per the example in the README.md.

closed 2020-02-09 by mbentley