mbentley/docker-omada-controller #36: Unable to run on Raspberry Pi 3 Model B+

kipusoep, 2020-05-12:
Hi, I'm trying to run the arm64 image on a RPi 3 B+ with the command I copied from the readme:
```
docker run -d \
  --name omada-controller \
  --restart unless-stopped \
  -e TZ=Etc/UTC \
  -e SMALL_FILES=false \
  -p 8088:8088 \
  -p 8043:8043 \
  -p 27001:27001/udp \
  -p 27002:27002 \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29812:29812 \
  -p 29813:29813 \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:3.2-arm64
```

The container is crashing with the following error:
`standard_init_linux.go:211: exec user process caused "exec format error"`

AFAIK this relates to the architecture, but the arm64 image should be compatible with RPi3 right?

mbentley, 2020-05-12:
I'm assuming that you're running Raspbian.  Raspbian runs `armv7l` so it is a different architecture.  See https://github.com/mbentley/docker-omada-controller/issues/31 as well.

kipusoep, 2020-05-12:
Nah I'm running DietPi, but same issue I guess, it's 32 bit right?

mbentley, 2020-05-12:
Probably.  `uname -m` would confirm what architecture it is.

kipusoep, 2020-05-12:
Yeah:
```
root@DietPi:/home/dietpi# uname -m
armv7l
```

So I should look for another distro I guess :-)

mbentley, 2020-05-12:
For a 3b+, it's more than sufficient as you aren't going to see the benefits of a 64 bit OS.  I would supply an `armv7l` if I could get the Debian packages to actually work using the cross-compilation support in Docker.  I think there was something funky in the .deb files that were being used so I will have to try again.

kipusoep, 2020-05-12:
I'll try the images from the repo mentioned in #31 for now :-)

mbentley, 2020-05-12:
👍 Probably the best to get up and running quicker as I just tired to build the image using the new .deb and either they're doing something weird in the .deb or it is something with the multi-arch Docker buildx capabilities.  I don't have any `armv7l` operating systems right now to test natvely.

kipusoep, 2020-05-13:
Thanks, got the other image up and running :-)

closed 2020-05-13 by kipusoep

Soxcks, 2020-05-25:
@mbentley I'm trying this on an old v1 Model B. `name -m` returns `armv61`. I am receiving the same error:
```standard_init_linux.go:190: exec user process caused "exec format error"```

Any ideas/suggestions?

mbentley, 2020-05-25:
That error is due to the architecture difference. Not sure if anyone has an `armv6l` image.

mbentley, 2020-05-31:
I just added an `armv7l` image which should work for a Raspberry Pi.