mbentley/docker-omada-controller #37: Omada error: Failed to login!

Nephelo, 2020-05-14:
Thank you very much for this docker image, which simplifies the usage of the omada controller :-)

Unfortunately I've a strange issue: I setup the Omada Controller together with my AP. After some time I'm not able to sign into the oama controller any more. I get the message "Failed to login!". I'm using the correct password for sure. After that, I deleted the volumes and configured Omada again and now, after a few days the same happened again.

Do you have an idea what's going wrong or where I can find some additinal logs?

mbentley, 2020-05-14:
Could you provide the Docker commands used to run the controller?  Anything you're seeing around what is happening like did you re-create/ re-deploy the container?

Nephelo, 2020-05-14:
Thanks for your quick answer!

Sure! That's my run command:

```
docker run -d \
  --name omada-controller \
  --restart unless-stopped \
  -e TZ=Etc/UTC \
  -e SMALL_FILES=false \
  -p 8088:8088 \
  -p 8043:8043 \
  -p 27001:27001/udp \
  -p 27002:27002 \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29812:29812 \
  -p 29813:29813 \
  -v /home/administrator/volumes/omada/data:/opt/tplink/EAPController/data \
  -v /home/administrator/volumes/omada/work:/opt/tplink/EAPController/work \
  -v /home/administrator/volumes/omada/logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:3.2
```
I tried to re-create the container (`docker rm -f omada-controller`) without success. I also tried to restart the system (Ubuntu 18.04.4 VM). The only thing that happend in the background was a reboot of the VM-Host-System (Microsoft Patchday), but this shouldn't affect the container.

mbentley, 2020-05-14:
Very strange.  I have mine auto-updating and re-creating the container weekly with no issues.  I'd check the container logs to see if there is any indication of what is going on.

Nephelo, 2020-05-16:
Thanks for your help. Unfortunately I can't see anything interersting in the logs:
```
2020-05-12 15:58:15 [main] [INFO]-[SourceFile:139] - mongodb process id is :33
2020-05-12 15:58:15 [main] [INFO]-[SourceFile:120] - Mongo DB server started
2020-05-12 15:58:17 [main] [INFO]-[ContextHandler.java:2040] - Initializing Spring root WebApplicationContext
2020-05-12 15:58:17 [pool-5-thread-1] [INFO]-[SourceFile:91] - Omada Controller isn't prepared to handle event
2020-05-12 15:58:22 [pool-5-thread-2] [INFO]-[SourceFile:91] - Omada Controller isn't prepared to handle event
2020-05-12 15:58:25 [main] [INFO]-[SourceFile:41] - monitor context initialing...
2020-05-12 15:58:25 [main] [INFO]-[ContextHandler.java:2040] - Initializing Spring FrameworkServlet 'springMVC'
2020-05-12 15:58:25 [main] [INFO]-[SourceFile:152] - no need to compatible db.
2020-05-12 15:58:26 [main] [INFO]-[SourceFile:20] - Omada Controller started
2020-05-12 15:58:26 [Thread-18] [INFO]-[SourceFile:83] - Starting PF module.
2020-05-12 15:58:26 [Thread-18] [INFO]-[SourceFile:160] - Starting SCMP Server.
2020-05-12 15:59:05 [qtp434398524-61] [INFO]-[SourceFile:232] - CloudClient is stopped.
2020-05-12 15:59:05 [qtp434398524-61] [INFO]-[SourceFile:178] - Cloud service is stopped.
2020-05-12 15:59:05 [qtp434398524-61] [INFO]-[SourceFile:94] - Stopping PF module.
2020-05-12 15:59:05 [Thread-19] [INFO]-[SourceFile:534] - cleanExpiredCacheThread is interrupted.
2020-05-12 15:59:05 [qtp434398524-61] [INFO]-[SourceFile:554] - Clean expired cache thread is destroyed.
2020-05-12 15:59:05 [qtp434398524-61] [INFO]-[SourceFile:181] - Stopping SCMP Server.
2020-05-12 15:59:05 [Thread-20] [INFO]-[SourceFile:393] - handleRequestThread is interrupted.
2020-05-12 15:59:05 [Thread-20] [INFO]-[SourceFile:435] - SCMP request handle thread is ended.
2020-05-12 15:59:05 [qtp434398524-61] [INFO]-[SourceFile:453] - SCMP request handle thread is destroyed.
2020-05-12 15:59:05 [qtp434398524-61] [INFO]-[SourceFile:287] - clear proxy connection info
2020-05-13 16:59:50 [pool-6-thread-1] [INFO]-[SourceFile:50] - Connected device time out for mac: XXXXXXX
2020-05-13 16:59:50 [pool-6-thread-1] [INFO]-[SourceFile:104] - Ap(XXXXXXX) is disconnected; timeout type : CONNECTED_TIMEOUT.
2020-05-13 17:35:45 [pool-6-thread-1] [INFO]-[SourceFile:50] - Connected device time out for mac: XXXXXX
2020-05-13 17:35:45 [pool-6-thread-1] [INFO]-[SourceFile:104] - Ap(XXXXXXX) is disconnected; timeout type : CONNECTED_TIMEOUT.
2020-05-14 18:13:20 [main] [INFO]-[SourceFile:139] - mongodb process id is :34
2020-05-14 18:13:20 [main] [INFO]-[SourceFile:120] - Mongo DB server started
2020-05-14 18:13:21 [main] [INFO]-[ContextHandler.java:2040] - Initializing Spring root WebApplicationContext
2020-05-14 18:13:23 [pool-5-thread-1] [INFO]-[SourceFile:91] - Omada Controller isn't prepared to handle event
2020-05-14 18:13:28 [pool-5-thread-2] [INFO]-[SourceFile:91] - Omada Controller isn't prepared to handle event
2020-05-14 18:13:28 [main] [INFO]-[SourceFile:41] - monitor context initialing...
2020-05-14 18:13:28 [main] [INFO]-[ContextHandler.java:2040] - Initializing Spring FrameworkServlet 'springMVC'
2020-05-14 18:13:28 [main] [INFO]-[SourceFile:152] - no need to compatible db.
2020-05-14 18:13:28 [main] [INFO]-[SourceFile:20] - Omada Controller started
2020-05-14 18:13:28 [Thread-17] [INFO]-[SourceFile:47] - Cloud service is forbidden.
2020-05-14 18:21:52 [main] [INFO]-[SourceFile:139] - mongodb process id is :31
2020-05-14 18:21:52 [main] [INFO]-[SourceFile:120] - Mongo DB server started
2020-05-14 18:21:46 [main] [INFO]-[ContextHandler.java:2040] - Initializing Spring root WebApplicationContext
2020-05-14 18:21:46 [pool-5-thread-1] [INFO]-[SourceFile:91] - Omada Controller isn't prepared to handle event
2020-05-14 18:21:51 [pool-5-thread-2] [INFO]-[SourceFile:91] - Omada Controller isn't prepared to handle event
2020-05-14 18:21:56 [pool-5-thread-3] [INFO]-[SourceFile:91] - Omada Controller isn't prepared to handle event
2020-05-14 18:22:01 [pool-5-thread-1] [INFO]-[SourceFile:91] - Omada Controller isn't prepared to handle event
2020-05-14 18:22:06 [pool-5-thread-2] [INFO]-[SourceFile:91] - Omada Controller isn't prepared to handle event
2020-05-14 18:22:08 [main] [INFO]-[SourceFile:41] - monitor context initialing...
2020-05-14 18:22:09 [main] [INFO]-[ContextHandler.java:2040] - Initializing Spring FrameworkServlet 'springMVC'
2020-05-14 18:22:09 [main] [INFO]-[SourceFile:152] - no need to compatible db.
2020-05-14 18:22:09 [main] [INFO]-[SourceFile:20] - Omada Controller started
2020-05-14 18:22:09 [Thread-17] [INFO]-[SourceFile:47] - Cloud service is forbidden.
```

mbentley, 2020-05-31:
I'd suspect if there is an issue with data loss, it might be related to MongoDB as I'd expect all data to be stored there.  Anything in the `mongod.log` file?

Nephelo, 2020-06-02:
I don't think it's data loss. I backuped the volumes (also containing MongoDB) directly after configuring the controller. After restoring, I was still not able to login to the controller. An other strange thing: last time I enabled the Cloud login. Using the TP Cloud account I'm still able to access the Omada Controller, but no via the local web interface. 

I opened a ticket at the TP Support, but still waiting for a response.

Nephelo, 2020-06-12:
To keep you up to date: unfortunately, the TPLink support neither had an idea now to solve the problem nor how to find additional information in the logs. Discussing technical details was not possible. A bit of disappointing for a buisness product.

Luckily it is working now again (without changing anything). I assume there is some strange condition in my network, wich causes this effect. (My password contains only letters and numbers, no special characters)

If you like, you can close this issue. I think we won't find any further information. Nevertheless, thank you very much for your support!

mbentley, 2020-06-12:
Well that's disappointing.  Hopefully it doesn't happen again!

closed 2020-06-12 by mbentley