mbentley/docker-omada-controller #38: Many copies of index.html.xxxxx

yaronr, 2020-05-29:
Hi
I peeked inside a running container using Docker exec.
ls for /opt/tplink/EAPController showed that there are over 100k index.html.xxxxxx files (xxxxx seems to be sequential).
Using tag 3.2, image ID 6efd2937e619

mbentley, 2020-05-29:
Could you give me any more data about how you are deploying?  I have omada-controller running tag `3.2` running using `sha256:1d5eaaeda6da1bc3cd993c81ff1419b0cc9acef65e2a60adfe4c131653efeda9` and I do not see the `index.html` files as you mention.

Also, could you get me the following output:
1.) `docker images --digests --filter reference=mbentley/omada-controller`
2.) `docker diff <container-name-or-id>`
3.) `docker logs <container-name-or-id>`

yaronr, 2020-05-30:
Hi
I'm running it inside a NAS device called QNAP. Can you be more specific?
1/
mbentley/omada-controller   3.2                 sha256:2a7d128cf455840dbf2449c94fee6a410403935389f380cf1dfdbe159385af69   6efd2937e619        6 weeks ago         353MB
mbentley/omada-controller   latest              sha256:a00e26557808b82844d589a6a58e0d0c67e311756d605296d6e822ffda51531b   8bc7cac98c7b        6 weeks ago         353MB
mbentley/omada-controller   <none>              sha256:fd3bf5c0fef90e26405b93ba68c50d05ac2d760ed4f8243fad8cfd3301a6f322   7879bec35bd2        8 months ago        349MB

2/ [~] # docker diff c743832a3286
C /opt/tplink/EAPController
A /opt/tplink/EAPController/index.html
A /opt/tplink/EAPController/index.html.1
A /opt/tplink/EAPController/index.html.10
A /opt/tplink/EAPController/index.html.100
A /opt/tplink/EAPController/index.html.1000
A /opt/tplink/EAPController/index.html.10000
A /opt/tplink/EAPController/index.html.100000
A /opt/tplink/EAPController/index.html.100001
A /opt/tplink/EAPController/index.html.100002
A /opt/tplink/EAPController/index.html.100003
A /opt/tplink/EAPController/index.html.100004
A /opt/tplink/EAPController/index.html.100005
A /opt/tplink/EAPController/index.html.100006
A /opt/tplink/EAPController/index.html.100007
A /opt/tplink/EAPController/index.html.100008
A /opt/tplink/EAPController/index.html.100009
A /opt/tplink/EAPController/index.html.10001
A /opt/tplink/EAPController/index.html.100010
... it goes on and on...
A /opt/tplink/EAPController/index.html.99999
C /tmp
A /tmp/mongodb-27217.sock
C /var/lib/apt/lists
D /var/lib/apt/lists/auxfiles
A /var/lib/apt/lists/lock
D /var/lib/apt/lists/partial
C /var/lib/dpkg/lock
C /var/lib/dpkg/lock-frontend

(I tried to do apt-get update from within the container, so that's why apt is here)
(I'm running in an un-tagged vlan environment, and although I could get omada to finally see and manage the EAP 245s, it is unable to access the Internet - still trying to figure out what's going on there. I don't think it's related to index.html - but any ideas?)

3/ May 29, 2020 6:27:30 PM org.hibernate.validator.internal.util.Version <clinit>
INFO: HV000001: Hibernate Validator 4.3.1.Final
Omada Controller started
INFO: Time zone set to 'Etc/UTC'
INFO: Starting Omada Controller as user omada
SLF4J: com.tp_link.eap.util.m.a
SLF4J: The following set of substitute loggers may have been accessed
SLF4J: during the initialization phase. Logging calls during this
SLF4J: phase were not honored. However, subsequent logging calls to these
SLF4J: loggers will work as normally expected.
SLF4J: See also http://www.slf4j.org/codes.html#substituteLogger
SLF4J: com.tp_link.eap.util.m.a
startup...
May 30, 2020 8:41:24 AM org.hibernate.validator.internal.util.Version <clinit>
INFO: HV000001: Hibernate Validator 4.3.1.Final
Omada Controller started

mbentley, 2020-05-31:
Not really sure what that might be.  Are you deploying the container any differently than the examples in the readme?

yaronr, 2020-05-31:
Not really.. Here's docker inspect:
`[~] # docker inspect c743832a3286
[
    {
        "Id": "c743832a32860ba60980cd74a93b3f0f097b57950cac6b25d1f853495cf41c85",
        "Created": "2020-04-13T14:51:48.18283681Z",
        "Path": "/entrypoint.sh",
        "Args": [
            "/opt/tplink/EAPController/jre/bin/java",
            "-server",
            "-Xms128m",
            "-Xmx1024m",
            "-XX:MaxHeapFreeRatio=60",
            "-XX:MinHeapFreeRatio=30",
            "-XX:+HeapDumpOnOutOfMemoryError",
            "-XX:-UsePerfData",
            "-Deap.home=/opt/tplink/EAPController",
            "-cp",
            "/opt/tplink/EAPController/lib/*:",
            "com.tp_link.eap.start.EapLinuxMain"
        ],
        "State": {
            "Status": "running",
            "Running": true,
            "Paused": false,
            "Restarting": false,
            "OOMKilled": false,
            "Dead": false,
            "Pid": 32754,
            "ExitCode": 0,
            "Error": "",
            "StartedAt": "2020-05-30T08:40:38.581168169Z",
            "FinishedAt": "2020-05-30T08:38:34.340448093Z",
            "Health": {
                "Status": "healthy",
                "FailingStreak": 0,
                "Log": [
                    {
                        "Start": "2020-05-31T08:52:20.045617328+03:00",
                        "End": "2020-05-31T08:52:21.948020207+03:00",
                        "ExitCode": 0,
                        "Output": ""
                    },
                    {
                        "Start": "2020-05-31T08:52:52.493627139+03:00",
                        "End": "2020-05-31T08:52:54.415663837+03:00",
                        "ExitCode": 0,
                        "Output": ""
                    },
                    {
                        "Start": "2020-05-31T08:53:24.659622931+03:00",
                        "End": "2020-05-31T08:53:26.506951624+03:00",
                        "ExitCode": 0,
                        "Output": ""
                    },
                    {
                        "Start": "2020-05-31T08:53:56.650607235+03:00",
                        "End": "2020-05-31T08:53:58.695161152+03:00",
                        "ExitCode": 0,
                        "Output": ""
                    },
                    {
                        "Start": "2020-05-31T08:54:28.816751811+03:00",
                        "End": "2020-05-31T08:54:30.687228896+03:00",
                        "ExitCode": 0,
                        "Output": ""
                    }
                ]
            }
        },
[... 256 lines cut ...]
            "MacAddress": "",
            "Networks": {
                "qnet-static-bond0-945f0e": {
                    "IPAMConfig": {
                        "IPv4Address": "10.1.1.11"
                    },
                    "Links": null,
                    "Aliases": [
                        "c743832a3286"
                    ],
                    "NetworkID": "7bf1d0b3eec6019dd80b3d336a81eed5d25c8207731d34862c5983f22d27afeb",
                    "EndpointID": "bce4c515f50f30f3b00df169f3ad397d77dcf1213f58adff085587eba78ec9a8",
                    "Gateway": "10.1.0.1",
                    "IPAddress": "10.1.1.11",
                    "IPPrefixLen": 23,
                    "IPv6Gateway": "",
                    "GlobalIPv6Address": "",
                    "GlobalIPv6PrefixLen": 0,
                    "MacAddress": "02:42:dd:59:4b:78",
                    "DriverOpts": null
                }
            }
        }
    }
]`

mbentley, 2020-05-31:
Ah, I am almost certain it's the health check.  I personally do not use the health check so that is why I didn't see it.

commit 2a5cda4e (2020-05-31): Fixed healthcheck to not save file; changed output; fixes #38
commit fbfc5781 (2020-05-31): Merge pull request #40 from mbentley/fix-healthcheck

Fixed healthcheck to not save file; changed output; fixes #38
closed 2020-05-31 by mbentley