mbentley/docker-omada-controller #39: Constantly Restarting

boydspace, 2020-05-31:
After installing via:

```
docker run -d \
  --name omada-controller \
  --restart unless-stopped \
  -e TZ=America/New_York \
  -e SMALL_FILES=true \
  -p 8088:8088 \
  -p 8043:8043 \
  -p 27001:27001/udp \
  -p 27002:27002 \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29812:29812 \
  -p 29813:29813 \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:3.2-arm64
```

I then run docker container ls -a

and constantly get an output of `Restarting (1) X seconds ago `

It never seems to start. What is going on and where do I look for more information?

mbentley, 2020-05-31:
The output from `docker logs omada-controller` and the output of `uname -a` would be a good start.

boydspace, 2020-05-31:
docker logs omada-controller:

```
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
```

uname -a:
`
Linux pihole 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux`

mbentley, 2020-05-31:
That’d be why - different architecture. See #31

closed 2020-05-31 by mbentley

mbentley, 2020-05-31:
I just added an `armv7l` image which should work for a Raspberry Pi.

boydspace, 2020-05-31:
You are amazing! Thank you!
—
Fr. Andrew Boyd
Parochial Vicar
Catholic Community of Meadville
aboyd@eriercd.org <mailto:aboyd@eriercd.org>
https://fatherboyd.com <https://fatherboyd.com/>
Office: (814) 336-1112