mbentley/docker-omada-controller #42: I can't login to Omada

adrianfime27, 2020-06-02:
I try enter in the soft Omada but i can't, i create an account on the official page Omada Controller but not work when want enter on the docker. It throws me a popup that my username is invalid. And if instead I try to recover my password, it also launches a pop-up window that says Email server is not set. Please check your settings and try again.

adrianfime27, 2020-06-03:
I run the container like this:
docker run -d   --name omada-controller   --restart unless-stopped   -e TZ=Etc/UTC   -e SMALL_FILES=false   -p 8088:8088   -p 8043:8043   -p 27001:27001/udp   -p 27002:27002   -p 29810:29810/udp   -p 29811:29811   -p 29812:29812   -p 29813:29813   -v omada-data:/opt/tplink/EAPController/data   -v omada-work:/opt/tplink/EAPController/work   -v omada-logs:/opt/tplink/EAPController/logs   mbentley/omada-controller:latest

mbentley, 2020-06-03:
So have you setup the Omada controller successfully through the setup wizard?  If you haven't and can start clean, I would suggest giving that a shot

```
docker stop omada-controller
docker rm omada controller
docker volume rm omada-data omada-work omada-logs
```

and then run the container again.

mbentley, 2020-07-16:
If this continues to cause issues, please reopen this.

closed 2020-07-16 by mbentley