mbentley/docker-omada-controller #43: Create SSL certificate

bartlammers, 2020-06-02:
Hi, thanks for your work on this container! 

I have it running using a docker-compose file, but when trying to access the container it keeps complaining about an invalid SSL certificate (because there is none). Would it be possible to add one? If so, can you point me in the right direction perhaps?

The container can be ran without one, by accepting the risks ofc, but a selfsigned certificate might be nicer

mbentley, 2020-06-03:
It would probably be a bit of a pain to do it natively in the image from what I can see in these steps: https://community.tp-link.com/en/business/forum/topic/150083

I personally just run an nginx reverse proxy in front of the Omada controller.  If you join nginx to the same Docker network as the Omada Controller container, you can make sure that nginx only exposes your environment with your valid SSL certificate installed as the idea would be that the connection between the containers would be a trusted network that isn't externally available from the custom bridge network.

mbentley, 2020-06-03:
Sure thing.  Let me know if you run into any issues.  I actually don't have a reverse proxy w/nginx and SSL in Docker example ready but it should be pretty straight forward.