mbentley/docker-omada-controller #44: Cannot access the login page on a Synology NAS

AlexanderSanchezNL, 2020-06-03:
I've tried to find the answer all over the Internet, but unfortunately I cannot find a solution for my problem: I get a ERR_CONNECTION_REFUSED when I try to open the Omada Controller on my Synology NAS. I use the NAS IP on my local LAN and port 8043 for https (8088 on http also doesn't work). Am I doing something wrong here?

I have a Synology DS718+ and I installed Docker + you Omada controller (latest version). The installation went smoothly and I don't see any errors in the logs. It's just that my browser connection is being refused. I disabled the Firewall on the NAS.

Can you help me?

mbentley, 2020-06-03:
You didn't expose any of the ports it would appear.  Could you give me your run command?

AlexanderSanchezNL, 2020-06-03:
Ehmmm, without sounding like a complete dumb*ss: I installed Docker, searched for Omada and installed your package. Then I clicked "Launch" and went into Advanced Settings where I enabled auto-restart. I didn't meddle with the Port Settings, they looked ok to me. And voilà, here we are :-)

So, I didn't run a run command script...

mbentley, 2020-06-03:
Ah ok.  I am not familiar with how Synology packages up Docker containers but the gist of it is that the following ports need to be exposed from the container:

```
  -p 8088:8088 \
  -p 8043:8043 \
  -p 27001:27001/udp \
  -p 27002:27002 \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29812:29812 \
  -p 29813:29813 \
```

AlexanderSanchezNL, 2020-06-03:
Yeah, I now figured as much... :-) Sorry.

I ran the following run command, and now it seems to be working:

docker run -d \
  --name omada-controller \
  --restart unless-stopped \
  -e TZ=Europe/Amsterdam \
  -e SMALL_FILES=false \
  -p 8088:8088 \
  -p 8043:8043 \
  -p 27001:27001/udp \
  -p 27002:27002 \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29812:29812 \
  -p 29813:29813 \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:latest

mbentley, 2020-06-03:
👍  Happy to help if you run into any other issues.

closed 2020-06-03 by mbentley