mbentley/docker-omada-controller #45: Discovery requires network_mode: host

Javerre, 2020-06-19:
Hi, 

I've just pulled de2206826ae4 and created an application with the following docker-compose settings:

```
version: '3'
networks:
  omada_net:
services:
  omada:
    container_name: omada
    image: mbentley/omada-controller:latest
    networks:
      - omada_net
    ports:
      - '8088:8088'
      - '8043:8043'
      - '27001:27001/udp'
      - '27002:27002'
      - '27217:27217'
      - '29810:29810/udp'
      - '29811:29811'
      - '29812:29812'
      - '29813:29813'
    volumes:
      - /share/Container/Omada/data:/opt/tplink/EAPController/data
      - /share/Container/Omada/work:/opt/tplink/EAPController/work
      - /share/Container/Omada/logs:/opt/tplink/EAPController/logs
      - /etc/localtime:/etc/localtime:ro
    restart: unless-stopped
```

This can run, connect to the web, register with the TP-Link cloud service, etc. but does **not detect my EAP devices**. Checking the logs, it appears that MongoDB is not starting.

If instead I use the following docker-compose file:

```
version: '3'
services:
  omada:
    network_mode: host
    container_name: omada
    image: mbentley/omada-controller:latest
    volumes:
      - /share/Container/Omada/data:/opt/tplink/EAPController/data
      - /share/Container/Omada/work:/opt/tplink/EAPController/work
      - /share/Container/Omada/logs:/opt/tplink/EAPController/logs
      - /etc/localtime:/etc/localtime:ro
    restart: unless-stopped
```

everything works as expected, MongoDB starts and Omada detects my EAP devices fine.

mbentley, 2020-06-19:
Well, you're right about it not working just with ports published.  I honestly don't run mine that way; I use the docker `macvlan` driver but using `host` networking also does work.  I'll have to do some searching to see if I can figured out how the discovery works exactly to see if there is any way to get it working without needing to use host networking.

mbentley, 2020-06-19:
Ah, I found a reference that it’s using broadcast and with the private network being on a separate multicast domain, it won’t be able to see the APs.  So you either need to use host networking or macvlan.

closed 2020-06-19 by mbentley

Javerre, 2020-06-24:
Well, I read [this article from TP-link](https://community.tp-link.com/en/business/kb/detail/362) and there it makes it clear that the EAP devices are the ones that broadcast to UDP 29810 so if the docker network is mapped onto the local subnet though a port directive it will get hit and discovery should work. I don’t think this is the issue, I think the container simply malfunctions for some other internal reason when its not on host or macvlan but is in a locked down network environment.

Furthermore there are other users out there who are managing remote sites and getting discovery to work with a firewall between the controller and the EAP units having only 29810/UDP plus 29811-29813 open.

mbentley, 2020-06-24:
Using broadcast like that works both ways - the broadcast traffic can’t cross network domains.

Javerre, 2020-06-24:
That's not what the documentation shows:
[image]
In this situation I think the EAP should discover the controller's docker host IP just fine

Javerre, 2020-06-24:
Going the other way would of course be a problem as the controller running in docker cannot scan the host subnet. So if (and I don't know this is the case) the _controller_ tries to initiate discovery by broadcasting to _EAP units_ when its powered on, _that_ would fail. Perhaps that was what I experienced? I'm not sure I power cycled the EAPs after starting the controller so they may have stopped broadcasting by the time the controller started to listen. 

I can check later this week.

mbentley, 2020-06-24:
I also tried to setup a new controller with a new WAP the other day when you posted originally and I couldn’t get the WAP to show up at all when mapping ports.  I am curious...the diagram is showing that the controller responding with its IP address. If that’s the case, it wouldn’t be able to find the IP of the host network interface to return to the AP.  Not sure if there is any way to override that setting to tell it the IP it should respond with.

Javerre, 2020-06-24:
Sorry to be so dim, I think you are absolutely right. [Reading around](https://community.tp-link.com/en/business/forum/topic/107014), to detect EAPs on a different subnet the controller has to be running on a system with a public IP that can be routed to from the EAP.

Sorry to have wasted your time, but thanks your support!

dmtroyer, 2020-06-25:
I'd be curious to try out macvlan mode, as my eaps are constantly dropping to disconnected on the controller, but I'm a little bit overwhelmed by the networking side of it. Any pointers?

mbentley, 2020-06-25:
The gist of it is that you need to create a `macvlan` network:

```
docker network create -d macvlan --subnet=192.168.0.0/24 --gateway=192.168.0.1 -o parent=eth0 macvlan1
```

Update the `subnet`, `gateway`, and `parent` with options to match your network's settings and your network interface.

Then when you run the container, you need to specify the `macvlan` network and IP you want to assign (think of this just like a static IP on your network):

```
--network macvlan1
--ip 192.168.0.100
```

There are some gotchas around macvlan such as you won't be able to access the IP from the container host itself unless you utilize a dedicate network interface for macvlan traffic.  If you don't need to access the IP for your container from the host itself, I wouldn't worry too much about it.  In my case, I had extra NICs so I could easily dedicate one.  Here is my `interfaces` file that just brings up the network interface that I can use for `macvlan`:

`/etc/network/interfaces`

```
allow-hotplug eth1
iface eth1 inet manual
        pre-up ip link set $IFACE up
        post-down ip link set $IFACE down
```

https://docs.docker.com/network/macvlan/

dmtroyer, 2020-06-26:
@mbentley thank you so much! that was very helpful, I got the macvlan to work, but my APs still cycle between connected and disconnected basically every 30 seconds in the controller :-(

mbentley, 2020-06-26:
Hmm, so it is consistent with multiple APs?  Tried doing some quick tests like continually pinging the APs to see if there are issues?  How is your network setup?  Using mesh?  Hard wired?

dmtroyer, 2020-06-29:
@mbentley ping is good
```
--- 10.0.0.6 ping statistics ---
500 packets transmitted, 500 received, 0% packet loss, time 504048ms
rtt min/avg/max/mdev = 0.394/1.235/2.490/0.671 ms
```
```
--- 10.0.0.7 ping statistics ---
500 packets transmitted, 500 received, 0% packet loss, time 503669ms
rtt min/avg/max/mdev = 0.372/1.299/2.327/0.681 ms
```
mesh is turned on (as it is by default?), but they are both hard wired. I couldn't find a whole lot of info about mesh on these units when hardwired, but it seemed like there were some ancillary benefits, anyhow I'll try turning that off.

Networking is a little weird, there are lots of places for it to go wrong. The container is running in an ubuntu vm on a freenas (freebsd) installation. I'm going to try running the controller on my laptop and see if I see the same behavior.

mbiebl, 2020-07-16:
Hm, I've been using bridged network + exposed ports and didn't have any problems so far with APs not being discovered.

mbentley, 2020-07-16:
Hmm, that's odd.  I couldn't get it to work on my server.  Makes me really curious what is different/required to make that work.

returntrip, 2020-08-28:
> Hm, I've been using bridged network + exposed ports and didn't have any problems so far with EAPs not being discovered.

Same here, discovery is working with bridged network and exposed ports.

DePingus, 2020-09-18:
> > Hm, I've been using bridged network + exposed ports and didn't have any problems so far with EAPs not being discovered.
> 
> Same here, discovery is working with bridged network and exposed ports.

Hi. Any chance you can post your docker compose with the ports you used?

mbentley, 2020-09-18:
I'm pretty sure it was exactly what was posted in the original post of this issue.

DePingus, 2020-09-18:
> I'm pretty sure it was exactly what was posted in the original post of this issue.

Excellent. Thanks. The controller is up but my EAP225 doesn't arrive until tomorrow. I'll report back with my findings this weekend.

DePingus, 2020-09-18:
Just to follow up my preveious post. I setup a v3.2 Omada controller using your container image (thanks for keeping that available). My new EAP225 (hardware v3.8 / firmware v2.7.0 Build 20200113 Rel. 38287) was discovered immediately and adopted/provisioned with no problems. I opted to go with controller v3.2 because I don't want any of the SDN garbage in v4.1.

Here is my docker command:
```
docker create \
  --name omada-controller \
  --restart unless-stopped \
  --mount source=omada-data,target=/opt/tplink/EAPController/data \
  --mount source=omada-work,target=/opt/tplink/EAPController/work \
  --mount source=omada-logs,target=/opt/tplink/EAPController/logs \
  -p 8088:8088 \
  -p 8043:8043 \
  -p 27001:27001/udp \
  -p 27002:27002 \
  -p 27217:27217 \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29812:29812 \
  -p 29813:29813 \
  -e MANAGE_HTTP_PORT=8088 \
  -e MANAGE_HTTPS_PORT=8043 \
  -e PORTAL_HTTP_PORT=8088 \
  -e PORTAL_HTTPS_PORT=8843 \
  -e SHOW_SERVER_LOGS=true \
  -e SHOW_MONGODB_LOGS=false \
  -e TZ=America/New_York \
  -e PUID=1000 \
  -e PGID=102 \
  mbentley/omada-controller:3.2
```