mbentley/docker-omada-controller #47: New (major) upstream release 4.1.5

mbiebl, 2020-07-14:
Hi,

there is a new major upstream release available at https://static.tp-link.com/2020/202007/20200714/Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz

See also https://www.tp-link.com/de/omada-sdn/controller-upgrade/#content-1

mbentley, 2020-07-14:
Awesome, thanks for the heads up.  Time to see what we can do :)

mbiebl, 2020-07-14:
A first cursory glance: The provided tar.gz no longer ships a bundled JRE (so I guess we should pull java8-runtime-headless) and no mongodb-server.
The latter is more tricky. We now either need a separate docker container providing a mongodb server or install and start a mongodb server ourselves afaics

mbentley, 2020-07-14:
Yeah, I was noticing the same thing.  We won't be able to run a separate container of mongodb as it still expected to be able to launch mongodb the same way older versions did.  I have it building and working; I'm just trying to clean up some janky things I am seeing in the .properties files (relative paths which basically mean you have to launch it from the bin directory unless you want it to fail 🤦 )

mbentley, 2020-07-14:
See what I've pushed in #48.  I still need to handle the `SMALL_FILES` option as the properties file has been consolidated into an omada.properties.  It seems that variables are no longer parsed in the properties files as my attempts to use `eap.home=/opt/tplink/EAPController` have failed.

mbentley, 2020-07-14:
Also, for some reason the logs aren't going to the console; currently doing some hacky stuff...
https://github.com/mbentley/docker-omada-controller/pull/48/files#diff-8967333c08582da27f10d610dc6443dfR52

mbentley, 2020-07-14:
I just tested the upgraded version and unfortunately it is not yet working so I can't fully test it as it would appear there are not supported firmwares available for my access points.

mbentley, 2020-07-15:
So far, so good!

mbentley, 2020-07-15:
And just to update; everything regarding the container seems to be fine; I'll be curious to see what TP-Links recommendations are in terms of any specifics around requirements for Java and MongoDB.  I've seen a few things fail related to auto-backups and some emails failing to send on events - not sure if it a Java version issue or an app issue.

Example failure when it should be sending an email:

```
Jul 15, 2020 7:22:05 AM com.google.common.eventbus.EventBus$LoggingHandler handleException
SEVERE: Exception thrown by subscriber method a(com.tplink.omada.manager.log.mail.a) on subscriber com.tplink.omada.manager.log.mail.MailLogEventListener@3a225534 when dispatching event: com.tplink.omada.manager.log.mail.a@79426e82
java.lang.NullPointerException
        at com.tplink.omada.manager.log.mail.MailLogEventListener.a(SourceFile:172)
        at com.tplink.omada.manager.log.mail.MailLogEventListener.a(SourceFile:94)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at com.google.common.eventbus.Subscriber.invokeSubscriberMethod(Subscriber.java:87)
        at com.google.common.eventbus.Subscriber$SynchronizedSubscriber.invokeSubscriberMethod(Subscriber.java:144)
        at com.google.common.eventbus.Subscriber$1.run(Subscriber.java:72)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
```

mbiebl, 2020-07-15:
The `.deb` package provided by TP-Link has the following dependencies:
```
 Depends: adduser, jsvc (>=1.0.8), curl,
  mongodb-server (>= 3.0.15) | mongodb-10gen (>= 3.0.15) | mongodb-org-server (>= 3.0.15),
  mongodb-server (<< 1:3.6.18) | mongodb-10gen (<< 3.6.18) |  mongodb-org-server (<< 3.6.18),
  java8-runtime-headless
```

mbentley, 2020-07-15:
We should be set; I am not sure if `curl` is actually required or if it was just part of the install/config (I seem to remember something where it complained if I didn't have `curl` installed when running the install script) but I will add it anyway.

mbiebl, 2020-07-15:
> We should be set; I am not sure if `curl` is actually required or if it was just part of the install/config (I seem to remember something where it complained if I didn't have `curl` installed when running the install script) but I will add it anyway.

I just checked the `.deb`:
```
opt/tplink/EAPController/bin/control.sh:# curl
opt/tplink/EAPController/bin/control.sh:CURL=$(command -v curl)
opt/tplink/EAPController/bin/control.sh:	echo "${DESC}: curl not found, please install curl!"
opt/tplink/EAPController/bin/control.sh:    http_code=$(curl -I -m 10 -o /dev/null -s -w %{http_code} http://localhost:${HTTP_PORT}/status)
```
Seems `curl` is only used in the init script, which we don't use. So should be safe to omit.

mbiebl, 2020-07-15:
> Seems `curl` is only used in the init script, which we don't use. So should be safe to omit.

The curl dependency increases the image size by 5M.
If we want to keep curl, we could replace our wget usage.

mbentley, 2020-07-15:
Sounds good; `wget` is smaller so let's just leave it as is.  I just removed `curl` as we can add it back if there is some sort of functionality that is broken due to it being missing later.

mbentley, 2020-07-15:
Fixed by #48

closed 2020-07-15 by mbentley