mbentley/docker-omada-controller #49: Unable to create docker image

joopy, 2020-07-15:
Dockerfile.v4.1.x used

root@omada:~/mydockerbuild# docker build - < Dockerfile.v4.1.x
Sending build context to Docker daemon  4.096kB
Step 1/10 : FROM ubuntu:18.04
18.04: Pulling from library/ubuntu
db182fa07ba5: Pull complete
3e85650936bd: Pull complete
258b890879c4: Pull complete
d0f858a12ece: Pull complete
Digest: sha256:e5b0b89c846690afe2ce325ac6c6bc3d686219cfa82166fc75c812c1011f0803
Status: Downloaded newer image for ubuntu:18.04
 ---> cd3a27be997b
Step 2/10 : MAINTAINER Matt Bentley <mbentley@mbentley.net>
 ---> Running in f2fd2efe0e68
Removing intermediate container f2fd2efe0e68
 ---> 4e94001814c8
Step 3/10 : RUN   echo "**** Install Dependencies ****" &&  apt-get update &&  DEBIAN_FRONTEND="noninteractive" apt-get install --no-install-recommends -y gosu mongodb-server-core net-tools openjdk-8-jre-headless tzdata wget &&  rm -rf /var/lib/apt/lists/* &&  echo "**** Download Omada Controller ****" &&  cd /tmp &&  wget -nv "https://static.tp-link.com/2020/202007/20200714/Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz" &&  echo "**** Extract and Install Omada Controller ****" &&  tar zxvf Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz &&  rm Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz &&  cd Omada_SDN_Controller_* &&  mkdir /opt/tplink/EAPController -vp &&  cp bin /opt/tplink/EAPController -r &&  cp data /opt/tplink/EAPController -r &&  cp properties /opt/tplink/EAPController -r &&  cp webapps /opt/tplink/EAPController -r &&  cp keystore /opt/tplink/EAPController -r &&  cp lib /opt/tplink/EAPController -r &&  cp install.sh /opt/tplink/EAPController -r &&  cp uninstall.sh /opt/tplink/EAPController -r &&  ln -sf $(which mongod) /opt/tplink/EAPController/bin/mongod &&  chmod 755 /opt/tplink/EAPController/bin/* &&  echo "**** Cleanup ****" &&  cd /tmp &&  rm -rf /tmp/Omada_SDN_Controller* &&  echo "**** Setup omada User Account ****" &&  groupadd -g 508 omada &&  useradd -u 508 -g 508 -d /opt/tplink/EAPController omada &&  mkdir /opt/tplink/EAPController/logs /opt/tplink/EAPController/work &&  chown -R omada:omada /opt/tplink/EAPController/data /opt/tplink/EAPController/logs /opt/tplink/EAPController/work
 ---> Running in 9ba4fb09a4e7
**** Install Dependencies ****
Get:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease [242 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [74.6 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports bionic/main armhf Packages [1277 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports bionic/multiverse armhf Packages [157 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports bionic/universe armhf Packages [11.0 MB]
Get:8 http://ports.ubuntu.com/ubuntu-ports bionic/restricted armhf Packages [12.5 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main armhf Packages [834 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe armhf Packages [1170 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports bionic-updates/restricted armhf Packages [11.0 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports bionic-updates/multiverse armhf Packages [4408 B]
Get:13 http://ports.ubuntu.com/ubuntu-ports bionic-backports/main armhf Packages [8247 B]
Get:14 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe armhf Packages [8101 B]
Get:15 http://ports.ubuntu.com/ubuntu-ports bionic-security/multiverse armhf Packages [1849 B]
Get:16 http://ports.ubuntu.com/ubuntu-ports bionic-security/restricted armhf Packages [5504 B]
Get:17 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe armhf Packages [694 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports bionic-security/main armhf Packages [567 kB]
Fetched 16.2 MB in 9s (1776 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
Package mongodb-server-core is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

**E: Package 'mongodb-server-core' has no installation candidate**
The command '/bin/sh -c echo "**** Install Dependencies ****" &&  apt-get update &&  DEBIAN_FRONTEND="noninteractive" apt-get install --no-install-recommends -y gosu mongodb-server-core net-tools openjdk-8-jre-headless tzdata wget &&  rm -rf /var/lib/apt/lists/* &&  echo "**** Download Omada Controller ****" &&  cd /tmp &&  wget -nv "https://static.tp-link.com/2020/202007/20200714/Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz" &&  echo "**** Extract and Install Omada Controller ****" &&  tar zxvf Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz &&  rm Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz &&  cd Omada_SDN_Controller_* &&  mkdir /opt/tplink/EAPController -vp &&  cp bin /opt/tplink/EAPController -r &&  cp data /opt/tplink/EAPController -r &&  cp properties /opt/tplink/EAPController -r &&  cp webapps /opt/tplink/EAPController -r &&  cp keystore /opt/tplink/EAPController -r &&  cp lib /opt/tplink/EAPController -r &&  cp install.sh /opt/tplink/EAPController -r &&  cp uninstall.sh /opt/tplink/EAPController -r &&  ln -sf $(which mongod) /opt/tplink/EAPController/bin/mongod &&  chmod 755 /opt/tplink/EAPController/bin/* &&  echo "**** Cleanup ****" &&  cd /tmp &&  rm -rf /tmp/Omada_SDN_Controller* &&  echo "**** Setup omada User Account ****" &&  groupadd -g 508 omada &&  useradd -u 508 -g 508 -d /opt/tplink/EAPController omada &&  mkdir /opt/tplink/EAPController/logs /opt/tplink/EAPController/work &&  chown -R omada:omada /opt/tplink/EAPController/data /opt/tplink/EAPController/logs /opt/tplink/EAPController/work' returned a non-zero code: 100

mbentley, 2020-07-15:
Yeah, that's expected at the moment - I'll need to create a Dockerfile for the `armv7l`.  We had to use `ubuntu:16.04` in the previous version because of the dependency on MongoDB and I believe there wasn't a package in the 18.04 version but I will check and see.  I didn't want to invest too much time in both `amd64` and `armv7l` right away until I confirmed that there weren't any issues.  Luckily it shouldn't be _too_ bad.  I'll see if I can work on adapting it from https://github.com/mbentley/docker-omada-controller/blob/master/Dockerfile.v3.2.x-armv7l tonight.

The main things would be to start from `Dockerfile.v4.1.x`, change the base image from 18.04 to 16.04, changing the mongodb package to `mongodb` which should be present in 16.04.

joopy, 2020-07-15:
thnx in advance. If I need to test anything let me know.

mbentley, 2020-07-16:
I just successfully built an image for `armv7l`: `mbentley/omada-controller:4.1-armv7l`.  If you're able to trying it out, that would be a huge help!

```
+ docker buildx build --cache-to=type=local,dest=/var/lib/jenkins/buildx-cache/omada-controller-armv7l-test,mode=max --push --progress plain --platform linux/arm/v7 -t mbentley/omada-controller:4.1-armv7l -f Dockerfile.v4.1.x-armv7l .
#1 [internal] load build definition from Dockerfile.v4.1.x-armv7l
#1 transferring dockerfile: 2.49kB done
#1 DONE 0.0s

#2 [internal] load .dockerignore
#2 transferring context: 61B done
#2 DONE 0.0s

#3 [internal] load metadata for docker.io/library/ubuntu:16.04
#3 DONE 1.1s

#4 [1/4] FROM docker.io/library/ubuntu:16.04@sha256:69bc24edd22c270431d1a9e...
#4 resolve docker.io/library/ubuntu:16.04@sha256:69bc24edd22c270431d1a9e6dbf57cfc4a77b2da199462d0251b145fdd7fa538 done
#4 CACHED

#6 [internal] load build context
#6 transferring context: 3.00kB done
#6 DONE 0.0s

#5 [2/4] RUN   echo "**** Install Dependencies ****" &&  apt-get update && ...
#5 0.362 **** Install Dependencies ****
#5 1.207 Get:1 http://ports.ubuntu.com/ubuntu-ports xenial InRelease [247 kB]
#5 1.858 Get:2 http://ports.ubuntu.com/ubuntu-ports xenial-updates InRelease [109 kB]
#5 1.909 Get:3 http://ports.ubuntu.com/ubuntu-ports xenial-backports InRelease [107 kB]
#5 2.046 Get:4 http://ports.ubuntu.com/ubuntu-ports xenial-security InRelease [109 kB]
#5 4.481 Get:5 http://ports.ubuntu.com/ubuntu-ports xenial/main armhf Packages [1486 kB]
#5 5.431 Get:6 http://ports.ubuntu.com/ubuntu-ports xenial/restricted armhf Packages [8491 B]
#5 5.433 Get:7 http://ports.ubuntu.com/ubuntu-ports xenial/universe armhf Packages [9531 kB]
#5 6.008 Get:8 http://ports.ubuntu.com/ubuntu-ports xenial/multiverse armhf Packages [149 kB]
#5 7.072 Get:9 http://ports.ubuntu.com/ubuntu-ports xenial-updates/main armhf Packages [1080 kB]
#5 7.356 Get:10 http://ports.ubuntu.com/ubuntu-ports xenial-updates/restricted armhf Packages [8488 B]
#5 7.356 Get:11 http://ports.ubuntu.com/ubuntu-ports xenial-updates/universe armhf Packages [926 kB]
#5 7.399 Get:12 http://ports.ubuntu.com/ubuntu-ports xenial-updates/multiverse armhf Packages [14.0 kB]
#5 9.267 Get:13 http://ports.ubuntu.com/ubuntu-ports xenial-backports/main armhf Packages [7936 B]
#5 9.389 Get:14 http://ports.ubuntu.com/ubuntu-ports xenial-backports/universe armhf Packages [8707 B]
#5 11.53 Get:15 http://ports.ubuntu.com/ubuntu-ports xenial-security/main armhf Packages [754 kB]
#5 12.24 Get:16 http://ports.ubuntu.com/ubuntu-ports xenial-security/restricted armhf Packages [8480 B]
#5 12.36 Get:17 http://ports.ubuntu.com/ubuntu-ports xenial-security/universe armhf Packages [553 kB]
#5 12.39 Get:18 http://ports.ubuntu.com/ubuntu-ports xenial-security/multiverse armhf Packages [3561 B]
#5 12.66 Fetched 15.1 MB in 11s (1275 kB/s)
#5 12.66 Reading package lists...
#5 21.19 Reading package lists...
#5 29.27 Building dependency tree...
#5 30.16 Reading state information...
#5 30.78 The following additional packages will be installed:
#5 30.78   ca-certificates ca-certificates-java fontconfig-config fonts-dejavu-core
#5 30.78   java-common libavahi-client3 libavahi-common-data libavahi-common3
#5 30.78   libboost-filesystem1.58.0 libboost-program-options1.58.0
#5 30.78   libboost-system1.58.0 libboost-thread1.58.0 libcups2 libdbus-1-3 libexpat1
#5 30.78   libffi6 libfontconfig1 libfreetype6 libgmp10 libgnutls30
#5 30.78   libgoogle-perftools4 libgssapi-krb5-2 libhogweed4 libidn11 libjpeg-turbo8
#5 30.78   libjpeg8 libk5crypto3 libkeyutils1 libkrb5-3 libkrb5support0 liblcms2-2
#5 30.78   libnettle6 libnspr4 libnss3 libnss3-nssdb libp11-kit0 libpcap0.8
#5 30.78   libpcrecpp0v5 libpcsclite1 libpng12-0 libsnappy1v5 libsqlite3-0 libssl1.0.0
#5 30.78   libtasn1-6 libtcmalloc-minimal4 libv8-3.14.5 libx11-6 libx11-data libxau6
#5 30.78   libxcb1 libxdmcp6 libxext6 libxi6 libxrender1 libxtst6 libyaml-cpp0.5v5
#5 30.78   mongodb-clients mongodb-server openssl ucf x11-common
#5 30.78 Suggested packages:
#5 30.78   default-jre cups-common gnutls-bin krb5-doc krb5-user liblcms2-utils pcscd
#5 30.78   libnss-mdns fonts-dejavu-extra fonts-ipafont-gothic fonts-ipafont-mincho
#5 30.78   fonts-wqy-microhei fonts-wqy-zenhei fonts-indic
#5 30.78 Recommended packages:
#5 30.78   dbus krb5-locales
#5 30.79 The following NEW packages will be installed:
#5 30.79   ca-certificates ca-certificates-java fontconfig-config fonts-dejavu-core
#5 30.79   gosu java-common libavahi-client3 libavahi-common-data libavahi-common3
[... 715 lines cut ...]
#9 exporting config sha256:e346c3fb4276cdb628db46babc0722525046c42eb57386c3935b1da558acffb9 0.0s done
#9 pushing layers
#9 pushing layers 49.2s done
#9 pushing manifest for docker.io/mbentley/omada-controller:4.1-armv7l
#9 pushing manifest for docker.io/mbentley/omada-controller:4.1-armv7l 0.8s done
#9 DONE 67.4s

#10 exporting cache
#10 preparing build cache for export 0.1s done
#10 writing layer sha256:0b760e2e684f25b6d807a37487630e4b7c8ea9a10ea0272b499fb9ad58a4741c
#10 writing layer sha256:0b760e2e684f25b6d807a37487630e4b7c8ea9a10ea0272b499fb9ad58a4741c 0.0s done
#10 writing layer sha256:15bcafb819cb1e8d340ca68bdf919d156cde5bb998a5b59d3bc8db1c6c1ab55c done
#10 writing layer sha256:1a94f4ef63bde77e0559a918f38faa154d9170ae4b63880064d2a99b7675fa0d done
#10 writing layer sha256:75ee4e72f0eac9907725f07d80b1a8b4adeee7b9af0097ce074454bec7e24a03
[... cut ...]
#10 writing layer sha256:b862de1ffd0c696287b3ffbe1ba7a10e4118ac683f6759a1a2a6e58bd4e02b77 0.0s done
#10 writing layer sha256:be0d01e22a1f0d428dbbe6be2da34bbdb47878547862e614fb5eac9a9dc17cae
#10 writing layer sha256:be0d01e22a1f0d428dbbe6be2da34bbdb47878547862e614fb5eac9a9dc17cae 0.6s done
#10 writing config sha256:292db9eb60b3f9bd19900ac9bf29ae46c702227c7a86e8eb8632d12b84097b13 done
#10 writing manifest sha256:35419b1955e0f16a96cc97198de71e9e80ca17a071754f2d9681c0e4567bfc26 done
#10 DONE 3.5s
Run condition [Always] enabling perform for step [Builder to mark whether executed]
```

mbentley, 2020-07-16:
I went ahead and merged https://github.com/mbentley/docker-omada-controller/pull/50 based on how my arm64 tests went.  Let me know if you're able to test and if there are any issues, I'll take a look.  I'll hold this open until I get confirmation.

joopy, 2020-07-16:
root@omada:~/mydockerbuild# docker build - < Dockerfile.v4.1.x-armv7l
Sending build context to Docker daemon  4.096kB
Step 1/10 : FROM ubuntu:16.04
16.04: Pulling from library/ubuntu
be0d01e22a1f: Pull complete
0b760e2e684f: Pull complete
15bcafb819cb: Pull complete
aaa652f3eed9: Pull complete
Digest: sha256:69bc24edd22c270431d1a9e6dbf57cfc4a77b2da199462d0251b145fdd7fa538
Status: Downloaded newer image for ubuntu:16.04
 ---> 5d186976a248
Step 2/10 : MAINTAINER Matt Bentley <mbentley@mbentley.net>
 ---> Running in b0c11a543526
Removing intermediate container b0c11a543526
 ---> 2ff99ecad0e5
Step 3/10 : RUN   echo "**** Install Dependencies ****" &&  apt-get update &&  DEBIAN_FRONTEND="noninteractive" apt-get install --no-install-recommends -y gosu mongodb net-tools openjdk-8-jre-headless tzdata wget &&  rm -rf /var/lib/apt/lists/* &&  echo "**** Download Omada Controller ****" &&  cd /tmp &&  wget -nv "https://static.tp-link.com/2020/202007/20200714/Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz" &&  echo "**** Extract and Install Omada Controller ****" &&  tar zxvf Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz &&  rm Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz &&  cd Omada_SDN_Controller_* &&  mkdir /opt/tplink/EAPController -vp &&  cp bin /opt/tplink/EAPController -r &&  cp data /opt/tplink/EAPController -r &&  cp properties /opt/tplink/EAPController -r &&  cp webapps /opt/tplink/EAPController -r &&  cp keystore /opt/tplink/EAPController -r &&  cp lib /opt/tplink/EAPController -r &&  cp install.sh /opt/tplink/EAPController -r &&  cp uninstall.sh /opt/tplink/EAPController -r &&  ln -sf $(which mongod) /opt/tplink/EAPController/bin/mongod &&  chmod 755 /opt/tplink/EAPController/bin/* &&  echo "**** Cleanup ****" &&  cd /tmp &&  rm -rf /tmp/Omada_SDN_Controller* &&  echo "**** Setup omada User Account ****" &&  groupadd -g 508 omada &&  useradd -u 508 -g 508 -d /opt/tplink/EAPController omada &&  mkdir /opt/tplink/EAPController/logs /opt/tplink/EAPController/work &&  chown -R omada:omada /opt/tplink/EAPController/data /opt/tplink/EAPController/logs /opt/tplink/EAPController/work
 ---> Running in c3a49d128a79
**** Install Dependencies ****
Get:1 http://ports.ubuntu.com/ubuntu-ports xenial InRelease [247 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports xenial-updates InRelease [109 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports xenial-backports InRelease [107 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports xenial-security InRelease [109 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports xenial/main armhf Packages [1486 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports xenial/restricted armhf Packages [8491 B]
Get:7 http://ports.ubuntu.com/ubuntu-ports xenial/universe armhf Packages [9531 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports xenial/multiverse armhf Packages [149 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports xenial-updates/main armhf Packages [1080 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports xenial-updates/restricted armhf Packages [8488 B]
Get:11 http://ports.ubuntu.com/ubuntu-ports xenial-updates/universe armhf Packages [926 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports xenial-updates/multiverse armhf Packages [14.0 kB]
Get:13 http://ports.ubuntu.com/ubuntu-ports xenial-backports/main armhf Packages [7936 B]
Get:14 http://ports.ubuntu.com/ubuntu-ports xenial-backports/universe armhf Packages [8707 B]
Get:15 http://ports.ubuntu.com/ubuntu-ports xenial-security/main armhf Packages [754 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports xenial-security/restricted armhf Packages [8480 B]
Get:17 http://ports.ubuntu.com/ubuntu-ports xenial-security/universe armhf Packages [553 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports xenial-security/multiverse armhf Packages [3561 B]
Fetched 15.1 MB in 11s (1346 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  ca-certificates ca-certificates-java fontconfig-config fonts-dejavu-core
  java-common libavahi-client3 libavahi-common-data libavahi-common3
  libboost-filesystem1.58.0 libboost-program-options1.58.0
  libboost-system1.58.0 libboost-thread1.58.0 libcups2 libdbus-1-3 libexpat1
  libffi6 libfontconfig1 libfreetype6 libgmp10 libgnutls30
  libgoogle-perftools4 libgssapi-krb5-2 libhogweed4 libidn11 libjpeg-turbo8
  libjpeg8 libk5crypto3 libkeyutils1 libkrb5-3 libkrb5support0 liblcms2-2
  libnettle6 libnspr4 libnss3 libnss3-nssdb libp11-kit0 libpcap0.8
  libpcrecpp0v5 libpcsclite1 libpng12-0 libsnappy1v5 libsqlite3-0 libssl1.0.0
  libtasn1-6 libtcmalloc-minimal4 libv8-3.14.5 libx11-6 libx11-data libxau6
  libxcb1 libxdmcp6 libxext6 libxi6 libxrender1 libxtst6 libyaml-cpp0.5v5
  mongodb-clients mongodb-server openssl ucf x11-common
Suggested packages:
  default-jre cups-common gnutls-bin krb5-doc krb5-user liblcms2-utils pcscd
  libnss-mdns fonts-dejavu-extra fonts-ipafont-gothic fonts-ipafont-mincho
  fonts-wqy-microhei fonts-wqy-zenhei fonts-indic
Recommended packages:
  dbus krb5-locales
The following NEW packages will be installed:
  ca-certificates ca-certificates-java fontconfig-config fonts-dejavu-core
  gosu java-common libavahi-client3 libavahi-common-data libavahi-common3
  libboost-filesystem1.58.0 libboost-program-options1.58.0
  libboost-system1.58.0 libboost-thread1.58.0 libcups2 libdbus-1-3 libexpat1
  libffi6 libfontconfig1 libfreetype6 libgmp10 libgnutls30
  libgoogle-perftools4 libgssapi-krb5-2 libhogweed4 libidn11 libjpeg-turbo8
[... cut ...]
Omada_SDN_Controller_v4.1.5_linux_x64/lib/checker-qual-2.5.2.jar
Omada_SDN_Controller_v4.1.5_linux_x64/lib/omada-compatible-4.1.5.jar
mkdir: created directory '/opt/tplink'
mkdir: created directory '/opt/tplink/EAPController'
**** Cleanup ****
**** Setup omada User Account ****
Removing intermediate container c3a49d128a79
 ---> 14b574f46d27
Step 4/10 : COPY entrypoint-4.1.x.sh /entrypoint.sh
COPY failed: stat /var/lib/docker/tmp/docker-builder536441664/entrypoint-4.1.x.sh: no such file or directory

joopy, 2020-07-16:
same output:

root@omada:~/mydockerbuild# docker build -t mbentley/omada-controller:4.1-armv7l -f Dockerfile.v4.1.x-armv7l .
Sending build context to Docker daemon  4.608kB
Step 1/10 : FROM ubuntu:16.04
16.04: Pulling from library/ubuntu
be0d01e22a1f: Pull complete
0b760e2e684f: Pull complete
15bcafb819cb: Pull complete
aaa652f3eed9: Pull complete
Digest: sha256:69bc24edd22c270431d1a9e6dbf57cfc4a77b2da199462d0251b145fdd7fa538
Status: Downloaded newer image for ubuntu:16.04
--
--
--
Omada_SDN_Controller_v4.1.5_linux_x64/lib/checker-qual-2.5.2.jar
Omada_SDN_Controller_v4.1.5_linux_x64/lib/omada-compatible-4.1.5.jar
mkdir: created directory '/opt/tplink'
mkdir: created directory '/opt/tplink/EAPController'
**** Cleanup ****
**** Setup omada User Account ****
Removing intermediate container eeb8cd1cda53
 ---> 783ca410bf60
Step 4/10 : COPY entrypoint-4.1.x.sh /entrypoint.sh
COPY failed: stat /var/lib/docker/tmp/docker-builder820045872/entrypoint-4.1.x.sh: no such file or directory

mbentley, 2020-07-16:
Did you clone the repo or did you just download the Dockerfile?
In your working directory, `~/mydockerbuild`, what all do you have?  You will need both the [Dockerfile.v4.1.x-armv7l](https://github.com/mbentley/docker-omada-controller/blob/master/Dockerfile.v4.1.x-armv7l) and [entrypoint-4.1.x.sh](https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-4.1.x.sh) files in order to build the Docker image.

joopy, 2020-08-06:
The image build went succesfull but i'm a real newby with docker. I can't get it to work with an container.

joopy, 2020-08-06:
Never mind. It's working now! thnx a lot.

mbentley, 2020-08-06:
Fantastic, thank you for verifying!

closed 2020-08-06 by mbentley