mbentley/docker-omada-controller #55: Unable to start docker image - port error - 8443 already in use

skawikk, 2020-07-17:
Hi All;)

I have little problem, i'm trying too run Omada on my server, but port 8443 is already in use, and Omada image won't start.
But in Docker file exposed port is 8043.
What part of controller is using port 8443?

M.

mbentley, 2020-07-17:
What's your run command for the container?

skawikk, 2020-07-17:
I'm running that in Rancher orchestrator.
I'm using host network mode, and because of that i have overlap between Omada and another container.
In Dockerfile and endtrypoint.sh i didn't find anything related to port 8443, and because of that i wrote my question.
M.

mbentley, 2020-07-17:
When running in host mode, it will require that the following ports be available on the host:
8088
8043
27001/udp
27002
29810/udp
29811
29812
29813

skawikk, 2020-07-17:
I've checked that already in Dockerfile;)

Please, check logs:
~> `docker logs a3d
INFO: Time zone set to 'Europe/Warsaw'
INFO: Starting Omada Controller as user omada
2020-07-17 15:53:46 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2020-07-17 15:53:47 [main] [INFO]-[SourceFile:75] - going to start local mongod.
2020-07-17T15:53:47.270+0200 I CONTROL  [main] ***** SERVER RESTARTED *****
2020-07-17T15:53:47.276+0200 I CONTROL  [initandlisten] MongoDB starting : pid=38 port=27217 dbpath=../data/db 64-bit host=origin
2020-07-17T15:53:47.276+0200 I CONTROL  [initandlisten] db version v3.6.3
2020-07-17T15:53:47.276+0200 I CONTROL  [initandlisten] git version: 9586e557d54ef70f9ca4b43c26892cd55257e1a5
2020-07-17T15:53:47.276+0200 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1  11 Sep 2018
2020-07-17T15:53:47.276+0200 I CONTROL  [initandlisten] allocator: tcmalloc
2020-07-17T15:53:47.276+0200 I CONTROL  [initandlisten] modules: none
2020-07-17T15:53:47.276+0200 I CONTROL  [initandlisten] build environment:
2020-07-17T15:53:47.276+0200 I CONTROL  [initandlisten]     distarch: x86_64
2020-07-17T15:53:47.276+0200 I CONTROL  [initandlisten]     target_arch: x86_64
2020-07-17T15:53:47.276+0200 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db", journal: { enabled: true } }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2020-07-17T15:53:47.276+0200 I -        [initandlisten] Detected data files in ../data/db created by the 'wiredTiger' storage engine, so setting the active storage engine to 'wiredTiger'.
2020-07-17T15:53:47.276+0200 I STORAGE  [initandlisten] wiredtiger_open config: create,cache_size=4427M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
2020-07-17T15:53:47.935+0200 I STORAGE  [initandlisten] WiredTiger message [1594994027:935932][38:0x7fe70826d0c0], txn-recover: Main recovery loop: starting at 16/5632
2020-07-17T15:53:48.000+0200 I STORAGE  [initandlisten] WiredTiger message [1594994028:726][38:0x7fe70826d0c0], txn-recover: Recovering log 16 through 17
2020-07-17T15:53:48.050+0200 I STORAGE  [initandlisten] WiredTiger message [1594994028:50683][38:0x7fe70826d0c0], txn-recover: Recovering log 17 through 17
2020-07-17T15:53:48.207+0200 I CONTROL  [initandlisten] 
2020-07-17T15:53:48.207+0200 I CONTROL  [initandlisten] ** WARNING: Access control is not enabled for the database.
2020-07-17T15:53:48.207+0200 I CONTROL  [initandlisten] **          Read and write access to data and configuration is unrestricted.
2020-07-17T15:53:48.207+0200 I CONTROL  [initandlisten] 
2020-07-17T15:53:48.394+0200 I FTDC     [initandlisten] Initializing full-time diagnostic data capture with directory '../data/db/diagnostic.data'
2020-07-17T15:53:48.394+0200 I NETWORK  [initandlisten] waiting for connections on port 27217
2020-07-17 15:53:48 [main] [INFO]-[SourceFile:158] - mongodb process id is 38
2020-07-17 15:53:48 [main] [INFO]-[SourceFile:139] - Mongo DB server started
2020-07-17T15:53:48.806+0200 I NETWORK  [listener] connection accepted from 127.0.0.1:46822 #1 (1 connection now open)
2020-07-17T15:53:48.812+0200 I NETWORK  [conn1] received client metadata from 127.0.0.1:46822 conn: { driver: { name: "mongo-java-driver", version: "3.6.4" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.5.13-arch2-1" }, platform: "Java/Private Build/1.8.0_252-8u252-b09-1~18.04-b09" }
2020-07-17T15:53:48.835+0200 I NETWORK  [listener] connection accepted from 127.0.0.1:46824 #2 (2 connections now open)
2020-07-17T15:53:48.836+0200 I NETWORK  [conn2] received client metadata from 127.0.0.1:46824 conn: { driver: { name: "mongo-java-driver", version: "3.6.4" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.5.13-arch2-1" }, platform: "Java/Private Build/1.8.0_252-8u252-b09-1~18.04-b09" }
2020-07-17T15:53:48.853+0200 I NETWORK  [conn2] end connection 127.0.0.1:46824 (1 connection now open)
2020-07-17T15:53:48.854+0200 I NETWORK  [conn1] end connection 127.0.0.1:46822 (0 connections now open)
2020-07-17 15:53:48 [main] [ERROR]-[SourceFile:103] - Port 8843 is already in use.
2020-07-17 15:53:48 [main] [ERROR]-[SourceFile:128] - Port 8843 is already in use. Release the port and try again.
2020-07-17 15:53:48 [main] [ERROR]-[SourceFile:249] - Environment is not ok for controller running
Bootstrap prepare failed, going to exit.
2020-07-17T15:53:48.864+0200 I CONTROL  [signalProcessingThread] got signal 15 (Terminated), will terminate after current cmd ends
2020-07-17T15:53:48.864+0200 I NETWORK  [signalProcessingThread] shutdown: going to close listening sockets...
2020-07-17T15:53:48.864+0200 I NETWORK  [signalProcessingThread] removing socket file: /tmp/mongodb-27217.sock
2020-07-17T15:53:48.864+0200 I FTDC     [signalProcessingThread] Shutting down full-time diagnostic data capture
2020-07-17T15:53:48.864+0200 I STORAGE  [signalProcessingThread] WiredTigerKVEngine shutting down
2020-07-17T15:53:48.957+0200 I STORAGE  [signalProcessingThread] shutdown: removing fs lock...
2020-07-17T15:53:48.957+0200 I CONTROL  [signalProcessingThread] now exiting
2020-07-17T15:53:48.957+0200 I CONTROL  [signalProcessingThread] shutting down with code:0
`

mbentley, 2020-07-17:
Huh, interesting.  It is using 8843 on 4.1.  Must be some new requirement as there isn't a ton of documentation on 4.1 yet.

skawikk, 2020-07-17:
I had problems with connection to TTY in container, it won't work at all. I couldn't check that whap app is a problem in container;/

mbentley, 2020-07-17:
Ah ok, so I found some settings in the properties file:
```
#web config
manage.http.port=8088
manage.https.port=8043
portal.http.port=8088
portal.https.port=8843
```

`manage` should be the management console, `portal` is probably for portal authentication for guest networks.  I imagine I should be able to make these settings configurable.

skawikk, 2020-07-17:
I saw that, but still there are no 8443;)

mbentley, 2020-07-17:
Damn it, I read `8843` as `8443`. 🤦

skawikk, 2020-07-17:
I'm blind;) i read logs again... 8843, 8843;d my bad;)

mbentley, 2020-07-17:
Whew, OK, glad I wasn't going crazy :)  I'll see if I can add an update to the entrypoint to be able to update those values on demand via environment variables.

mbentley, 2020-07-18:
OK, new feature is in.  You will not be able to modify the ports if the MongoDB has already been bootstrapped.  The persistent data will need to be reset and then you can use the environment variables [described in the README](https://github.com/mbentley/docker-omada-controller#optional-variables).

commit 49e5fff7 (2020-07-18): Merge pull request #56 from mbentley/change-ports-entrypoint

Added ability to change ports on first run; fixes #55. allow user to set which logs to show on STDOUT
commit 579c48cc (2020-07-18): Added ability to change ports on first run; fixes #55. allow user to set which logs to show on STDOUT
closed 2020-07-18 by mbentley