mbentley/docker-omada-controller #57: Fails start, re-start loop

dougle03, 2020-07-19:
Updated container on Synology NAS.
Followed additional instruction; cleared out existing folders (logs, work, data)
Restarted container, get the following in the log:

INFO: Time zone set to 'Europe/London'
INFO: Starting Omada Controller as user omada
error: exec: "/opt/tplink/EAPController/jre/bin/java": stat /opt/tplink/EAPController/jre/bin/java: no such file or directory
INFO: Time zone set to 'Europe/London'

And repeat!

I'm no expert so using the Docker GUI on DSM
Thanks,
[image]

mbentley, 2020-07-19:
How are you running the container?  Are you specifying your own command instead of relying on the default?  I'm assuming you're using the 4.1 image.  The JRE is no longer included in the installed bundle so it is installed in the system path.

dougle03, 2020-07-19:
Hi, I'm an idiot on this!
Is below what you are refering to?
I'm using the DSM Docker GUI provided by synology ;(
[image]

I've used the Docker registry within the Syno Docker GUI to download the 'Latest' image, so assume the latest tag is 4.1....?

dougle03, 2020-07-19:
I don't think I have any control over the 'entry point' as above, it's pre-determined with the docker image I presume?

mbentley, 2020-07-19:
Yes, latest is tagged to 4.1.  I'd suggest picking a version tag like 4.1 or 3.2 so you're locking yourself in on a version as I change out latest without notice. Especially in the case of 4.1, the breaking changes might be unexpected.

I'm not familiar with how Synology allows you to run containers but the 4.1 image has the correct entrypoint and command as defined in the Dockerfile.  I'll double check to make sure there isn't something funky going on with the auto builds on Hub though.

dougle03, 2020-07-19:
Thanks. I was on 3.2.10, so might go back to that as I kept a copy of the v data folders

mbentley, 2020-07-19:
I verified that the `latest` tag has the correct CMD and ENTRYPOINT:

```
# docker inspect docker.io/mbentley/omada-controller:latest --format '{{json .ContainerConfig.Entrypoint}}' | jq .
[
  "/entrypoint.sh"
]

# docker inspect docker.io/mbentley/omada-controller:latest --format '{{json .ContainerConfig.Cmd}}' | jq .
[
  "/bin/sh",
  "-c",
  "#(nop) ",
  "CMD [\"/usr/bin/java\" \"-server\" \"-Xms128m\" \"-Xmx1024m\" \"-XX:MaxHeapFreeRatio=60\" \"-XX:MinHeapFreeRatio=30\" \"-XX:+HeapDumpOnOutOfMemoryError\" \"-cp\" \"/opt/tplink/EAPController/lib/*:\" \"com.tplink.omada.start.OmadaLinuxMain\"]"
]
```

I'm wondering if the Synology inspects the entrypoint and command from the image when it is originally deployed and essentially overrides the entrypoint and command on the container.  It would have saved what was originally specified on the 3.2.x image as that's what I would expect for the older version.  At least that would explain the behavior.

dougle03, 2020-07-19:
I manually attempted to update the container following the syno docker instructions:
download 'latest' container
Stop the container
'clear' the container
Start the container.

After doing the above I checked the logs and saw the comment about 4.1 being non-standard update, so then deleted the V folders and recreated them empty.
This got rid of the upgrade warnings, but then the JRE warning appeared.

since the synology does retain settings like volume maps and ports ect, it's possible it also retains the entrypoint too....?

mbentley, 2020-07-19:
Sounds plausible.

dougle03, 2020-07-19:
Would make sense, pitty, good job I've not got too many AP's to sort out! 
It's late now, I'll have another go in the morning when rested.
Might be worth adding a warning to Synology users about this? Especially the GUI syno docker users (Like me)...
Thanks Matt.
P

commit 0e45ffe4 (2020-07-20): Merge pull request #59 from mbentley/synology-note

Added note about upgrading on a Synology NAS; fixes #57
commit 486c9e11 (2020-07-20): Added note about upgrading on a Synology NAS; fixes #57
closed 2020-07-20 by mbentley