mbentley/docker-omada-controller #60: SSL certificate names

ulug79, 2020-07-21:
Please add environment variables for customizing certificate names

mbentley, 2020-07-21:
Just curious, what's the use case where the certificate names are an issue?

ulug79, 2020-07-21:
I am using traefik 2.2 with letsencrypt certificate wildcard generation. A special container extracts the certificate in json format and puts them in a folder named cert.pem, fullchain.pem and key.pem

ulug79, 2020-07-21:
at home i am using Unifi Controller container from https://hub.docker.com/r/jacobalberty/unifi

and in it you can set non-standard names through environment variables

mbentley, 2020-07-21:
Are you expecting to just be able to update the certificate name or the whole path?

ulug79, 2020-07-21:
For example in Unifi Controller i mount certficate folder as volume in container, In enviroment variable I just indicate the different name. Please create 2 environent variable for full chain name and key name. Thanks.

For example:
If your certificate or private key have different names, you can set the environment variables CERTNAME and CERT_PRIVATE_NAME to the name of your certificate/private key, e.g. CERTNAME=my-cert.pem and CERT_PRIVATE_NAME=my-privkey.pem.

mbentley, 2020-07-21:
👍  Sounds good.  #61 should resolve this and add the ability to have custom names.

commit 8f5d303b (2020-07-21): Merge pull request #61 from mbentley/custom-cert-names

Added custom cert names; fixes #60
commit a0cc211b (2020-07-21): Added custom cert names; fixes #60
closed 2020-07-21 by mbentley