mbentley/docker-omada-controller #63: Error Code -1 (General Error) on  External Portal Server authorize API (Omada Controller v.4.1.5)

kevinp1005, 2020-09-19:
Omada Controller version : **4.1.5**
Device Model: **EAP225(US) v3.0** 
Firmware Version: **2.20.1 Build 20200630 Rel. 41064**

Hi, I'm trying to implement external portal server based on this [FAQ](https://www.tp-link.com/us/support/faq/2907/),
and is currently stuck at Step 8.

I was able to login to the Omada Controller (Step 7) and received the CSRF Token.

But received an error when trying to authorize the client (Step 8):

URL:
https://192.168.1.64:8043/api/v2/hotspot/extPortal/auth?token=05a5c488581148d7a17e0214f68ba0dc

Body:
```
{
    "clientMac" : "3C-15-C2-DE-D2-88",
    "ssidName" : "RUA",
    "apMac" : "50-D4-F7-5B-87-78",
    "radioId" : "1",
    "t" : "1600588644",
    "site" : "Default",
    "time" : "21600"
}
```
Note: Headers and Cookies have been set correctly as well

Response:
```
{
    "errorCode": -1,
    "msg": "General error."
}
```

server.log
```
2020-09-19 15:58:01 [qtp1976804832-1868] [ERROR]-[SourceFile:98] - java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException: null
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:1.8.0_265]
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:1.8.0_265]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:1.8.0_265]
	at java.lang.reflect.Method.invoke(Method.java:498) ~[?:1.8.0_265]
	at com.tplink.omada.apigateway.dispatch.chain.a.a(SourceFile:79) [omada-api-gateway-4.1.5.jar:?]
	at com.tplink.omada.apigateway.dispatch.RestDispatcher.a(SourceFile:74) [omada-api-gateway-4.1.5.jar:?]
	at com.tplink.omada.web.controller.ApiController.a(SourceFile:57) [omada-web-4.1.5.jar:?]
	at sun.reflect.GeneratedMethodAccessor606.invoke(Unknown Source) ~[?:?]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:1.8.0_265]
	at java.lang.reflect.Method.invoke(Method.java:498) ~[?:1.8.0_265]
	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:189) [spring-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:138) [spring-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:102) [spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:892) [spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:797) [spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) [spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1038) [spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:942) [spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1005) [spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.FrameworkServlet.doPost(FrameworkServlet.java:908) [spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:707) [javax.servlet-api-3.1.0.jar:3.1.0]
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:882) [spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790) [javax.servlet-api-3.1.0.jar:3.1.0]
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:873) [jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1623) [jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
	at com.tplink.omada.web.filter.CacheControlFilter.doFilter(SourceFile:51) [omada-web-4.1.5.jar:?]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1610) [jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.apache.shiro.web.servlet.ProxiedFilterChain.doFilter(ProxiedFilterChain.java:61) [shiro-web-1.4.0.jar:1.4.0]
	at org.apache.shiro.web.servlet.AdviceFilter.executeChain(AdviceFilter.java:108) [shiro-web-1.4.0.jar:1.4.0]
	at org.apache.shiro.web.servlet.AdviceFilter.doFilterInternal(AdviceFilter.java:137) [shiro-web-1.4.0.jar:1.4.0]
[... 38 lines cut ...]
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:267) [jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:305) [jetty-io-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103) [jetty-io-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.io.ssl.SslConnection$DecryptedEndPoint.onFillable(SslConnection.java:427) [jetty-io-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.io.ssl.SslConnection.onFillable(SslConnection.java:321) [jetty-io-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.io.ssl.SslConnection$2.succeeded(SslConnection.java:159) [jetty-io-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103) [jetty-io-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117) [jetty-io-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:333) [jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:310) [jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:168) [jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:126) [jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366) [jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:765) [jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:683) [jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_265]
Caused by: java.lang.NullPointerException
	at com.tplink.omada.service.auth.impl.AuthServiceImpl.a(SourceFile:661) ~[omada-service-4.1.5.jar:?]
	at com.tplink.omada.service.auth.impl.AuthServiceImpl.a(SourceFile:150) ~[omada-service-4.1.5.jar:?]
	at com.tplink.omada.service.LockWrapperService.j(SourceFile:237) ~[omada-service-4.1.5.jar:?]
	at com.tplink.omada.service.LockWrapperService.n(SourceFile:119) ~[omada-service-4.1.5.jar:?]
	at com.tplink.omada.service.auth.impl.AuthServiceImpl.a(SourceFile:176) ~[omada-service-4.1.5.jar:?]
	at com.tplink.omada.service.api.impl.ClientApiServiceImpl.a(SourceFile:410) ~[omada-service-4.1.5.jar:?]
	... 84 more
```

mbentley, 2020-09-19:
I can't say I've tried to set that up. I would suggest posting to https://community.tp-link.com/en/business/forum/582 as they should have visibility into the code as well.

closed 2020-09-19 by mbentley

arrie1992, 2023-03-02:
@kevinp1005 Have you ever figured this out? Also getting that -1 General error. Token and cookies are set! Using controller version 5.0.30! Getting that error on after every api call.. When i want to fetch a list of all the sites, im getting this error too