mbentley/docker-omada-controller #64: AutoBackup failed because autobackup schedule time was expired when TZ is set to Amsterdam/Europe

Roos-AID, 2020-10-01:
First a compliment, thanks for your great Omada package !

Running docker image on Synology Docker. Works great, but since upgrading from version 3 to 4.1.5 the Autobackup fails with message AutoBackup failed because autobackup schedule time was expired. 
The TZ environment variable is set to Europe/Amsterdam.  The log shows : INFO: Time zone set to 'Europe/Amsterdam' . The Omada user interface timezone settings show Time Zone: (UTC+01:00) Amsterdam, Berlin, Bern, Rome, Stockholm, Vienna
The autobackup schedule is set to 05:00 CET. 
However on the scheduled autobackup time we get error as mentioned. 
In the log file
# Time: 2020-10-01 05:00:08
# Controller version: 4.1.5
2020-10-01 05:00:06	System  	Error       	Auto Backup failed to execute with generating file AutoBackup failed because autobackup schedule time was expired.  

When I set timezone to UTC is works correctly. It seems that the Docker time is used to compare the schedule with the actual time (Docker time is UTC and which cannot be changed).

Is there a solution to this ?

Thanks,
Rob

mbentley, 2020-10-01:
It's a bug in the software.  I filed the same over at the TP Link Community forums: https://community.tp-link.com/en/business/forum/topic/217542

mbentley, 2020-12-02:
FYI, I believe this is fixed in the 4.2.4 release.

Roos-AID, 2020-12-03:
Thanks, it seems it is ok now. But we have to wait until summertime as according to release notes, the bug was related to summertime.