mbentley/docker-omada-controller #65: Certificate import

xrs444, 2020-10-21:
Hello, 

Using the latest version of the container, I placed a certificate and key file into the certs folder in the omada-data volume, but the container does not seem to pick them up, whether or not I use the environment variables to specify the tls.key and tls.crt file names.

I don't see any relevant errors in the logs, and the certificate was generated by FreeIPA/Certmonger. If I package them up and import them via the web interface they work as desired.

Is this no longer possible in the latest version, or do I have a mistake somewhere?

Thanks,

Thomas

mbentley, 2020-10-22:
Do you have your certificates mounted inside the container under the `/cert` directory and are they named `tls.key` and `tls.crt`?  Did you ensure that the `tls.crt` file had the certs stacked in the correct order (see https://github.com/mbentley/docker-omada-controller#custom-certificates)?

I currently have mine working with the certs bind mounted like the above.

xrs444, 2020-10-23:
Certs are named tls.key and tls.crt, and I have the chain in there.

I'll check the order and see how it goes.

xrs444, 2020-10-28:
Looks like my certs are in the right order, container then CA.

I've not had a chance to dig into this further, and I don't see anything obvious in the log. I'll see if I can test the entrypoint script with the certs outside of the container and see if I see any errors.

xrs444, 2020-10-28:
I opened the console in the container and when I tried running the entrypoint script it was looking for the certs in /cert, which does not exist in that container.

To test I did an ln -s for /cert pointing to the cert folder in the omada-data folder mount and the entrypoint script reported it ran correctly when I tried it again. 

However when I restarted the container it now seems to work apart from the web interface, and does not appear in the running containers in portainer, but does contact the cloud service, even after stopping the whole service and restarting.

I'll dig into this more tomorrow.

xrs444, 2020-10-28:
I think i may have found the issue, I opened the terminal in the container and ran the entrypoint script, and it appears it is looking for /cert, but for my container this didn't exist, just the cert folder in the omada-data volume mount. 

I did a ln -s for /cert to the mounted location and the entrypoint script ran successfully.

I then mapped the cert directory in the omada-data folder to /cert in the container, pushed the updated service, and this seems to be working fine.

mbentley, 2020-10-28:
Ah yes.  It has to be the explicit `/cert` directory, not a directory under the `omada-data` volume.  See https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-4.1.x.sh#L95-L115

closed 2020-10-28 by mbentley