mbentley/docker-omada-controller #69: New (major) upstream release 4.2.4

jobhh, 2020-11-16:
Linux version of 2.4.2 has been released :) 

https://static.tp-link.com/2020/202011/20201110/Omada_SDN_Controller_v4.2.4_linux_x64.tar.gz

https://www.tp-link.com/support/download/omada-software-controller/#Controller_Software

mbentley, 2020-11-16:
Thanks! @mc1271 submitted a PR to work on this https://github.com/mbentley/docker-omada-controller/pull/68

jobhh, 2020-11-16:
Ah cool, I didn't see the PR :)

I contacted TP-Link support a couple weeks back regarding an issue and they informed me it was fixed in 4.2.4, the Windows version was available and Linux would follow "soon". Last week I asked when "soon" would be and they told me on Monday, so I figured I'd post it here.

I'll close this issue.

closed 2020-11-16 by jobhh

jobhh, 2020-11-17:
I've received another email from TP-Link support. Apparently both a tarball and .deb package will be provided for the release after 4.2.4.

> The Omada Controller V4.2.4 for Linux version has been released, please take your time to run the tests and let us know if this fixes the issue.
> Note: Linux controller v4.2.4 only release .tar.gz achieve, but the subsequent version will provide both .deb packet and .tar.gz achieve.

mbentley, 2020-11-17:
👍  Luckily we use the .tar.gz mostly.

mbentley, 2020-11-17:
There are some test images available: https://github.com/mbentley/docker-omada-controller/pull/68#issuecomment-729031065

jobhh, 2020-11-22:
The update worked :) I did got this response on the 4.2 version. The latest version worked fine.

> docker pull mbentley/omada-controller:4.2
> Error response from daemon: manifest for mbentley/omada-controller:4.2 not found: manifest unknown: manifest unknown

mbentley, 2020-11-22:
Oh whoops. Forgot to add the auto build