mbentley/docker-omada-controller #70: Failed to start omada controller

JoeNobody76, 2020-11-20:
Not sure if I'm doing something wrong here, but can't seem to get it up and running.  Installing on Unraid/Docker.

I've attached the docker log.

[Log.txt](https://github.com/mbentley/docker-omada-controller/files/5571919/Log.txt)

dvandonkelaar, 2020-11-20:
the log gives the following message: `java.io.IOException: No space left on device`
Is there any free storage space left?

JoeNobody76, 2020-11-21:
My current Docker vDisk size is 20Gb, should it be higher?  What size does it need?

mbentley, 2020-11-21:
That is going to depend on how much space you have available and will depend on where you're storing your persistent data.  This isn't what I would call a great example but it was just for testing but in this case, it is writing data out to the Docker root dir:

```
$ docker info --format '{{ .DockerRootDir}}'
/var/lib/docker
```

```
docker run -it --rm \
  --name omada-controller \
  -p 8088:8088 \
  -p 8043:8043 \
  -e SHOW_SERVER_LOGS=true \
  -e SHOW_MONGODB_LOGS=false \
  -e TZ=Etc/UTC \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:3.2
```

The short answer is that MongoDB will, by default, pre-allocate a lot of data.  You can use the `SMALL_FILES` option to have it use less.  See [SMALL_FILES](https://github.com/mbentley/docker-omada-controller#small-files)

JoeNobody76, 2020-11-22:
Space is not an issue, it runs off a 1Tb cache drive so I can allocate more space. So how much would you recommend I allocate to avoid the issue?

mbentley, 2020-11-22:
How are you starting the container?  There is either a filesystem somewhere that is in use where there is not enough space.  It could be where the Docker data root is or possibly where the persistent data is.  It is either going to be related to the  Docker data root (the path returned from `docker info --format '{{ .DockerRootDir}}'` or wherever you have the persistent data mounted but it's hard to tell for sure which.

I wouldn't expect it to be complaining about space unless there is a space issue or for some reason, it not being able to write to a file or directory that is expected (unexpected or odd filesystem type?)

Just as an example of the default amount of space required to just get the container up and running from a standpoint of the directories that need persistent data:

```
root@38e8abba5287:/opt/tplink/EAPController# du -hs data/ work/ logs/
209M    data/
4.5M    work/
40K     logs/
```

A quick `docker diff` shows that nothing else is being modified in the container:

```
$ docker diff omada-controller
C /tmp
A /tmp/mongodb-27217.sock
```

JoeNobody76, 2020-11-23:
Think I've figured it out.  After a couple of uninstalls and re-installs I've figured out changing the Host paths to /mnt/user/ it installed without any issues and I've successfully launched and adopted my AP's.  I've also successfully connected it with my Omada cloud account.  There are also no errors in my log file.

Thanks for your replies and advice, this program runs great!

mbentley, 2020-11-23:
Great, glad to hear!

closed 2020-11-23 by mbentley

dark4sorrow, 2022-07-29:
> Think I've figured it out. After a couple of uninstalls and re-installs I've figured out changing the Host paths to /mnt/user/ it installed without any issues and I've successfully launched and [...]

I was running into the same issue, and just changing the Host Paths to /mnt/user/ fixed my problems as well.
Thanks for that. :)