mbentley/docker-omada-controller #71: Omada container default user and PW

keelhaulin, 2020-11-20:
Thanks much for creating this container. I can access the login page but the documented TP-Link default user name and password (admin, admin) do not work. Are their different credentials in your image? I am using mbentley/omada-controller:3.2.1
[image]

mbentley, 2020-11-20:
There are no default credentials.  It should take you to the wizard on initial run.  If it doesn't, you have persistent data that exists.
[image]

I would make sure to stop the container, clear the persistent data and start the container again.

keelhaulin, 2020-11-21:
Thank you for the very quick response and pointing me in the right
direction. I found the install directories and removed them. I am using
Portainer and incorrectly assumed that deleting the container and image
would clear all persistent data. I have it working now after deploying a
fresh container.

Thanks much for the original effort and the timely support

mbentley, 2020-11-21:
Good to hear, happy to help!

closed 2020-11-21 by mbentley