mbentley/docker-omada-controller #73: Feature Suggestion: Support v4.2.4

bland328, 2020-11-21:
Thanks very much for this project!

I'm no Docker/Github whiz, but I [forked](https://github.com/bland328/docker-omada-controller) your project to semi-crudely tweak it to install Omada Software Controller v4.2.4 after running into an [SSID Override bug](https://community.tp-link.com/en/business/forum/topic/238082?page=1) in v4.1.5.

The newer version did fix the bug, and while there's nothing pull-request-worthy about the changes I made, I want to let you know of a pain point I ran into: I couldn't for the life of me get access to the v4.2.4 interface until I ultimately substituted the v4.1.5 `eap.cer` and `eap.keystore` files for those that shipped with v4.2.4.

I don't honestly know if that was a smart or dumb move, and you won't find that addressed in my fork, as I found it easier to simply add an `omada-logs:/opt/tplink/EAPController/keystore` bind mount and replace them post-deployment.

If you decide to support v4.2.4, I hope some part of this helps 👍

mbentley, 2020-11-21:
I just merged in changes for 4.2.4. Let me know if there are issues.

bland328, 2020-11-21:
That was quick! Thanks, and I'll take your version for a spin soon.

Does v4.2.4 work fine with you without having to mess with the certificate files?

mbentley, 2020-11-21:
Yes works fine out of the box if you don't want to provide your own certs.

bland328, 2020-11-21:
Just tried your version, and I agree...works just fine. I've no idea what went weird with my attempt, but I'm happy to be back to using your project. Thanks again!

closed 2020-11-21 by bland328