mbentley/docker-omada-controller #74: Certificate does not mount correctly

ruslanguns, 2020-11-30:
Before everything, thanks for supporting this repository, you are doing a great job.

The container is not reading the `cert/tls.key` and `cert/tls.crt` it is correctly setup and is right in the / path from the container. But my guess is that it isn't configured correctly to know that this path has the certificates.

In this image you will see my configuration, it is pretty much the same as your command, I use docker-compose for confort.

[image]
 
And in this another image you will see that I did `docker exec` to get inside the container, I have printed the tls.key, as you can see it is there. 
[image]
 
But at the end of the day, omada is just getting the self signed certificate and is ignoring the mine.
[image]

Please let me know if you can give me some clues for fixing it, I am having issues with Apple hotspot.

mbentley, 2020-11-30:
What do the container logs show?  At the beginning, it should try to add the cert to the JKS.

mbentley, 2020-12-01:
I removed the image that had the private key in it.  Could you also provide the output of the following command?:

```
openssl x509 -in tls.crt -text -noout
```

ruslanguns, 2020-12-01:
So sorry for the delay, I am here now! 

## This is all the outputs when is created: 
```bash
Creating omada-controller ... done
Attaching to omada-controller
omada-controller | INFO: Time zone set to 'Etc/UTC'
omada-controller | INFO: Starting Omada Controller as user omada
omada-controller | 2020-12-01 18:45:26 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
omada-controller | 2020-12-01 18:45:27 [main] [INFO]-[SourceFile:89] - going to start local mongod.
omada-controller | 2020-12-01 18:45:29 [main] [INFO]-[SourceFile:167] - mongodb process id is 29
omada-controller | 2020-12-01 18:45:29 [main] [INFO]-[SourceFile:148] - Mongo DB server started
omada-controller | 2020-12-01 18:45:31 [main] [WARN]-[AbstractConnector.java:205] - Acceptors should be <= availableProcessors: ServerConnector@55de24cc{HTTP/1.1,[http/1.1]}{0.0.0.0:0}
omada-controller | 2020-12-01 18:45:31 [main] [WARN]-[AbstractConnector.java:205] - Acceptors should be <= availableProcessors: ServerConnector@7dfb0c0f{HTTP/1.1,[http/1.1]}{0.0.0.0:0}
omada-controller | 2020-12-01 18:46:02 [main] [INFO]-[SourceFile:35] - monitor context initialing...
omada-controller | 2020-12-01 18:46:06 [main] [INFO]-[SourceFile:290] - restart all customize tasks.
omada-controller | 2020-12-01 18:46:09 [main] [INFO]-[SourceFile:126] - no need to compatible db.
omada-controller | 2020-12-01 18:46:09 [main] [INFO]-[SourceFile:20] - Omada Controller started
omada-controller | 2020-12-01 18:46:10 [device-discovery-server] [INFO]-[SourceFile:54] - Start device discovery server
omada-controller | 2020-12-01 18:46:10 [device-manage-server] [INFO]-[SourceFile:111] - Start device manage server
omada-controller | 2020-12-01 18:46:10 [device-adopt-server] [INFO]-[SourceFile:82] - Start device adopt server
omada-controller | 2020-12-01 18:46:12 [Thread-16] [INFO]-[SourceFile:100] - CloudClient is started.
omada-controller | 2020-12-01 18:46:12 [Thread-16] [INFO]-[SourceFile:110] - Connect SEF server automatically.
omada-controller | 2020-12-01 18:46:12 [comm-pool-3] [INFO]-[SourceFile:84] - Starting PF module.
omada-controller | 2020-12-01 18:46:12 [comm-pool-3] [INFO]-[SourceFile:146] - Starting SCMP Server.
omada-controller | 2020-12-01 18:46:12 [comm-pool-3] [INFO]-[SourceFile:120] - Cloud service is started.
omada-controller | 2020-12-01 18:46:13 [Thread-16] [INFO]-[SourceFile:194] - The result of connection is true, service host is n-devs-smb.tplinkcloud.com, port is 443.
omada-controller | 2020-12-01 18:46:13 [Thread-16] [INFO]-[SourceFile:114] - Close connection to SEF server.
omada-controller | 2020-12-01 18:46:13 [Thread-19] [INFO]-[SourceFile:403] - javax.net.ssl.SSLException: Socket closed
omada-controller | javax.net.ssl.SSLException: Socket closed
omada-controller |      at sun.security.ssl.Alert.createSSLException(Alert.java:127) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.TransportContext.fatal(TransportContext.java:324) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.TransportContext.fatal(TransportContext.java:267) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.TransportContext.fatal(TransportContext.java:262) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl.handleException(SSLSocketImpl.java:1303) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl.access$300(SSLSocketImpl.java:72) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:831) ~[?:1.8.0_275]
omada-controller |      at com.tplink.eap.cloudsdk.client.p.a(SourceFile:120) ~[cloudsdk-1.0.5.jar:?]
omada-controller |      at com.tplink.eap.cloudsdk.client.o.run(SourceFile:397) [cloudsdk-1.0.5.jar:?]
omada-controller |      at java.lang.Thread.run(Thread.java:748) [?:1.8.0_275]
omada-controller | Caused by: java.net.SocketException: Socket closed
omada-controller |      at java.net.SocketInputStream.socketRead0(Native Method) ~[?:1.8.0_275]
omada-controller |      at java.net.SocketInputStream.socketRead(SocketInputStream.java:116) ~[?:1.8.0_275]
omada-controller |      at java.net.SocketInputStream.read(SocketInputStream.java:171) ~[?:1.8.0_275]
omada-controller |      at java.net.SocketInputStream.read(SocketInputStream.java:141) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketInputRecord.read(SSLSocketInputRecord.java:457) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketInputRecord.bytesInCompletePacket(SSLSocketInputRecord.java:68) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl.readApplicationRecord(SSLSocketImpl.java:1095) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl.access$200(SSLSocketImpl.java:72) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:815) ~[?:1.8.0_275]
omada-controller |      ... 3 more
omada-controller | 2020-12-01 18:46:13 [Thread-19] [INFO]-[SourceFile:407] - ssl recv thread is stopped.
omada-controller | 2020-12-01 18:46:13 [Thread-16] [INFO]-[SourceFile:127] - Connect service server automatically, ConnectionType is PERSISTENT_CONNECTION.
omada-controller | 2020-12-01 18:46:14 [Thread-16] [INFO]-[SourceFile:366] - The result of connection is true.
omada-controller | 2020-12-01 18:46:25 [device-event-bus-0] [WARN]-[SourceFile:88] - Realtimeap is null, ap mac 74-DA-88-F9-D8-CF
[... cut ...]
         fe:20:aa:6f:a7:40:b2:fe:ae:a2:bf:e9:97:7b:fa:19:bc:fa:
         86:05:0d:e2:bc:fc:55:65:30:ac:91:a7:b8:dd:30:21:6e:81:
         2f:1f:cc:d8:85:99:f1:56:16:49:49:32:cd:f5:5e:2c:14:b6:
         6a:37:f9:99:08:b0:cd:75:6a:b5:64:ad:f3:3f:47:11:b3:ec:
         c1:8e:bf:f8:3b:31:f8:d6:bf:c2:31:b5:d6:1d:db:38:07:8d:
         d8:c5:9d:13:e2:51:a1:93:0e:27:02:46:ae:5c:03:81:1d:57:
         30:bc:ed:d7:10:c7:68:0c:6c:5b:8f:2b:99:eb:65:db:a3:b7:
         8e:74:46:3e:25:9c:e3:20:69:8a:48:5f:10:e8:37:ba:b3:f4:
         0a:da:53:3e
```

mbentley, 2020-12-01:
No worries - looks like you might have an older image as I don't see any messages about importing the cert.  Try to re-pull the image and try it again.

ruslanguns, 2020-12-01:
Thanks for quick reply, I have modify the docker-compose file in order to use the latest image. Here is the logs 

```bash
root@omada:~/omada# docker-compose up -d && docker-compose logs -f
Pulling omada (mbentley/omada-controller:latest)...
latest: Pulling from mbentley/omada-controller
f22ccc0b8772: Pull complete
3cf8fb62ba5f: Pull complete
e80c964ece6a: Pull complete
60739ce57848: Pull complete
b7f256e7ce76: Pull complete
Digest: sha256:6a38816b1b29ffaafb4bff348f7006b14c2fdbfb3418cca1c04f63e99ec2ed46
Status: Downloaded newer image for mbentley/omada-controller:latest
Creating omada-controller ... done
Attaching to omada-controller
omada-controller | INFO: Time zone set to 'Etc/UTC'
omada-controller | INFO: Starting Omada Controller as user omada
omada-controller | 2020-12-01 19:09:56 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
omada-controller | 2020-12-01 19:09:56 [main] [INFO]-[SourceFile:89] - going to start local mongod.
omada-controller | 2020-12-01 19:09:58 [main] [INFO]-[SourceFile:167] - mongodb process id is 29
omada-controller | 2020-12-01 19:09:58 [main] [INFO]-[SourceFile:148] - Mongo DB server started
omada-controller | 2020-12-01 19:09:59 [main] [WARN]-[AbstractConnector.java:205] - Acceptors should be <= availableProcessors: ServerConnector@30f842ca{HTTP/1.1,[http/1.1]}{0.0.0.0:0}
omada-controller | 2020-12-01 19:09:59 [main] [WARN]-[AbstractConnector.java:205] - Acceptors should be <= availableProcessors: ServerConnector@169bb4dd{HTTP/1.1,[http/1.1]}{0.0.0.0:0}
omada-controller | 2020-12-01 19:10:29 [main] [INFO]-[SourceFile:35] - monitor context initialing...
omada-controller | 2020-12-01 19:10:34 [main] [INFO]-[SourceFile:290] - restart all customize tasks.
omada-controller | 2020-12-01 19:10:37 [main] [INFO]-[SourceFile:126] - no need to compatible db.
omada-controller | 2020-12-01 19:10:37 [main] [INFO]-[SourceFile:20] - Omada Controller started
omada-controller | 2020-12-01 19:10:37 [device-discovery-server] [INFO]-[SourceFile:54] - Start device discovery server
omada-controller | 2020-12-01 19:10:37 [device-manage-server] [INFO]-[SourceFile:111] - Start device manage server
omada-controller | 2020-12-01 19:10:37 [device-adopt-server] [INFO]-[SourceFile:82] - Start device adopt server
omada-controller | 2020-12-01 19:10:40 [Thread-11] [INFO]-[SourceFile:100] - CloudClient is started.
omada-controller | 2020-12-01 19:10:40 [Thread-11] [INFO]-[SourceFile:110] - Connect SEF server automatically.
omada-controller | 2020-12-01 19:10:40 [comm-pool-3] [INFO]-[SourceFile:84] - Starting PF module.
omada-controller | 2020-12-01 19:10:40 [comm-pool-3] [INFO]-[SourceFile:146] - Starting SCMP Server.
omada-controller | 2020-12-01 19:10:40 [comm-pool-3] [INFO]-[SourceFile:120] - Cloud service is started.
omada-controller | 2020-12-01 19:10:41 [Thread-11] [INFO]-[SourceFile:194] - The result of connection is true, service host is n-devs-smb.tplinkcloud.com, port is 443.
omada-controller | 2020-12-01 19:10:41 [Thread-11] [INFO]-[SourceFile:114] - Close connection to SEF server.
omada-controller | 2020-12-01 19:10:41 [Thread-14] [INFO]-[SourceFile:403] - javax.net.ssl.SSLException: Socket closed
omada-controller | javax.net.ssl.SSLException: Socket closed
omada-controller |      at sun.security.ssl.Alert.createSSLException(Alert.java:127) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.TransportContext.fatal(TransportContext.java:324) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.TransportContext.fatal(TransportContext.java:267) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.TransportContext.fatal(TransportContext.java:262) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl.handleException(SSLSocketImpl.java:1303) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl.access$300(SSLSocketImpl.java:72) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:831) ~[?:1.8.0_275]
omada-controller |      at com.tplink.eap.cloudsdk.client.p.a(SourceFile:120) ~[cloudsdk-1.0.5.jar:?]
omada-controller |      at com.tplink.eap.cloudsdk.client.o.run(SourceFile:397) [cloudsdk-1.0.5.jar:?]
omada-controller |      at java.lang.Thread.run(Thread.java:748) [?:1.8.0_275]
omada-controller | Caused by: java.net.SocketException: Socket closed
omada-controller |      at java.net.SocketInputStream.socketRead0(Native Method) ~[?:1.8.0_275]
omada-controller |      at java.net.SocketInputStream.socketRead(SocketInputStream.java:116) ~[?:1.8.0_275]
omada-controller |      at java.net.SocketInputStream.read(SocketInputStream.java:171) ~[?:1.8.0_275]
omada-controller |      at java.net.SocketInputStream.read(SocketInputStream.java:141) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketInputRecord.read(SSLSocketInputRecord.java:457) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketInputRecord.bytesInCompletePacket(SSLSocketInputRecord.java:68) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl.readApplicationRecord(SSLSocketImpl.java:1095) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl.access$200(SSLSocketImpl.java:72) ~[?:1.8.0_275]
omada-controller |      at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:815) ~[?:1.8.0_275]
omada-controller |      ... 3 more
[... cut ...]
omada-controller | 2020-12-01 19:10:53 [device-event-bus-0] [WARN]-[SourceFile:88] - Realtimeap is null, ap mac 74-DA-88-F9-D8-CF
omada-controller | 2020-12-01 19:10:54 [nioEventLoopGroup-9-1] [INFO]-[SourceFile:51] - success to send InitialSettingSync to 74-DA-88-F9-D8-CF
omada-controller | 2020-12-01 19:11:07 [scheduled-pool-8] [INFO]-[SourceFile:122] - Start checking last firmwares of managed devices.
omada-controller | 2020-12-01 19:11:07 [scheduled-pool-8] [INFO]-[SourceFile:132] - Checking last firmwares of managed devices succeed.

mbentley, 2020-12-01:
Ah ok.  Interesting.  It seems like it isn't finding the files or that if statement isn't working.  Could you try to swap in this image which I added some debugging to?: `mbentley/omada-controller:4.2-test`

Example output:

```
INFO: Time zone set to 'Etc/UTC'
===============
START DEBUGGING
===============
+ set +e
+ ls -l /cert
total 8
-rw-r--r-- 1 root root 3579 Oct 30 00:16 tls.crt
-rw-r--r-- 1 root root 1679 Jan 10  2020 tls.key
+ grep -E Subject:|Issuer:
+ openssl x509 -in /cert/tls.crt -text -noout
        Issuer: C = US, O = Let's Encrypt, CN = Let's Encrypt Authority X3
        Subject: CN = ucp.test.demo.dckr.org
+ [ -f /cert/tls.key ]
+ [ -f /cert/tls.crt ]
+ echo INFO: Importing Cert from /cert/tls.[key|crt]
INFO: Importing Cert from /cert/tls.[key|crt]
+ openssl pkcs12 -export -inkey /cert/tls.key -in /cert/tls.crt -certfile /cert/tls.crt -name eap -out /opt/tplink/EAPController/keystore/cert.p12 -passout pass:tplink
+ rm /opt/tplink/EAPController/keystore/eap.keystore
+ keytool -importkeystore -deststorepass tplink -destkeystore /opt/tplink/EAPController/keystore/eap.keystore -srckeystore /opt/tplink/EAPController/keystore/cert.p12 -srcstoretype
PKCS12 -srcstorepass tplink
Importing keystore /opt/tplink/EAPController/keystore/cert.p12 to /opt/tplink/EAPController/keystore/eap.keystore...
Entry for alias eap successfully imported.
Import command completed:  1 entries successfully imported, 0 entries failed or cancelled

Warning:
The JKS keystore uses a proprietary format. It is recommended to migrate to PKCS12 which is an industry standard format using "keytool -importkeystore -srckeystore /opt/tplink/EAPCo
ntroller/keystore/eap.keystore -destkeystore /opt/tplink/EAPController/keystore/eap.keystore -deststoretype pkcs12".
+ set +x
=============
END DEBUGGING
=============
...
```

ruslanguns, 2020-12-01:
Again, sorry for late, Github does not notify me when new reply is there. I will have to find some external plugin at chrome to do it. Really I apologies.

[image]

mbentley, 2020-12-01:
Ah, it's a compose issue.

Remove the quotes around these values:
[image]

ruslanguns, 2020-12-01:
Seems like it is working now.. but please see the output

[image]

mbentley, 2020-12-01:
There we go!  Yup, working now.

ruslanguns, 2020-12-01:
Thank you so much! YTB 😎

closed 2020-12-01 by ruslanguns