mbentley/docker-omada-controller #79: multiple Dockerfiles for a given version

aeschbacher, 2020-12-29:
The Dockerfile for `amd64` and `armv7l` differ slightly, that's why there are two different Dockerfiles (`Dockerfile.v4.2.x` and `Dockerfile.v4.2.x-armv7l`).

It is possible to have one unique Dockerfile, putting the differences out of it.

If you think that it is acceptable to rely on `make` and use:

```shell
make
make armv7l
make arm64
```
instead of:

```shell
docker build ...
docker build -f Dockerfile.v4.2.x-armv7 ...l
docker build -f Dockerfile.v4.2.x-arm64 ...
```

then I think I can provide a PR

mbentley, 2020-12-29:
My main limitation is that I am relying on Docker Hub to perform the amd64 builds via auto builds so whatever is done in the Dockerfile would need to be portable to work on Docker Hub.  I do build the armv7l and arm64 images on my own CI though.  I have over 100 repos on Docker Hub so I rely on auto builds instead of running my own CI for everything as I only want to build on my own what I can't do easily on Docker Hub.  This could be better handled by consolidating the Dockerfiles by using an if statement to set what package to install; something like this:

```
...
# install omada controller (instructions taken from install.sh); then create a user & group and set the appropriate file system permissions
RUN \
  echo "**** Install Dependencies ****" &&\
  if [ "$(arch)" = "armv7l" ]; then MONGO_PKG="mongodb"; else MONGO_PKG="mongodb-server-core";fi &&\
  apt-get update &&\
  apt-get install --no-install-recommends -y gosu ${MONGO_PKG} net-tools openjdk-8-jre-headless tzdata wget &&\
...
```

That would allow Docker Hub to build the amd64 images just fine and I could adjust my own CI with no problem.  I don't have a problem with a Makefile though either - I just don't manually build anything if I do not have to.

mbentley, 2020-12-29:
...actually, I take that back.  I missed that I have to use `ubuntu:16.04` for the `armv7l` image because there are no mongodb packages built for `ubuntu:18.04`

aeschbacher, 2020-12-29:
Got it. I think it can be handled using (Docker Hub) hooks. If you think it is worth it, I can try.

PR #80 (2021-01-09): Factorize multiple Dockerfile into one
- new ARG `ARCH` to select target architecture (amd64|arm64|arm7vl)
- new ARG `BASE` to overload the Dockerfile's FROM if needed
- new script `install-<ver>.sh` encapsulating the differences between
  target architectures
- consequently, keep one unique Dockerfile for 4.2.x
- new script `build.sh`, whose main goal is to document how `docker build`
  shall be called

Closes: #79
commit b3cc765d (2021-01-09): Factorize multiple Dockerfile into one

- new ARG ARCH to select target architecture (amd64|arm64|arm7vl)
- new ARG BASE to overload the Dockerfile's FROM if needed
- new script install-<ver>.sh encapsulating the differences between
  target architectures
- consequently, keep one unique Dockerfile for 4.2.x
- new script build.sh, whose main goal is to document how 'docker build'
  shall be called

Closes: #79
closed 2021-01-09 by mbentley