mbentley/docker-omada-controller #82: Controller Mail Server

Kirk-GH, 2021-01-20:
Hi,

First off.  Great work and thank you very much.

Secondly, is it possible for the mail server settings in the controller to send email notifications?  I have put in the same settings as I have in my nas which works just fine.

Any assistance would be grateful.

Thanks,
Kirk

mbentley, 2021-01-20:
Thanks!  Yes, email works as one would expect.  If there are any mail failures, you might want to check to see if there are any rules that would prevent email coming from a container that could show up against an ACL as an IP address that isn't allowed or anything like that related to the controller running in a container.

[image]
[image]

Kirk-GH, 2021-01-20:
Thanks for the prompt response.  I did fail to mention that I am running the container in my Synology nas.  Are there maybe some environment variables I would have to include?

mbentley, 2021-01-20:
I do not know how your mail server is configured to send mail but there shouldn't be anything special about Omada Controller running in a container in terms of how to setup mail.

Kirk-GH, 2021-01-20:
Hi.  Here are my settings for nas and for Omada.  There is no access control list that I know of.

Nas Settings
[image]

Omada Settings
[image]

Thanks again,
Kirk

mbentley, 2021-01-20:
Check the logs of the omada controller container.  It will output detail information about what is happening with the SMTP test.  Are you filling in a `Sender Address`?

Kirk-GH, 2021-01-20:
I'm new to Docker.  Totally forgot about the logs.  Thanks.  I have tried with the sender address entered and also not.

[image]

Kirk

mbentley, 2021-01-20:
Are you sure that your Synology settings are correct?  Normally SMTP over TLS/SSL is 465 or 587 although I suppose they could be listening over TLS on 25. I'm on my phone or I'd check by doing a test connection.

Kirk-GH, 2021-01-20:
Totally agree with you.  These values work for my nas settings.

[image]

Kirk-GH, 2021-01-20:
Ok, well, no worries.  I'll keep at it.  I don't mean to take up too much of your time.

I really appreciate you assisting.  I'll post back once I figure it out.

Kirk

mbentley, 2021-01-20:
Yeah, I would try using port `587` with SSL and see how that works out for you.  I did some more poking around and I was able to get the same error message when I ticked the SSL box when I was connecting to a non-TLS/SSL enabled port but that doesn't match the instructions on the screen shot from Bell but I suppose that wouldn't be the first time documentation wasn't completely correct and I also wouldn't put it past the TP-Link team to have implemented something funny in how SMTP works.  I don't have a TLS enabled mail server here locally otherwise I would test that as well.

Kirk-GH, 2021-01-20:
Ya, I tried all different ports already.  25, 465, 587, etc.  As I said, it works coming directly from my nas with the same/proper setup.  It really has to be something about not getting out of the docker.  I have no problem giving you my email credentials should you feel you want to do some testing.

Kirk-GH, 2021-01-20:
Well, as you said, something funny.  I unchecked SSL and it sent.  Probably should have tried that before but why would I if it works as expected with SSL from other applications.

I'm really sorry for wasting so much of your time.  Hopefully this can be helpful (try the unexpected) to anyone else searching.

I appreciated your prompt responses and again thank you for your work.

mbentley, 2021-01-20:
There are some other comments on some oddities of the SMTP implementation over on the [TP-Link forums](https://community.tp-link.com/en/business/forum/topic/217556) as well where they mention the issues with StartTLS not working properly.

closed 2021-01-20 by mbentley

kylaris, 2021-02-24:
I fixed this issue by opening up the Synology firewall on port 465 / docker IP