mbentley/docker-omada-controller #84: Issues with 4.2

Timeteo, 2021-02-13:
Can't get 4.2 to work.

First errors were about log4j including this one:
`2021-02-13 04:56:49,562 log4j-thread ERROR Unable to create file ../logs/server.log java.io.IOException: Could not create directory /opt/tplink/logs`
This is fixed by adding a second volume mapping:
`-v omada-logs:/opt/tplink/EAPController/logs \`
`-v omada-logs:/opt/tplink/logs \`

Next attempt mongodb would not start:
`2021-02-13 05:05:35 [main] [WARN]-[SourceFile:78] - com.tplink.omada.start.a.a: Fail to start mongo DB server com.tplink.omada.start.a.a: Fail to start mongo DB server`
This is fixed by adding a second volume mapping:
`-v omada-logs:/opt/tplink/EAPController/data\`
`-v omada-logs:/opt/tplink/data\`

Next start attempt jetty would not start:
`2021-02-13 05:21:34 [main] [WARN]-[SourceFile:22] - OmadaJettyServer start failed!_can't create work directory com.tplink.omada.start.a.a: can't create work directory`
This is fixed by adding a second volume mapping:
`-v omada-logs:/opt/tplink/EAPController/work\`
`-v omada-logs:/opt/tplink/work\`

Leads to the the final error which I cannot resolve:
`2021-02-13 05:24:41 [main] [WARN]-[WebInfConfiguration.java:675] - Web application not found ../webapps/omada-web.war`
In the container I can see the war:
`root@677f6b155c6a:/opt/tplink/EAPController# ls -l /opt/tplink/EAPController/webapps`
`total 6048`
`-rw-r--r-- 1 root root 6189282 Feb 13 02:16 omada-web.war`

I am wondering if a directory variable (OMADA_HOME?) is set incorrectly but this is the extent that I can troubleshoot.

mbentley, 2021-02-13:
Could you tell me exactly how you're running the container?  The entrypoint script is responsible for setting the permissions so it should always work.  Are you skipping the entrypoint script by chance?  Your full command you're using to start the container would be good along with the full logs from the container for troubleshooting purposes.  I just built the image from the Dockerfile and it starts fine in a super basic config with no volumes mapped:

```
$ docker run -it --rm test
INFO: Time zone set to 'Etc/UTC'
INFO: Starting Omada Controller as user omada
tail: cannot open '/opt/tplink/EAPController/logs/server.log' for reading: No such file or directory
tail: '/opt/tplink/EAPController/logs/server.log' has appeared;  following new file
2021-02-13 14:56:47 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-02-13 14:56:47 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2021-02-13 14:56:48 [main] [INFO]-[SourceFile:167] - mongodb process id is 39
2021-02-13 14:56:48 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2021-02-13 14:57:07 [main] [INFO]-[SourceFile:35] - monitor context initialing...
2021-02-13 14:57:09 [main] [INFO]-[SourceFile:290] - restart all customize tasks.
2021-02-13 14:57:11 [device-discovery-server] [INFO]-[SourceFile:54] - Start device discovery server
2021-02-13 14:57:11 [device-manage-server] [INFO]-[SourceFile:111] - Start device manage server
2021-02-13 14:57:11 [device-adopt-server] [INFO]-[SourceFile:82] - Start device adopt server
2021-02-13 14:57:12 [main] [INFO]-[SourceFile:126] - no need to compatible db.
2021-02-13 14:57:12 [main] [INFO]-[SourceFile:20] - Omada Controller started
```

Same with volumes:

```
$ docker run -it --rm -v omada-data:/opt/tplink/EAPController/data -v omada-work:/opt/tplink/EAPController/work -v omada-logs:/opt/tplink/EAPController/logs test
INFO: Time zone set to 'Etc/UTC'
INFO: Starting Omada Controller as user omada
tail: cannot open '/opt/tplink/EAPController/logs/server.log' for reading: No such file or directory
tail: '/opt/tplink/EAPController/logs/server.log' has appeared;  following new file
2021-02-13 14:59:43 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-02-13 14:59:43 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2021-02-13 14:59:44 [main] [INFO]-[SourceFile:167] - mongodb process id is 38
2021-02-13 14:59:44 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2021-02-13 15:00:04 [main] [INFO]-[SourceFile:35] - monitor context initialing...
2021-02-13 15:00:07 [main] [INFO]-[SourceFile:290] - restart all customize tasks.
2021-02-13 15:00:08 [device-discovery-server] [INFO]-[SourceFile:54] - Start device discovery server
2021-02-13 15:00:08 [device-manage-server] [INFO]-[SourceFile:111] - Start device manage server
2021-02-13 15:00:08 [device-adopt-server] [INFO]-[SourceFile:82] - Start device adopt server
2021-02-13 15:00:09 [main] [INFO]-[SourceFile:126] - no need to compatible db.
2021-02-13 15:00:09 [main] [INFO]-[SourceFile:20] - Omada Controller started
```

Timeteo, 2021-02-13:
Thanks.
I am attempting to create/run it from Portainer creating the container setting based on the example docker run command in the readme.

closed 2021-02-13 by Timeteo