mbentley/docker-omada-controller #85: Unable to connect to Omada

RogerMCT, 2021-02-17:
I'm stuck.  I have tried several times to deploy this docker but I can't connect to it using the host 192.168.1.1 or localhost ON the host. 

This is running on Windows 10 using Docker Desktop.  Firewall is turned off on the host machine.

My Command:
docker run -d --name Omada-Controller --restart unless-stopped --net host -e MANAGE_HTTP_PORT=8088 -e MANAGE_HTTPS_PORT=8043 -e PORTAL_HTTP_PORT=8088 -e PORTAL_HTTPS_PORT=8843  -e SHOW_SERVER_LOGS=true -e SHOW_MONGODB_LOGS=false -e TZ=America/Indianapolis -v $HOME/docker/omada/data:/opt/tplink/EAPController/data -v $HOME/docker/omada/work:/opt/tplink/EAPController/work -v $HOME/docker/omada/logs:/opt/tplink/EAPController/logs mbentley/omada-controller:latest
[image]
[image]
[image]

mbentley, 2021-02-17:
Are you trying the IP or localhost on 8088 via http? or https on 8043?

RogerMCT, 2021-02-17:
I have tried the host IP and those ports and localhost with those ports.  I have two other images running (minecraft, and terraria) successfully, but I'm unable to access this one.  If you would like to hope in and take a look, I can set that up.

mbentley, 2021-02-17:
Are the others running net=host?

RogerMCT, 2021-02-17:
No, Minecrasft is -p=25565 and Terreria is -p=7777

mbentley, 2021-02-17:
Give this a shot:

```
docker run -d --name Omada-Controller --restart unless-stopped -p 29810:29810 -p 29811:29811 -p 29812:29812 -p 29813:29813 -p 29810:29810/udp -p 29811:29811/udp -p 29812:29812/udp -p 29813:29813/udp -p 8088:8088 -p 8043:8043 -p 8843:8843 -e MANAGE_HTTP_PORT=8088 -e MANAGE_HTTPS_PORT=8043 -e PORTAL_HTTP_PORT=8088 -e PORTAL_HTTPS_PORT=8843 -e SHOW_SERVER_LOGS=true -e SHOW_MONGODB_LOGS=false -e TZ=America/Indianapolis -v $HOME/docker/omada/data:/opt/tplink/EAPController/data -v $HOME/docker/omada/work:/opt/tplink/EAPController/work -v $HOME/docker/omada/logs:/opt/tplink/EAPController/logs mbentley/omada-controller:latest
```

RogerMCT, 2021-02-17:
THAT WORKED PERFECT, and it found my access points (I need to update their firmware).  YOU ROCK MBENTLEY!!
[image]

mbentley, 2021-02-17:
Awesome, glad to hear it works!

closed 2021-02-17 by mbentley