mbentley/docker-omada-controller #90: This x.x.x.x page can’t be found

Vligsu, 2021-03-17:
Hi mbentley,
Hi everyone,

An error message is returned, when accessing the login page:
No webpage was found for the web address: http://x.x.x.x:7086/login
HTTP ERROR 404

Logs:
INFO: Time zone set to 'Europe/Berlin'
INFO: Starting Omada Controller as user omada
2021-03-17 22:05:51 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-03-17 22:05:52 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2021-03-17 22:05:56 [main] [INFO]-[SourceFile:167] - mongodb process id is 44
2021-03-17 22:05:56 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2021-03-17 22:06:09 [main] [INFO]-[SourceFile:35] - monitor context initialing...
2021-03-17 22:06:10 [main] [INFO]-[SourceFile:290] - restart all customize tasks.
2021-03-17 22:06:11 [main] [INFO]-[SourceFile:126] - no need to compatible db.
2021-03-17 22:06:11 [main] [INFO]-[SourceFile:20] - Omada Controller started
2021-03-17 22:06:12 [device-discovery-server] [INFO]-[SourceFile:54] - Start device discovery server
2021-03-17 22:06:12 [device-manage-server] [INFO]-[SourceFile:111] - Start device manage server
2021-03-17 22:06:12 [device-adopt-server] [INFO]-[SourceFile:82] - Start device adopt server
2021-03-17 22:06:42 [scheduled-pool-5] [INFO]-[SourceFile:111] - Controller has not bound owner, cancel online checking Firmware tasks.
2021-03-17 22:07:10 [scheduled-pool-3] [INFO]-[SourceFile:106] - Start Controller Dst Task.

Smells like an insufficient permission, but I can't figure it out on my own :(

Docker-compose via Portainer:
---
version: "2.1"
services:
  omada-controller:
    image: mbentley/omada-controller
    container_name: omada
    environment:
      - TZ=Europe/Berlin
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
    volumes:
      - /volume6/docker/omada/data:/opt/tplink/EAPController/data
      - /volume6/docker/omada/logs:/opt/tplink/EAPController/logs
      - /volume6/docker/omada/work:/opt/tplink/EAPController/work
    ports:
      - 7086:8088  # manage http port
      - 7087:8043  # manage https port
      - 7088:8843  # portal https port
      - 29810:29810
      - 29810:29810/udp
      - 29811:29811
      - 29811:29811/udp
      - 29812:29812
      - 29812:29812/udp
      - 29813:29813
      - 29813:29813/udp
    restart: unless-stopped

Help is much appreciated

Cheers

mbentley, 2021-03-17:
If I remember correctly, there is an issue with mapping the ports where they differ from the host to inside the container but try https://x.x.x.x:7087/login instead of http.  I believe the issue is that it doesn't know what port to redirect http requests to https because the ports are mapped by Docker and not the app itself.

Vligsu, 2021-03-17:
Thank you for this prompt reply! Though it didn't work like this - https://x.x.x.x:7087/login - it did work with defaults 8088:8088, 8043:8043, 8843:8843, more specifically with https://x.x.x.x:8043/login. Problem solved, thanks for pointing out!

Maybe a question for another thread/issue. Is it possible to implement custom port mapping?

mbentley, 2021-03-17:
Yes it is possible but you need to specify the environment variables to modify the ports inside the container - i.e these:

```
- MANAGE_HTTP_PORT=8088
- MANAGE_HTTPS_PORT=8043
- PORTAL_HTTP_PORT=8088
- PORTAL_HTTPS_PORT=8843
```

closed 2021-03-18 by mbentley