mbentley/docker-omada-controller #92: Unable to start container due to mongodb

tundrastar1, 2021-03-18:
I havent been able to get the container to start due to issues with mongodb. I have done all the usual things:

Have ensured that there is plenty of space both in docker root and the bind mount folders
Ensured permissions are correct

The container is being run using the command on the webpage except for the -v option

Any suggestions?

But the logs show this:

```
INFO: Time zone set to 'Etc/UTC'

WARNING: smallfiles was passed but is not supported in >= 4.1 with the WiredTiger engine in use by MongoDB

INFO: skipping setting smallfiles option

WARNING: owner or group (0:0) not set correctly on '/opt/tplink/EAPController/data'

INFO: setting correct permissions

WARNING: owner or group (0:0) not set correctly on '/opt/tplink/EAPController/work'

INFO: setting correct permissions

WARNING: owner or group (0:0) not set correctly on '/opt/tplink/EAPController/logs'

INFO: setting correct permissions

INFO: Starting Omada Controller as user omada

2021-03-18 17:18:24 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties

2021-03-18 17:18:25 [main] [INFO]-[SourceFile:89] - going to start local mongod.

2021-03-18 17:20:05 [main] [WARN]-[SourceFile:151] - Fail to start mongo DB server

2021-03-18 17:20:05 [main] [WARN]-[SourceFile:78] - com.tplink.omada.start.a.a: Fail to start mongo DB server

com.tplink.omada.start.a.a: Fail to start mongo DB server

	at com.tplink.omada.start.b.d.a(SourceFile:152) ~[omada-start.jar:?]

	at com.tplink.omada.start.b.b.e(SourceFile:75) [omada-start.jar:?]

	at com.tplink.omada.start.OmadaBootstrap.d(SourceFile:223) [omada-start.jar:?]

	at com.tplink.omada.start.OmadaBootstrap.q(SourceFile:287) [omada-start.jar:?]

	at com.tplink.omada.start.OmadaBootstrap.a(SourceFile:100) [omada-start.jar:?]

	at com.tplink.omada.start.OmadaBootstrap.e(SourceFile:245) [omada-start.jar:?]

	at com.tplink.omada.start.OmadaLinuxMain.b(SourceFile:80) [omada-start.jar:?]

	at com.tplink.omada.start.OmadaLinuxMain.main(SourceFile:25) [omada-start.jar:?]
```

tundrastar1, 2021-03-18:
Forgot to add, the behaviour was the same with or without the SMALL_FILES option...

mbentley, 2021-03-18:
I know your said you're using the commands provided but could you include the exact command you're using?

tundrastar1, 2021-03-18:
Ooh, that was quick many thanks

`sudo docker run -d   --name omada-controller   --restart unless-stopped   -p 8088:8088   -p 8043:8043   -p 8843:8843   -p 29810:29810   -p 29810:29810/udp   -p 29811:29811   -p 29811:29811/udp   -p 29812:29812   -p 29812:29812/udp   -p 29813:29813   -p 29813:29813/udp   -e SMALL_FILES=true   -e MANAGE_HTTP_PORT=8088   -e MANAGE_HTTPS_PORT=8043   -e PORTAL_HTTP_PORT=8088   -e PORTAL_HTTPS_PORT=8843   -e SHOW_SERVER_LOGS=true   -e SHOW_MONGODB_LOGS=false   -e SSL_CERT_NAME="tls.crt"   -e SSL_KEY_NAME="tls.key"   -e TZ=Etc/UTC   -v /srv/dev-disk-by-label-Volume/Appdata/Omada/omada-data:/opt/tplink/EAPController/data   -v /srv/dev-disk-by-label-Volume/Appdata/Omada/omada-work:/opt/tplink/EAPController/work   -v /srv/dev-disk-by-label-Volume/Appdata/Omada/omada-logs:/opt/tplink/EAPController/logs   mbentley/omada-controller:4.2`

tundrastar1, 2021-03-18:
The folders are being created and logs are being written to the folders.

mbentley, 2021-03-18:
Have you ran a container previously with the same directories mounted?  Like maybe with a previous version?  If so and you do not have data yet you care about, I would try clearing your directories and re-running the container.  Otherwise we need to do more debugging.

I simplified your command and ran it interactively to test and it comes up fine:
```
docker run -it --rm --name omada-controller -p 8088:8088 -p 8043:8043 -p 8843:8843 -p 29810:29810 -p 29810:29810/udp -p 29811:29811 -p 29811:29811/udp -p 29812:29812 -p 29812:29812/udp -p 29813:29813 -p 29813:29813/udp -e MANAGE_HTTP_PORT=8088 -e MANAGE_HTTPS_PORT=8043 -e PORTAL_HTTP_PORT=8088 -e PORTAL_HTTPS_PORT=8843 -e SHOW_SERVER_LOGS=true -e SHOW_MONGODB_LOGS=false -e SSL_CERT_NAME="tls.crt" -e SSL_KEY_NAME="tls.key" -e TZ=Etc/UTC -v omada-data:/opt/tplink/EAPController/data -v omada-work:/opt/tplink/EAPController/work -v omada-logs:/opt/tplink/EAPController/logs mbentley/omada-controller:4.2
INFO: Time zone set to 'Etc/UTC'
INFO: Starting Omada Controller as user omada
tail: cannot open '/opt/tplink/EAPController/logs/server.log' for reading: No such file or directory
tail: '/opt/tplink/EAPController/logs/server.log' has appeared;  following new file
2021-03-18 17:33:26 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-03-18 17:33:26 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2021-03-18 17:33:27 [main] [INFO]-[SourceFile:167] - mongodb process id is 61
2021-03-18 17:33:27 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2021-03-18 17:33:48 [main] [INFO]-[SourceFile:35] - monitor context initialing...
2021-03-18 17:33:49 [main] [INFO]-[SourceFile:290] - restart all customize tasks.
2021-03-18 17:33:50 [device-discovery-server] [INFO]-[SourceFile:54] - Start device discovery server
2021-03-18 17:33:50 [device-manage-server] [INFO]-[SourceFile:111] - Start device manage server
2021-03-18 17:33:50 [device-adopt-server] [INFO]-[SourceFile:82] - Start device adopt server
2021-03-18 17:33:51 [main] [INFO]-[SourceFile:126] - no need to compatible db.
2021-03-18 17:33:51 [main] [INFO]-[SourceFile:20] - Omada Controller started
```

For more verbose logging from MongoDB, try adding `-e SHOW_MONGODB_LOGS=true` and see what it outputs since it seems to be related to MongoDB.

tundrastar1, 2021-03-18:
Sadly no joy for me after removing all directories. This is a new install completely.

Here is the mongo log

`2021-03-18T17:42:18.863+0000 I CONTROL  [initandlisten] MongoDB starting : pid=45 port=27217 dbpath=../data/db 64-bit host=9ed79c202940
2021-03-18T17:42:18.864+0000 I CONTROL  [initandlisten] db version v3.6.3
2021-03-18T17:42:18.864+0000 I CONTROL  [initandlisten] git version: 9586e557d54ef70f9ca4b43c26892cd55257e1a5
2021-03-18T17:42:18.865+0000 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1  11 Sep 2018
2021-03-18T17:42:18.865+0000 I CONTROL  [initandlisten] allocator: tcmalloc
2021-03-18T17:42:18.865+0000 I CONTROL  [initandlisten] modules: none
2021-03-18T17:42:18.866+0000 I CONTROL  [initandlisten] build environment:
2021-03-18T17:42:18.866+0000 I CONTROL  [initandlisten]     distarch: x86_64
2021-03-18T17:42:18.866+0000 I CONTROL  [initandlisten]     target_arch: x86_64
2021-03-18T17:42:18.867+0000 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db", journal: { enabled: true } }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2021-03-18T17:42:18.870+0000 I STORAGE  [initandlisten] wiredtiger_open config: create,cache_size=1308M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
2021-03-18T17:42:19.971+0000 E STORAGE  [initandlisten] WiredTiger error (1) [1616089339:971243][45:0x7faeeef5f0c0], connection: ../data/db/WiredTiger.wt: handle-open: open: Operation not permitted
2021-03-18T17:42:19.974+0000 E -        [initandlisten] Assertion: 28595:1: Operation not permitted src/mongo/db/storage/wiredtiger/wiredtiger_kv_engine.cpp 413
2021-03-18T17:42:19.976+0000 I STORAGE  [initandlisten] exception in initAndListen: Location28595: 1: Operation not permitted, terminating
2021-03-18T17:42:19.976+0000 I NETWORK  [initandlisten] shutdown: going to close listening sockets...
2021-03-18T17:42:19.976+0000 I NETWORK  [initandlisten] removing socket file: /tmp/mongodb-27217.sock
2021-03-18T17:42:19.976+0000 I CONTROL  [initandlisten] now exiting
2021-03-18T17:42:19.977+0000 I CONTROL  [initandlisten] shutting down with code:100
2021-03-18T17:44:00.922+0000 I CONTROL  [main] ***** SERVER RESTARTED *****
2021-03-18T17:44:00.941+0000 I CONTROL  [initandlisten] MongoDB starting : pid=80 port=27217 dbpath=../data/db 64-bit host=9ed79c202940
2021-03-18T17:44:00.941+0000 I CONTROL  [initandlisten] db version v3.6.3
2021-03-18T17:44:00.942+0000 I CONTROL  [initandlisten] git version: 9586e557d54ef70f9ca4b43c26892cd55257e1a5
2021-03-18T17:44:00.942+0000 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1  11 Sep 2018
2021-03-18T17:44:00.942+0000 I CONTROL  [initandlisten] allocator: tcmalloc
2021-03-18T17:44:00.943+0000 I CONTROL  [initandlisten] modules: none
2021-03-18T17:44:00.943+0000 I CONTROL  [initandlisten] build environment:
2021-03-18T17:44:00.943+0000 I CONTROL  [initandlisten]     distarch: x86_64
2021-03-18T17:44:00.943+0000 I CONTROL  [initandlisten]     target_arch: x86_64
2021-03-18T17:44:00.943+0000 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db", journal: { enabled: true } }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2021-03-18T17:44:00.945+0000 I STORAGE  [initandlisten] wiredtiger_open config: create,cache_size=1308M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
2021-03-18T17:44:02.047+0000 E STORAGE  [initandlisten] WiredTiger error (17) [1616089442:47839][80:0x7f2e025640c0], connection: ../data/db/WiredTiger.wt: handle-open: open: File exists
2021-03-18T17:44:02.048+0000 I STORAGE  [initandlisten] WiredTiger message unexpected file WiredTiger.wt found, renamed to WiredTiger.wt.1
2021-03-18T17:44:02.050+0000 E STORAGE  [initandlisten] WiredTiger error (1) [1616089442:50240][80:0x7f2e025640c0], connection: ../data/db/WiredTiger.wt: handle-open: open: Operation not permitted
2021-03-18T17:44:02.053+0000 E -        [initandlisten] Assertion: 28595:1: Operation not permitted src/mongo/db/storage/wiredtiger/wiredtiger_kv_engine.cpp 413
2021-03-18T17:44:02.054+0000 I STORAGE  [initandlisten] exception in initAndListen: Location28595: 1: Operation not permitted, terminating
2021-03-18T17:44:02.054+0000 I NETWORK  [initandlisten] shutdown: going to close listening sockets...
2021-03-18T17:44:02.055+0000 I NETWORK  [initandlisten] removing socket file: /tmp/mongodb-27217.sock
2021-03-18T17:44:02.055+0000 I CONTROL  [initandlisten] now exiting
2021-03-18T17:44:02.055+0000 I CONTROL  [initandlisten] shutting down with code:100
2021-03-18T17:45:43.124+0000 I CONTROL  [main] ***** SERVER RESTARTED *****
2021-03-18T17:45:43.140+0000 I CONTROL  [initandlisten] MongoDB starting : pid=116 port=27217 dbpath=../data/db 64-bit host=9ed79c202940
2021-03-18T17:45:43.141+0000 I CONTROL  [initandlisten] db version v3.6.3
[... cut ...]
2021-03-18T17:45:44.263+0000 E -        [initandlisten] Assertion: 28595:1: Operation not permitted src/mongo/db/storage/wiredtiger/wiredtiger_kv_engine.cpp 413
2021-03-18T17:45:44.264+0000 I STORAGE  [initandlisten] exception in initAndListen: Location28595: 1: Operation not permitted, terminating
2021-03-18T17:45:44.265+0000 I NETWORK  [initandlisten] shutdown: going to close listening sockets...
2021-03-18T17:45:44.265+0000 I NETWORK  [initandlisten] removing socket file: /tmp/mongodb-27217.sock
2021-03-18T17:45:44.265+0000 I CONTROL  [initandlisten] now exiting
2021-03-18T17:45:44.266+0000 I CONTROL  [initandlisten] shutting down with code:100
`

mbentley, 2021-03-18:
What is the filesystem that is in use with your persistent storage?  Seems like there is a problem with the WiredTiger storage engine interacting with the file system from my best guess.

tundrastar1, 2021-03-19:
Thanks for the pointer above. Seeing the earlier issues, I had moved the container to another mounted storage filesystem. Unfortunately, this was ntfs. Clearly, mongodb does not like this. I reformatted the filesystem to ext4 and it worked like a charm. I am still new to containers and this has been a learning experience for me.

Once again, many thanks for this, really appreciated, your efforts in creating the image and more importantly helping resolve issues even though they were caused by me.

tundrastar1, 2021-03-19:
BTW, for anyone else: I am running this as a docker container under the openmediavault NAS appliance, on a box that has only 8 GB of boot disk. The container creates a big image on the directory where docker is installed and if there is no space, it will fail. Not only does one need to move the location where container images are stored (if you have less memory like I did), but also the location where the logs will be stored/

closed 2021-03-19 by tundrastar1