mbentley/docker-omada-controller #93: Impossible to change http port? (container stuck in starting loop)

tophee, 2021-03-28:
Hi!

I'm using the following docker compose in portioner to create my container:

~~~
version: '2.2'

services:
  omada:
    image: mbentley/omada-controller:4.2
    restart: unless-stopped
    container_name: omada-controller
    network_mode: host
    volumes:
      - "/etc/localtime:/etc/localtime:ro"
      - "/etc/timezone:/etc/timezone:ro"
      - "/zfs/NAS/config/omada/data:/opt/tplink/EAPController/data"
      - "/zfs/NAS/config/omada/work:/opt/tplink/EAPController/work"
      - "/zfs/NAS/config/omada/logs:/opt/tplink/EAPController/logs"
    environment:
      - MANAGE_HTTP_PORT=9088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT:=9088
      - PORTAL_HTTPS_PORT=8843
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - TZ=Europe/Stockholm
~~~

I changed the HTTP port because 8088 is apparently already in use (I'm not aware of anything using that port, but the easiest solution was to just use a different port). However, it looks like the container refuses to use the new port (9088) because it still complains about 8088 being in use. See the logs here:

~~~
INFO: Time zone set to 'Europe/Stockholm'
INFO: Setting 'manage.http.port' to 9088
INFO: Database directory missing; creating '/opt/tplink/EAPController/data/db'
done
INFO: Starting Omada Controller as user omada
2021-03-28 03:55:53 [main] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-03-28 03:55:53 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2021-03-28 03:55:55 [main] [INFO]-[SourceFile:167] - mongodb process id is 38
2021-03-28 03:55:55 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2021-03-28 03:55:55 [main] [ERROR]-[SourceFile:103] - Port 8088 is already in use.
2021-03-28 03:55:55 [main] [ERROR]-[SourceFile:128] - Port 8088 is already in use. Release the port and try again.
2021-03-28 03:55:55 [main] [ERROR]-[SourceFile:263] - Environment is not ok for controller running
Bootstrap prepare failed, going to exit.
ShutdownHook: service stopped.
INFO: Time zone set to 'Europe/Stockholm'
WARNING: Unable to change 'manage.http.port' to 9088 after initial run; change the ports via the web UI
INFO: Starting Omada Controller as user omada
2021-03-28 03:56:04 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-03-28 03:56:04 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2021-03-28 03:56:06 [main] [INFO]-[SourceFile:167] - mongodb process id is 32
2021-03-28 03:56:06 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2021-03-28 03:56:06 [main] [ERROR]-[SourceFile:103] - Port 8088 is already in use.
2021-03-28 03:56:06 [main] [ERROR]-[SourceFile:128] - Port 8088 is already in use. Release the port and try again.
2021-03-28 03:56:06 [main] [ERROR]-[SourceFile:263] - Environment is not ok for controller running
Bootstrap prepare failed, going to exit.
ShutdownHook: service stopped.
...
~~~
 and so on. 

Is this a bug or am I doing something wrong?

mbentley, 2021-03-28:
If I remember correctly, there is something odd about how the Omada Controller software deals with the ports.  I believe the environment variable only works on the initial run.  After that, Omada Controller gets the ports it should use from a database and I am not sure how to access it.  If you haven't used the Omada Controller yet, I would just remove all persistent data and re-launch it with your new port config and that should take care of it.

tophee, 2021-03-28:
Unfortunately, I already tried that. I spent half the night removing the entire container as well as the persistent data in the `data` and `work` directories and then rebuilding the container, but the result is always the same. (I did it again right now, just to be sure that the tired version of myself didn't miss anything.)

The log I provided above is the start of the log after a clean start (you can see it because it's creating the database directory, and you are right, it looks like setting the http port only works the first time (`INFO: Setting 'manage.http.port' to 9088` and after that it complains about it (`WARNING: Unable to change 'manage.http.port' to 9088 after initial run; change the ports via the web UI`), but then it nevertheless complains about port 8088 being in use - and that is still during the first run. Indeed, it is the unavailability of the port that causes the restart (`Bootstrap prepare failed, going to exit.`)

Any other ideas? 

I understand that it is working fine for you, right? So I wonder what could be the issue here. I'm not very good at docker and my trial and error guess work has failed so far. Is there something wrong with my docker compose script?

mbentley, 2021-03-28:
No, I do not have access to the source and I am not sure what `SourceFIle:103` is referring to.  I just did a clean run and it looks like it worked as expected:

```
docker run -it --rm \
  --name omada-controller \
  --net=host \
  -e MANAGE_HTTP_PORT=9088 \
  -e MANAGE_HTTPS_PORT=8043 \
  -e PORTAL_HTTP_PORT=9088 \
  -e PORTAL_HTTPS_PORT=8843 \
  -e SHOW_SERVER_LOGS=true \
  -e SHOW_MONGODB_LOGS=false \
  -e TZ=Etc/UTC \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:4.2```

```
INFO: Time zone set to 'Etc/UTC'
INFO: Setting 'manage.http.port' to 9088
INFO: Setting 'portal.http.port' to 9088
INFO: Starting Omada Controller as user omada
tail: cannot open '/opt/tplink/EAPController/logs/server.log' for reading: No such file or directory
tail: '/opt/tplink/EAPController/logs/server.log' has appeared;  following new file
2021-03-28 13:48:28 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-03-28 13:48:28 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2021-03-28 13:48:29 [main] [INFO]-[SourceFile:167] - mongodb process id is 62
2021-03-28 13:48:29 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2021-03-28 13:48:47 [main] [INFO]-[SourceFile:35] - monitor context initialing...
2021-03-28 13:48:48 [main] [INFO]-[SourceFile:290] - restart all customize tasks.
2021-03-28 13:48:50 [device-discovery-server] [INFO]-[SourceFile:54] - Start device discovery server
2021-03-28 13:48:50 [device-manage-server] [INFO]-[SourceFile:111] - Start device manage server
2021-03-28 13:48:50 [device-adopt-server] [INFO]-[SourceFile:82] - Start device adopt server
2021-03-28 13:48:51 [main] [INFO]-[SourceFile:126] - no need to compatible db.
2021-03-28 13:48:51 [main] [INFO]-[SourceFile:20] - Omada Controller started
```

I can see the updated ports being used as expected:

```
root@builder-ssh:/# netstat -ltnp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 127.0.0.1:27217         0.0.0.0:*               LISTEN      525/mongod
tcp        0      0 0.0.0.0:29811           0.0.0.0:*               LISTEN      454/java
tcp        0      0 0.0.0.0:29812           0.0.0.0:*               LISTEN      454/java
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      27/sshd
tcp        0      0 127.0.0.11:37117        0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:9088            0.0.0.0:*               LISTEN      454/java
tcp        0      0 0.0.0.0:8843            0.0.0.0:*               LISTEN      454/java
tcp        0      0 0.0.0.0:8043            0.0.0.0:*               LISTEN      454/java
tcp6       0      0 :::22                   :::*                    LISTEN      27/sshd
tcp6       0      0 :::2376                 :::*                    LISTEN      41/dockerd
```

Are you sure that you're stopping the container, removing the volumes and then re-running it?

As you can see in the entrypoint [here](https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-4.1.x.sh#L28-L32), it is looking for an existing file in the data directory so that tells me that the volumes were not removed and the container is re-using existing data.

tophee, 2021-03-28:
> Are you sure that you're stopping the container, removing the volumes and then re-running it?

Yes, as sure as you can be when you do the following:

1. In portioner, select the container an click "Remove"
2. in the modal asking whether all non-persistent data should also be removed, select "yes" (or whatever it says).
3. Verify in portioner that the container is gone (I can triple check with the docker commands you provided when I'm back at my desk, but I have no reason to believe that any old data is left (see below). 
3. Via ssh, do `sudo rm -R *` in the data directory as well as in the work directory.
4. Double check with `ls` whether both directories are indeed empty.

> As you can see in the entrypoint here, it is looking for an existing file in the data directory so that tells me that the volumes were not removed and the container is re-using existing data.

Yes, but only the second time it runs, i.e. after bootstrapping failed. It is not seeing any existing data in the first round. So the existing data it is seeing is the one from the initial bootstrap.

tophee, 2021-03-28:
Quick update: I ran your docker command above and it also went through. So the problem seems to be either with my docker-compose script or with portainer...

So now I wonder: does this thing about changing the port only ay first run mean, that I will have problems when I restart the container? Or what is meant by first time?

mbentley, 2021-03-28:
I would expect that to be the case.  When you run the Docker command manually, it is likely using different named volumes than that of what the compose used, therefore it is starting with empty data volumes.  The issue is that the only easy method that I know of to change the ports is by a config file that the Omada Controller only reads the first time when the data in the database is populated.  After that, it reads the values from that db.  So it is not a problem to restart or recreate the container afterwards because it will get the ports from the database in the future.

The problem with portainer is that is is probably obscuring the fact that when the containers are removed via the compose that is deployed, it leaves the volumes.  That is expected because persistent data is persistent.  You have to instruct it to remove the volumes.

tophee, 2021-03-29:
> The problem with portainer is that is is probably obscuring the fact that when the containers are removed via the compose that is deployed, it leaves the volumes. That is expected because persistent data is persistent. You have to instruct it to remove the volumes.

I am instructing it to remove all non-persistent volumes. And I'm manually removing the persistent data from the `data` and `work` directories. Is there any other place where data is stored?

Another strange thing I observed: when I build the container using this command
~~~
docker run -it --rm   --name omada-controller   --net=host   -e MANAGE_HTTP_PORT=9088   -e MANAGE_HTTPS_PORT=8043   -e PORTAL_HTTP_PORT=9088   -e PORTAL_HTTPS_PORT=8843   -e SHOW_SERVER_LOGS=true   -e SHOW_MONGODB_LOGS=false   -e TZ=Etc/UTC   -v /zfs/NAS/config/omada/data:/opt/tplink/EAPController/data   -v /zfs/NAS/config/omada/work:/opt/tplink/EAPController/work   -v /zfs/NAS/config/omada/logs:/opt/tplink/EAPController/logs   mbentley/omada-controller:4.2
~~~

nothing is written to `/zfs/NAS/config/omada/work/`, `/zfs/NAS/config/omada/data/` or `/zfs/NAS/config/omada/logs` and when the container stops running `docker start omada-controller` will return `Error response from daemon: No such container: omada-controller` So everything just disappears into thin air! Very strange.

mbentley, 2021-03-29:
It's not writing to those directories because it is writing to named volumes.  Those are created under `/var/lib/docker/volumes`. You can stop and remove the container and the volumes and update the run command to use your volumes.

tophee, 2021-03-29:
Why would it write to `/var/lib/docker/volumes` when I tell it to write to `/zfs/NAS/config/omada/work/`?

mbentley, 2021-03-29:
Oh sorry, I missed that you were running it with the volumes mounted to the "correct" path.

tophee, 2021-03-29:
What baffles me more is where the container disappeared to:

~~~
$ ls /var/lib/docker/containers/
1a5b0fd271628c59c819016cee6ffb4ca224dbec79a769106985ea25f1c44b4c
1f577df0f620258fc463a43174cd33e746e3812577d0c1a1f46d57f1919914eb
29549f7cf795c72299de639da210b5094aaceada46c0f62f59ea8b72b26c2d2b
2f480a3fd00c56ee732e0bd7da41f1c9e05115cbf72cc839aef721d72db97ee8
40c715614e56c5c81526388944e1e0f0a899acb4b1fa2aa4b7192ec182accdbd
68b15e3dcf853617a2600be47fff17a1be73786c78ebd1b4e8730c6972f64305
8b563788d371d27c21c5302e64d829423a6392f378a921261a5fe79c67520197
8cea7ffcc71e751c14db59290c1a243da0e054a302a0a467a30a0c85056f2492
91561add00436acb445a00a8b738162abc5e583bdf12be6da9f6fc05af05f9ee
92cfe41a8a933b4322e309cc1885aede296b1eb4f9b6e358f5dd8bd4783e6903
baf88979cad96fcd5d49fb031029fb5bb2c6536803970754136300ac89edd9b3
bd334a7e5308fa13c20b86aba3479aed98786fc2c540b8bc801137038c28aa50
bdb8ea667fe4b9f64edf28d38c3521ac1097a1325dfc32f9dce9e8227c1f3e15
c37b372f432e61fe6e598a063ab0b7360219bddcdba7770ab36a5d6b0f511f6b
ca4f99c4721446ef38030dab35d0f84c2fdb6095fd4182274cb4563481bde451
ce80e38ed70f5cf072517fbc4941a88cdb475c54128f92a18dfe43ef03adcb6b
d4a01ebec572ae1a2b4304dc55082b52860ec82262cd824d96c935e5e5b749d5
d55b4e858ce20a9c03b77f4e28f744d68116253e4e524c35a56c7cdf0355044b
db2c2b4e4cebec24bbb1b9b067bef1ff1e84bbd000c7d5f6bbbd3529e1bac35b
df2f09996faed390e06c1ba46ff3f0f85b88eae8aa68cec1b7c885f5732e0d39
e0b41008c8b8a1314ba741cb251610d3dacefff356a00e08a6e29e463d303098
edfb71d279a74f71bb90937d0b1423d797c959e04fce17aca9ebc555957a3f14
$ 
~~~

mbentley, 2021-03-29:
So there should definitely be data in that directory, assuming it launched completely:

```
# docker exec -it omada-controller ls -l /opt/tplink/EAPController/work /opt/tplink/EAPController/data
/opt/tplink/EAPController/data:
total 18
drwxr-xr-x 2 omada omada  16 Mar 29 02:15 autobackup
drwxr-xr-x 4 omada omada 258 Mar 29 06:40 db
drwxr-xr-x 3 omada omada   3 Feb 19 19:04 device-firmware

/opt/tplink/EAPController/work:
total 1
drwxr-xr-x 4 omada omada 4 Mar 28 04:28 omada-web
```

Same as what I see directly on my host:
```
# ls -l /zfs/apps/omada-controller/data /zfs/apps/omada-controller/work
/zfs/apps/omada-controller/data:
total 18
drwxr-xr-x 2 omada omada  16 Mar 29 02:15 autobackup/
drwxr-xr-x 4 omada omada 258 Mar 29 06:40 db/
drwxr-xr-x 3 omada omada   3 Feb 19 19:04 device-firmware/

/zfs/apps/omada-controller/work:
total 1
drwxr-xr-x 4 omada omada 4 Mar 28 04:28 omada-web/
```

Those containers you see would show up with a `docker ps -a`.  They're just stopped containers.  The reason the Omada Controller container disappears after being stopped with the command you were using is that it has `--rm` which removes the container when stopped.

tophee, 2021-03-29:
> The reason the Omada Controller container disappears after being stopped with the command you were using is that it has --rm which removes the container when stopped.

Oh. ... That's embarrassing. Sorry about that. I should rely less on copy and paste...

tophee, 2021-05-21:
I'm still having issues with bootstrapping the container. I wanted to upgrade to the latest version and got the same boot-loop as described above (about the port already being in use). I solved (?) this by removing the database directory from the persistent storage, starting the container with the new image and once it successfully initialized, I put the database directory back where it was. After that, the restart of the container worked fine, except that the adoption of the Router (R605) failed. Also manual adoption fails and I cannot login to the routers web-interface either. Could it be that the new container managed to change the routers password?

In any case: what is the correct way to update the container when there is a new firmware?

mbentley, 2021-05-22:
Unless noted in the README for specific upgrades where there are breaking changes from the Omada Controller software, it should just be changing the image tag version number.  If that changes, a note will be added in the upgrade section.

tophee, 2021-05-22:
Yes, I saw the special procedure fro upgrading from 4.1, but this was not the case for me.

So you are saying that all you do to upgrade is to stop the container and issue the same docker command as previously, just with the new image version number? 

In my case, that didn't work because "a container with the same name already exits", obviously. So I removed it. I am not sure, but I think I did not remove non-persistent data at first but when that didn't work I did remove it. And when that didn't work I also removed the `/db` part of the persistent data.

Where did I go wrong?

OranguTech, 2021-05-24:
@tophee  - 
>So I removed it. I am not sure, but I think I did not remove non-persistent data at first but when that didn't work I did remove it. And when that didn't work I also removed the /db part of the persistent data.

That's where you went wrong. >:P 
You have to understand the difference between an image and a container/instance, and how the stateful data is stored. What I did when I encountered the same thing was (googled how to)rename the previous/existing container, then ran a new one w/ the new image (but same startup command, so it was loading the previously saved/used data). Once I verified everything was working, then deleted the old container and image.

tophee, 2021-06-21:
>What I did when I encountered the same thing was (googled how to)rename the previous/existing container, then ran a new one w/ the new image  (but same startup command, so it was loading the previously saved/used data).

@OranguTech The problem is that when I do exactly that, I get:

````
WARNING: Unable to change 'manage.http.port' to 9088 after initial run; change the ports via the web UI
WARNING: Unable to change 'portal.http.port' to 9088 after initial run; change the ports via the web UI
INFO: Starting Omada Controller as user omada
2021-06-21 23:29:03 [main] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-06-21 23:29:03 [main] [INFO]-[SourceFile:91] - going to start local mongod.
2021-06-21 23:29:05 [main] [INFO]-[SourceFile:167] - mongodb process id is 35
2021-06-21 23:29:05 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2021-06-21 23:29:05 [main] [ERROR]-[SourceFile:46] - Port 8088 is already in use.
Bootstrap prepare failed, going to exit.
2021-06-21 23:29:05 [main] [ERROR]-[SourceFile:71] - Port 8088 is already in use. Release the port and try again.
2021-06-21 23:29:05 [main] [ERROR]-[SourceFile:270] - Environment is not ok for controller running
ShutdownHook: service stopped.

````

So I'm stuck once again...

OranguTech, 2021-06-21:
"Port 8080 is in use". I'm guessing you have a container currently running using that, but if not, *something* on your system is using that. Look up how to view ports in use for your OS.

tophee, 2021-06-22:
According to `netstat`, 8088 is being used by `influxd`. I don't know how to change it and I would like to think that I don't need to change it because omada and influxdb have been living next to each other for months. The thing is: I changed the port for Omada to `9088` and that works, but for some strange reason that only works the first time you start the container.  So I think I somehow need to make Omada think that it is starting for the first time...

mbentley, 2021-06-22:
OK, so I can now reproduce this.

mbentley, 2021-06-22:
I am working on patching the entrypoint to fix this.  Looks good now:

```
$ docker exec -it omada-controller netstat -ltnpu
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:9088            0.0.0.0:*               LISTEN      -
tcp        0      0 127.0.0.11:36321        0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:8843            0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:8043            0.0.0.0:*               LISTEN      -
tcp        0      0 127.0.0.1:27217         0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:29811           0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:29812           0.0.0.0:*               LISTEN      -
tcp6       0      0 :::2376                 :::*                    LISTEN      -
udp        0      0 127.0.0.11:48254        0.0.0.0:*                           -
udp        0      0 0.0.0.0:27001           0.0.0.0:*                           -
udp        0      0 0.0.0.0:29810           0.0.0.0:*                           -
```

mbentley, 2021-06-22:
So it would appear that you can modify the portal ports after the first run via environment variables but not the controller port.

mbentley, 2021-06-22:
Could you pull the latest image that was just built and see if it fixes the issue?  I'll reopen this until you can confirm.

tophee, 2021-06-22:
So if I understand your netstats correctly, what we're seeing is that the container uses the non-default port at first run as well as when it is restarted, but not when the container is recreated? Or rather: it still reserves the non-default port but really uses the default one. Right? 

But what exactly does this mean:

> Looks like the Omada controller does actually pull the values from the omada.properties file for the portal. When the container is removed, it isn't setting the values again.

Are you saying that you previously assumed that the controller will forget the non-default port when the container is recreated but it turns out that it doesn't? Which explains why both the default and the non-default ports are active after recreating the container?

More specifically: you thought that none of the ports can be set via environment variables but it turns out that while this is the case for the controller port (whatever that is) you can actually set the portal port this way?

And then you fixed the image so that it wont use the default portal port if the saved config specifies a different one? (sorry, this is new territory for me)

Which means that with the new image it should no longer complain about the default port being occupied?

So in order to test this, I would need to put my influxdb back into host mode (or expose 8088) before I pull the new image?

And finally, just to make sure that I don't mess up the test: to recreate the container with the latest image, do I use "recreate" (with "pull new image") or do I run the same docker command as I did above when portainer didn't do the trick? Sounds like portainer should work now?

mbentley, 2021-06-22:
Correct.  The reason why the ports change on initial startup is that the entrypoint writes them to a `omada.properties` file which is read by the controller on startup.  On the first startup, the port for the controller itself (MANAGE_HTTP_PORT & MANAGE_HTTPS_PORT) appear to be read from the properties file, as do PORTAL_HTTP_PORT & PORTAL_HTTPS_PORT.  The behavior I noticed in the 4.x series is that after you have run the controller once, the "manage" ports are written to the mongodb and pulled from there when the controller starts.  Changing the values after the fact via environment variable did not have an effect on which ports are used.

I incorrectly assumed that the same behavior existed for the portal ports (used for guest authentication portal if you have it enabled) and that is what was conflicting with your influxdb.  So I set up the entrypoint to basically ignore the environment variables and not write an `omada.properties` and output a warning which is why when the container is removed after the first time you start the controller, the port conflict comes up.

The changes I made are to always write the file.  You can see the diff here: https://github.com/mbentley/docker-omada-controller/commit/503799f03788b19aba88f2d4adb946d9fc948c3e of the changes I made.

So in order to test this, yes it would be best if you switched influxdb back to either host mode or expose 8088 again and pull the new image.  You would need to stop the omada controller, remove it, and create it again as the changes are in the new image so you would need to make sure it is re-created from the new image.

tophee, 2021-09-10:
I finally got down to testing this (sorry for the delay) and as you might have guessed, it works flawlessly.

I switched influxdb back to host mode and then recreated the omada container in portainer (using the "pull latest image" option and the recreation worked fine, but I was still on 4.3.5 because I previously had specified `4.3` as my image. So I edited the container in portainer, changing the image to `latest` and deployed it. It also started without any issues.

So, thanks a million for helping me with this. I'm glad that all my stupid questions nevertheless lead to something like a bug-fix (implying that it wasn't *only* me being clueless). Thanks for maintaining this image and providing such excellent support.

commit 503799f0 (2021-06-22): Changed the port modification behavior; fixes #93
closed 2021-09-10 by tophee