mbentley/docker-omada-controller #94: Log filling up with warnings

Javerre, 2021-03-31:
Hi,

I have a problem with my Omada setup. The log is filling up with warnings such as:

```
2021-03-31 23:01:32 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /172.29.20.1/22 [/172.29.23.255], Exception: java.lang.NullPointerException                                                                                                                                                       
2021-03-31 23:06:32 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /172.29.56.1/22 [/172.29.59.255], Exception: java.lang.NullPointerException                                                                                                                                                       
2021-03-31 23:06:32 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /172.29.36.1/22 [/172.29.39.255], Exception: java.lang.NullPointerException                                                                                                                                                       
2021-03-31 23:06:32 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /172.29.32.1/22 [/172.29.35.255], Exception: java.lang.NullPointerException                                                                                                                                                       
2021-03-31 23:06:32 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /172.29.28.1/22 [/172.29.31.255], Exception: java.lang.NullPointerException                                                                                                                                                       
2021-03-31 23:06:32 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /172.29.24.1/22 [/172.29.27.255], Exception: java.lang.NullPointerException                                                                                                                                                       
2021-03-31 23:06:32 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /172.29.20.1/22 [/172.29.23.255], Exception: java.lang.NullPointerException   
```                                             

I'm running the latest container (ID 294f27cd5035). 
The bug is reproduced even if I blow away the config & recreate it. 
I've made sure to chown everything to 508:508.
My db/journal files are 300Mb in size (is that normal?)

Any suggestions?

mbentley, 2021-03-31:
This is nothing I've seen before - looks more like a software problem.  I'd check on the TP-Link forums (linked in the README).  Seems like something is being entered with a `/` in front of some IP range somewhere but I have no idea where that is coming from.

Walterlm1956, 2021-07-12:
Same issue here:

```
2021-07-12 12:53:40 [qtp1621002296-76] [WARN]-[SourceFile:151] - Failed to parse /10.253.0.1/32 [null], Exception: java.lang.NullPointerException
2021-07-12 12:54:47 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /10.253.0.1/32 [null], Exception: java.lang.NullPointerException
2021-07-12 12:56:39 [qtp1621002296-227] [WARN]-[SourceFile:151] - Failed to parse /10.253.0.1/32 [null], Exception: java.lang.NullPointerException
2021-07-12 12:58:28 [qtp1621002296-227] [WARN]-[SourceFile:151] - Failed to parse /10.253.0.1/32 [null], Exception: java.lang.NullPointerException
2021-07-12 12:59:47 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /10.253.0.1/32 [null], Exception: java.lang.NullPointerException
2021-07-12 13:04:08 [qtp1621002296-256] [WARN]-[SourceFile:151] - Failed to parse /10.253.0.1/32 [null], Exception: java.lang.NullPointerException
2021-07-12 13:09:47 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /10.253.0.1/32 [null], Exception: java.lang.NullPointerException
2021-07-12 13:14:47 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /10.253.0.1/32 [null], Exception: java.lang.NullPointerException
2021-07-12 13:19:47 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /10.253.0.1/32 [null], Exception: java.lang.NullPointerException
2021-07-12 13:19:47 [topology-task-0] [WARN]-[SourceFile:151] - Failed to parse /10.253.0.1/32 [null], Exception: java.lang.NullPointerException
```

mbentley, 2021-07-12:
Which version of Omada Controller are you running?  What image & tag?  The version will depend on the expected database size but I am running `mbentley/omada-controller:4.4` which is currently version `4.4.3`.  Also, could you provide how you're starting the container with your run command or compose file?  Here are the persistent data sizes for me:

```
# du -hs *
9.5K    cert
365M    data
3.4M    logs
12M     work
```

I just have two APs attached to my controller.  Not sure if the number of devices has an impact on the size of the data but this is what I am seeing for a long running controller (since July of 2020).

Initially, this looks like a software related problem which is something that you might need to report upstream through the [TP-Link forums](https://community.tp-link.com/en/business/forum/582) unless there is a specific way in which you're able to reproduce this through a setting where you either see it or don't.  I am using macvlan for my container networking, not sure if that has any impact but I do not see any `Failed to parse` errors in my logs but my guess could be that it is related to the bridge networking and how it is trying to discover devices maybe.

Walterlm1956, 2021-07-12:
So I'm running also latest tag: `mbentley/omada-controller:latest` and it is run on unraid so I've got 4 ports configured and running privileged in the host network.

[image]

mbentley, 2021-07-12:
I ran a container for a bit and couldn't get it to output any logs similar to what you're seeing:

```
# docker run -it --rm --privileged --net=host mbentley/omada-controller:latest
Unable to find image 'mbentley/omada-controller:latest' locally
latest: Pulling from mbentley/omada-controller
25fa05cd42bd: Pull complete
b15204a2880e: Pull complete
2184f3f87d02: Pull complete
82f25a0f361f: Pull complete
16d5d9963ca3: Pull complete
Digest: sha256:307f83366d1e33e7365082c73513fa2517e80c07e688ec3d92264318448c81e7
Status: Downloaded newer image for mbentley/omada-controller:latest
INFO: Time zone set to 'Etc/UTC'
INFO: Starting Omada Controller as user omada
tail: cannot open '/opt/tplink/EAPController/logs/server.log' for reading: No such file or directory
tail: '/opt/tplink/EAPController/logs/server.log' has appeared;  following new file
2021-07-12 13:58:37 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-07-12 13:58:37 [main] [INFO]-[SourceFile:91] - going to start local mongod.
2021-07-12 13:58:38 [main] [INFO]-[SourceFile:167] - mongodb process id is 42
2021-07-12 13:58:38 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2021-07-12 13:58:45 [main] [INFO]-[SourceFile:193] - User do not configure max.device, use defaultValue 1500.
2021-07-12 13:58:48 [main] [INFO]-[SourceFile:38] - monitor context initialing...
2021-07-12 13:58:48 [main] [INFO]-[SourceFile:193] - User do not configure max.device, use defaultValue 1500.
2021-07-12 13:58:48 [main] [INFO]-[SourceFile:193] - User do not configure max.device, use defaultValue 1500.
2021-07-12 13:58:49 [main] [INFO]-[SourceFile:463] - restart all customize tasks.
2021-07-12 13:58:49 [main] [INFO]-[SourceFile:471] - GeneralSetting not exist?may be quick setup unfinished. Skip Global Dst Job.
2021-07-12 13:58:49 [main] [INFO]-[SourceFile:73] - DisconnectRequestServer start
2021-07-12 13:58:50 [device-discovery-server] [INFO]-[SourceFile:59] - Start device discovery server
2021-07-12 13:58:50 [device-manage-server] [INFO]-[SourceFile:116] - Start device manage server
2021-07-12 13:58:50 [device-adopt-server] [INFO]-[SourceFile:87] - Start device adopt server
2021-07-12 13:58:51 [main] [INFO]-[SourceFile:152] - no need to compatible db.
2021-07-12 13:58:51 [main] [INFO]-[SourceFile:20] - Omada Controller started
2021-07-12 13:58:51 [site-event-bus-0] [INFO]-[SourceFile:107] - Success sending config to devices.
2021-07-12 13:58:51 [site-event-bus-0] [WARN]-[SourceFile:439] - restart hotspot task failed, site : Default, timeZone : null
2021-07-12 13:59:50 [scheduled-pool-8] [INFO]-[SourceFile:193] - User do not configure max.device, use defaultValue 1500.
2021-07-12 14:00:24 [scheduled-pool-1] [INFO]-[SourceFile:112] - Controller has not bound owner, cancel online checking Firmware tasks.
^C2021-07-12 14:23:42 [comm-pool-0] [WARN]-[SourceFile:142] - get InterruptedException while saving log.
2021-07-12 14:23:42 [Thread-1] [INFO]-[SourceFile:219] - Going to stop mongod which pid is 42
ShutdownHook: service stopped.
```

Just to make sure I am looking for the right thing, are these logs being outputted to the stdout/stderr of the container or just the log files?

mbentley, 2021-07-12:
I checked in both and do not see anything.  What sort of devices do you have managed by your controller?

Walterlm1956, 2021-07-12:
should be full log, i'm running a router but was looking to buy switch as well. but i'd like to fix this first

mbentley, 2021-07-12:
OK, I would suggest filing either a support case or create a forum post as it looks like a software issue to me.  I only have an AP so I can't test any other devices.

Walterlm1956, 2021-07-12:
Ok thanks anyway

closed 2021-10-16 by mbentley