mbentley/docker-omada-controller #96: Login Issue Docker/Container Station Qnap NAS

Securitet, 2021-04-07:
I was able to log in only once at the beginning, and only for a short period of time, since then after pushing the Log in button
I see a massage "Operation failed. Time out..."
Unexpectedly after pushing the Log in button like 20 times, I was able to log in again for a short period of time, and as you can see down below, container stopped working,
Console
2021-04-07 17:56:56 [qtp1241529534-106] [WARN]-[HttpChannel.java:597] - /theme/img/svg/NavigationBarIconMore.svg                                                                                                                                                                                                                                                           
javax.servlet.ServletException: org.apache.shiro.session.UnknownSessionException: There is no session with id [e89319a3-19f3-40e6-a448-397be618c1e6] 

Caused by: org.apache.shiro.session.UnknownSessionException: There is no session with id [e89319a3-19f3-40e6-a448-397be618c1e6] 
at org.apache.shiro.session.mgt.eis.AbstractSessionDAO.readSession(AbstractSessionDAO.java:170) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                                                      
        at org.apache.shiro.session.mgt.DefaultSessionManager.retrieveSessionFromDataSource(DefaultSessionManager.java:236) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                                  
        at org.apache.shiro.session.mgt.DefaultSessionManager.retrieveSession(DefaultSessionManager.java:222) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                                                
        at org.apache.shiro.session.mgt.AbstractValidatingSessionManager.doGetSession(AbstractValidatingSessionManager.java:118) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                             
        at org.apache.shiro.session.mgt.AbstractNativeSessionManager.lookupSession(AbstractNativeSessionManager.java:148) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                                    
        at org.apache.shiro.session.mgt.AbstractNativeSessionManager.lookupRequiredSession(AbstractNativeSessionManager.java:152) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                            
        at org.apache.shiro.session.mgt.AbstractNativeSessionManager.getAttribute(AbstractNativeSessionManager.java:249) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                                     
        at org.apache.shiro.session.mgt.DelegatingSession.getAttribute(DelegatingSession.java:141) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                                                           
        at org.apache.shiro.session.ProxiedSession.getAttribute(ProxiedSession.java:121) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                                                                     
        at org.apache.shiro.subject.support.DelegatingSubject.getRunAsPrincipalsStack(DelegatingSubject.java:473) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                                            
        at org.apache.shiro.subject.support.DelegatingSubject.getPrincipals(DelegatingSubject.java:157) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                                                      
        at org.apache.shiro.subject.support.DelegatingSubject.getPrincipal(DelegatingSubject.java:153) ~[shiro-core-1.4.0.jar:1.4.0]                                                                                                                                                                                                                                       
        at org.apache.shiro.web.servlet.ShiroHttpServletRequest.getSubjectPrincipal(ShiroHttpServletRequest.java:96) ~[shiro-web-1.4.0.jar:1.4.0]                                                                                                                                                                                                                          
        at org.apache.shiro.web.servlet.ShiroHttpServletRequest.getUserPrincipal(ShiroHttpServletRequest.java:112) ~[shiro-web-1.4.0.jar:1.4.0]                                                                                                                                                                                                                            
        at org.springframework.web.servlet.FrameworkServlet.getUsernameForRequest(FrameworkServlet.java:1159) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]                                                                                                                                                                                                             
        at org.springframework.web.servlet.FrameworkServlet.publishRequestHandledEvent(FrameworkServlet.java:1144) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]                                                                                                                                                                                                        
        at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1022) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]                                                                                                                                                                                                                    
        at org.springframework.web.servlet.FrameworkServlet.doGet(FrameworkServlet.java:897) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]                                                                                                                                                                                                                              
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:687) ~[javax.servlet-api-3.1.0.jar:3.1.0]                                                                                                                                                                                                                                                               
        at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:882) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]                                                                                                                                                                                                                            
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:790) ~[javax.servlet-api-3.1.0.jar:3.1.0]                                                                                                                       [... cut ...]                                                                                                          
^C2021-04-07 17:59:59 [comm-pool-0] [WARN]-[SourceFile:135] - get InterruptedException while saving log.                                                                                                                                                                                                                                                                   
ShutdownHook: service stopped.

mbentley, 2021-04-07:
I would say that sounds like an application issue.  I haven't seen anything like that related to it running in a container so your best bet is to report this on the [TP-Link Forums](https://community.tp-link.com/en/business/forum/582) and see if you can get traction with the team who has access to the code.

Securitet, 2021-04-07:
Thanks for the prompt respond. I'll try my luck then on TP-Link forum. In case they found a solution I'll post it here.

Aidan-OS, 2021-06-09:
I am sadly facing the same issue. Please do keep this ticket up to date, as I would like to know when I can actually start running my mesh properly.

closed 2021-10-16 by mbentley