mbentley/docker-omada-controller #97: Omada in Docker on Raspberry Pi 4 - MongoDB error "Got signal: 7 (Bus error)"

bbm-ro, 2021-04-08:
Hi,

I'm trying to install the latest version of the Omada Controller (4.2) using docker-compose on my Raspberry Pi. The container starts fine, but the interface is not accessible because the server is not starting, so nothing is listening on the management port.

When I check the logs, I see the following in the "server.log" file:

```
2021-04-08 14:25:35 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-04-08 14:25:35 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2021-04-08 14:27:16 [main] [WARN]-[SourceFile:151] - Fail to start mongo DB server
2021-04-08 14:27:16 [main] [WARN]-[SourceFile:78] - com.tplink.omada.start.a.a: Fail to start mongo DB server
com.tplink.omada.start.a.a: Fail to start mongo DB server
        at com.tplink.omada.start.b.d.a(SourceFile:152) ~[omada-start.jar:?]
        at com.tplink.omada.start.b.b.e(SourceFile:75) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.d(SourceFile:223) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.q(SourceFile:287) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.a(SourceFile:100) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.e(SourceFile:245) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaLinuxMain.b(SourceFile:80) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaLinuxMain.main(SourceFile:25) [omada-start.jar:?]
2021-04-08 14:28:58 [main] [WARN]-[SourceFile:151] - Fail to start mongo DB server
2021-04-08 14:28:58 [main] [WARN]-[SourceFile:78] - com.tplink.omada.start.a.a: Fail to start mongo DB server
com.tplink.omada.start.a.a: Fail to start mongo DB server
        at com.tplink.omada.start.b.d.a(SourceFile:152) ~[omada-start.jar:?]
        at com.tplink.omada.start.b.b.e(SourceFile:75) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.d(SourceFile:223) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.n(SourceFile:216) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.q(SourceFile:290) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.a(SourceFile:100) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.e(SourceFile:245) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaLinuxMain.b(SourceFile:80) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaLinuxMain.main(SourceFile:25) [omada-start.jar:?]
2021-04-08 14:29:00 [mongo-check-thread] [INFO]-[SourceFile:312] - mongod was shutdown, restarting it
2021-04-08 14:29:59 [main] [WARN]-[SourceFile:342] - Mongo Client connect error while init Jetty.com.mongodb.MongoTimeoutException: Timed out after 30000 ms while waiting to connect. Client view of cluster state is {type=UNKNOWN, servers=[{address=127.0.0.1:27217, type=UNKNOWN, state=CONNECTING, exception={com.mongodb.MongoSocketOpenException: Exception opening socket}, caused by {java.net.ConnectException: Connection refused (Connection refused)}}]
```

In the "mongod.log" file I have the the following errors, repeated indefinitely (the reported address is changing):

```
2021-04-08T14:25:36.040+0300 SEVERE: Invalid access at address: 0x489930b
2021-04-08T14:25:36.045+0300 SEVERE: Got signal: 7 (Bus error).
Backtrace:0x7639be 0x762f6e 0x76300a 0xf716e270 0x75ece4 0x75cd7e 0x75cdf4 0x1f0300 0x1fc81a 0x1fcb7e 0x1fcdc8 0x1c4e64 0xf715f8aa
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo15printStackTraceERSo+0x25) [0x7639be]
 /opt/tplink/EAPController/bin/mongod() [0x762f6e]
 /opt/tplink/EAPController/bin/mongod() [0x76300a]
 /lib/arm-linux-gnueabihf/libc.so.6(+0x25270) [0xf716e270]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11ProcessInfo10SystemInfo17collectSystemInfoEv+0x87f) [0x75ece4]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11ProcessInfo20initializeSystemInfoEv+0x7d) [0x75cd7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo36_mongoInitializerFunction_SystemInfoEPNS_18InitializerContextE+0x7) [0x75cdf4]
 /opt/tplink/EAPController/bin/mongod(_ZN5boost6detail8function17function_invoker1IPFN5mongo6StatusEPNS3_18InitializerContextEES4_S6_E6invokeERNS1_15function_bufferES6_+0x17) [0x1f0300]
 /opt/tplink/EAPController/bin/mongod(_ZNK5mongo11Initializer7executeERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EERKSt3mapIS7_S7_St4lessIS7_ESaISt4pairIKS7_S7_EEE+0x15d) [0x1fc81a]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo21runGlobalInitializersERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS6_EERKSt3mapIS6_S6_St4lessIS6_ESaISt4pairIKS6_S6_EEE+0x29) [0x1fcb7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo21runGlobalInitializersEiPKPKcS3_+0x22b) [0x1fcdc8]
 /opt/tplink/EAPController/bin/mongod(main+0x6f) [0x1c4e64]
 /lib/arm-linux-gnueabihf/libc.so.6(__libc_start_main+0x9d) [0xf715f8aa]
2021-04-08T14:27:18.300+0300 ***** SERVER RESTARTED *****
2021-04-08T14:27:18.302+0300 SEVERE: Invalid access at address: 0x47b130b
2021-04-08T14:27:18.308+0300 SEVERE: Got signal: 7 (Bus error).
Backtrace:0x7639be 0x762f6e 0x76300a 0xf7129270 0x75ece4 0x75cd7e 0x75cdf4 0x1f0300 0x1fc81a 0x1fcb7e 0x1fcdc8 0x1c4e64 0xf711a8aa
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo15printStackTraceERSo+0x25) [0x7639be]
 /opt/tplink/EAPController/bin/mongod() [0x762f6e]
 /opt/tplink/EAPController/bin/mongod() [0x76300a]
 /lib/arm-linux-gnueabihf/libc.so.6(+0x25270) [0xf7129270]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11ProcessInfo10SystemInfo17collectSystemInfoEv+0x87f) [0x75ece4]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11ProcessInfo20initializeSystemInfoEv+0x7d) [0x75cd7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo36_mongoInitializerFunction_SystemInfoEPNS_18InitializerContextE+0x7) [0x75cdf4]
 /opt/tplink/EAPController/bin/mongod(_ZN5boost6detail8function17function_invoker1IPFN5mongo6StatusEPNS3_18InitializerContextEES4_S6_E6invokeERNS1_15function_bufferES6_+0x17) [0x1f0300]
 /opt/tplink/EAPController/bin/mongod(_ZNK5mongo11Initializer7executeERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EERKSt3mapIS7_S7_St4lessIS7_ESaISt4pairIKS7_S7_EEE+0x15d) [0x1fc81a]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo21runGlobalInitializersERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS6_EERKSt3mapIS6_S6_St4lessIS6_ESaISt4pairIKS6_S6_EEE+0x29) [0x1fcb7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo21runGlobalInitializersEiPKPKcS3_+0x22b) [0x1fcdc8]
[... 74 lines cut ...]
      - TZ=Europe/Bucharest
    volumes:
      - './omada/omada-data:/opt/tplink/EAPController/data'
      - './omada/omada-work:/opt/tplink/EAPController/work'
      - './omada/omada-logs:/opt/tplink/EAPController/logs'
    ports:
      - 8088:8088
      - 8043:8043
      - 8843:8843
      - 29810:29810
      - 29810:29810/udp
      - 29811:29811
      - 29811:29811/udp
      - 29812:29812
      - 29812:29812/udp
      - 29813:29813
      - 29813:29813/udp
    restart: unless-stopped
```

Can you please tell me what I'm doing wrong and how to fix this?

I tried using the "arm64" version, but it doesn't even start the container. Only the "armv7l" version seems to start.

Thank you.

mbentley, 2021-04-08:
I might have to try to get one of the several old raspberry pis I have sitting around running Raspbian to check this out.  What happens when you try to run the arm64 image?  Any errors or other messages?  I would expect that it should work if architecture is showing as `aarch64`.  I have a Raspberry Pi 3 that I run Debian on with `aarch64` so I can also try on that.

bbm-ro, 2021-04-08:
Hi Matt,

Thank for the quick response!
When I try to use the "_mbentley/omada-controller:latest-arm64_" image the container is only showing up as "Restarting" and not "Up (starting)", there are no folders created under the "omada" directory and in my /var/log/syslog file I see the following messages (repeated periodically until I stop the container):

```
Apr  8 16:35:53 RPi4 systemd[1]: run-docker-runtime\x2drunc-moby-63f306cfcac692dd215ddec0fed59a0bc04f3399870868a7466c1c352e1d4c2a-runc.IDcARr.mount: Succeeded.
Apr  8 16:35:53 RPi4 systemd[778]: run-docker-runtime\x2drunc-moby-63f306cfcac692dd215ddec0fed59a0bc04f3399870868a7466c1c352e1d4c2a-runc.IDcARr.mount: Succeeded.
Apr  8 16:35:53 RPi4 systemd[3432]: run-docker-runtime\x2drunc-moby-63f306cfcac692dd215ddec0fed59a0bc04f3399870868a7466c1c352e1d4c2a-runc.IDcARr.mount: Succeeded.
Apr  8 16:36:02 RPi4 containerd[559]: time="2021-04-08T16:36:02.665114780+03:00" level=info msg="starting signal loop" namespace=moby path=/run/containerd/io.containerd.runtime.v2.task/moby/6fb35194b790ce304f3b0c615d1595820cb4ff2cb53a526ba7b87635c3c3571c pid=10597
Apr  8 16:36:02 RPi4 systemd[1]: run-docker-runtime\x2drunc-moby-6fb35194b790ce304f3b0c615d1595820cb4ff2cb53a526ba7b87635c3c3571c-runc.6v4wEX.mount: Succeeded.
Apr  8 16:36:02 RPi4 systemd[3432]: run-docker-runtime\x2drunc-moby-6fb35194b790ce304f3b0c615d1595820cb4ff2cb53a526ba7b87635c3c3571c-runc.6v4wEX.mount: Succeeded.
Apr  8 16:36:02 RPi4 systemd[778]: run-docker-runtime\x2drunc-moby-6fb35194b790ce304f3b0c615d1595820cb4ff2cb53a526ba7b87635c3c3571c-runc.6v4wEX.mount: Succeeded.
Apr  8 16:36:03 RPi4 dockerd[1095]: time="2021-04-08T16:36:03.690060974+03:00" level=info msg="ignoring event" container=6fb35194b790ce304f3b0c615d1595820cb4ff2cb53a526ba7b87635c3c3571c module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
Apr  8 16:36:03 RPi4 containerd[559]: time="2021-04-08T16:36:03.693315298+03:00" level=info msg="shim disconnected" id=6fb35194b790ce304f3b0c615d1595820cb4ff2cb53a526ba7b87635c3c3571c
Apr  8 16:36:03 RPi4 containerd[559]: time="2021-04-08T16:36:03.693547333+03:00" level=error msg="copy shim log" error="read /proc/self/fd/16: file already closed"
Apr  8 16:36:05 RPi4 docker-compose[10258]: #033[0m#033[33;1momada-controller exited with code 159
```

Meanwhile, I tried running the same docker-compose configuration as in my original message on a Raspberry Pi 3 Model B Rev 1.2, which has the same docker and docker-compose versions. It worked without any issues, so it looks like something is not correct with the RPi 4 with 64-bit kernel.

Thank you.

mjsabby, 2021-04-10:
Using OpenJDK 11 fixes the problem for me.

bbm-ro, 2021-04-10:
Hi Mukul,

Thanks for the suggestion, but it doesn't seem to work for me.
I installed the openjdk-11 packages on the Raspberry Pi 4 and when I start the container (amr-v7l image) I get the same errors in the MongoDB log:

```
2021-04-10T16:21:01.630+0300 ***** SERVER RESTARTED *****
2021-04-10T16:21:01.632+0300 SEVERE: Invalid access at address: 0x3efb30b
2021-04-10T16:21:01.637+0300 SEVERE: Got signal: 7 (Bus error).
Backtrace:0x7639be 0x762f6e 0x76300a 0xf731f270 0x75ece4 0x75cd7e 0x75cdf4 0x1f0300 0x1fc81a 0x1fcb7e 0x1fcdc8 0x1c4e64 0xf73108aa
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo15printStackTraceERSo+0x25) [0x7639be]
 /opt/tplink/EAPController/bin/mongod() [0x762f6e]
 /opt/tplink/EAPController/bin/mongod() [0x76300a]
 /lib/arm-linux-gnueabihf/libc.so.6(+0x25270) [0xf731f270]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11ProcessInfo10SystemInfo17collectSystemInfoEv+0x87f) [0x75ece4]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11ProcessInfo20initializeSystemInfoEv+0x7d) [0x75cd7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo36_mongoInitializerFunction_SystemInfoEPNS_18InitializerContextE+0x7) [0x75cdf4]
 /opt/tplink/EAPController/bin/mongod(_ZN5boost6detail8function17function_invoker1IPFN5mongo6StatusEPNS3_18InitializerContextEES4_S6_E6invokeERNS1_15function_bufferES6_+0x17) [0x1f0300]
 /opt/tplink/EAPController/bin/mongod(_ZNK5mongo11Initializer7executeERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EERKSt3mapIS7_S7_St4lessIS7_ESaISt4pairIKS7_S7_EEE+0x15d) [0x1fc81a]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo21runGlobalInitializersERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS6_EERKSt3mapIS6_S6_St4lessIS6_ESaISt4pairIKS6_S6_EEE+0x29) [0x1fcb7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo21runGlobalInitializersEiPKPKcS3_+0x22b) [0x1fcdc8]
 /opt/tplink/EAPController/bin/mongod(main+0x6f) [0x1c4e64]
 /lib/arm-linux-gnueabihf/libc.so.6(__libc_start_main+0x9d) [0xf73108aa]
```

Thank you.

mbentley, 2021-04-12:
Could you provide the output from `docker logs <container>` when trying to run the arm64 version?  I am pretty sure that the problem with mongodb not stating is because of the different architecture.

bbm-ro, 2021-04-12:
Hi Matt,

Here's what I get at the terminal when starting the container:

```
root@RPi4:/containers# docker-compose up -d omada
Pulling omada (mbentley/omada-controller:latest-arm64)...
latest-arm64: Pulling from mbentley/omada-controller
af9ca2923c1e: Pull complete
9a2babfbd9d0: Pull complete
f83b7f31d752: Pull complete
8c84f46dac3d: Pull complete
ec006aad55fd: Pull complete
2f6237343307: Pull complete
4f4fb700ef54: Pull complete
Digest: sha256:a4528fb6e2c8614e56ccb664b1e28b76411c68036e6a0006325cf4e82504cfde
Status: Downloaded newer image for mbentley/omada-controller:latest-arm64
Recreating omada-controller ... done
root@RPi4:/containers# docker-compose ps
      Name                    Command                  State                                                        Ports
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
omada-controller   /entrypoint.sh /usr/bin/ja ...   Restarting
pihole             /s6-init                         Up (healthy)   0.0.0.0:443->443/tcp, 0.0.0.0:53->53/tcp, 0.0.0.0:53->53/udp, 0.0.0.0:67->67/udp, 0.0.0.0:81->80/tcp
rutorrent          /init                            Up             443/tcp, 0.0.0.0:5000->5000/tcp, 0.0.0.0:51413->51413/tcp, 0.0.0.0:6881->6881/udp, 0.0.0.0:9080->80/tcp
```

When running ```docker logs``` or ```docker-compose``` logs for this container, I get nothing:

```
root@RPi4:/containers# docker logs omada-controller
root@RPi4:/containers# docker-compose logs omada
Attaching to omada-controller
root@RPi4:/containers#
```
As I previously mentioned, the logs inside the containers (_server.log_ and _mongod.log_) are not even created when using the _arm64_ image, as if it's not actually starting.

I get the same errors in /var/log/messages, as the container keeps restarting:
```
Apr 12 14:32:09 RPi4 dockerd[928]: time="2021-04-12T14:32:09.802497344+03:00" level=info msg="ignoring event" container=99a1471bfccd133447d96a5389335445c28726fef89b12c828dc4396747af51e module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
Apr 12 14:32:09 RPi4 containerd[567]: time="2021-04-12T14:32:09.803376485+03:00" level=info msg="shim disconnected" id=99a1471bfccd133447d96a5389335445c28726fef89b12c828dc4396747af51e
Apr 12 14:32:09 RPi4 containerd[567]: time="2021-04-12T14:32:09.803612742+03:00" level=error msg="copy shim log" error="read /proc/self/fd/17: file already closed"
...
Apr 12 14:32:11 RPi4 docker-compose[7163]: #033[0m#033[33;1momada-controller exited with code 159
```

Thank you.

mbentley, 2021-06-15:
Sorry, I totally forgot about this issue.  Were you able to figure out what was going on by chance?

bbm-ro, 2021-06-16:
Hi,
Unfortunately I haven't been able to make it work, so I just used the container on the Raspberry Pi 3, where it worked without issues.
If you still want to debug this further, let me know and I can try it again on the Pi4.

Thank you.

smartins, 2021-07-05:
Getting the exact same error on a Pi 3 Model B+

```
pi@pi-hole:~ $ lscpu
Architecture:          armv7l
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    1
Core(s) per socket:    4
Socket(s):             1
Model:                 4
Model name:            ARMv7 Processor rev 4 (v7l)
```

```
pi@pi-hole:~ $ docker logs omada-controller
INFO: Time zone set to 'Etc/UTC'
INFO: Starting Omada Controller as user omada
2021-07-05 12:13:43 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-07-05 12:13:44 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2021-07-05 12:15:24 [main] [WARN]-[SourceFile:151] - Fail to start mongo DB server
2021-07-05 12:15:24 [main] [WARN]-[SourceFile:78] - com.tplink.omada.start.a.a: Fail to start mongo DB server
com.tplink.omada.start.a.a: Fail to start mongo DB server
        at com.tplink.omada.start.b.d.a(SourceFile:152) ~[omada-start.jar:?]
        at com.tplink.omada.start.b.b.e(SourceFile:75) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.d(SourceFile:223) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.q(SourceFile:287) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.a(SourceFile:100) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaBootstrap.e(SourceFile:245) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaLinuxMain.b(SourceFile:80) [omada-start.jar:?]
        at com.tplink.omada.start.OmadaLinuxMain.main(SourceFile:25) [omada-start.jar:?]
```

mbentley, 2021-07-07:
OK, I finally was able to get some time to get a Raspberry Pi 3b+ with raspbian installed to try to reproduce this.  Will report back if I have any news on this.

smartins, 2021-07-09:
Thanks for the time taking to trying to figure out this @mbentley 

I turned on the mongodb logs to see if I could figure out the issue. Here's my docker command:

```
docker run -d \
  --name omada-controller \
  --restart unless-stopped \
  --net host \
  -e MANAGE_HTTP_PORT=8088 \
  -e MANAGE_HTTPS_PORT=8043 \
  -e PORTAL_HTTP_PORT=8088 \
  -e PORTAL_HTTPS_PORT=8843 \
  -e SHOW_SERVER_LOGS=true \
  -e SHOW_MONGODB_LOGS=true \
  -e SSL_CERT_NAME="tls.crt" \
  -e SSL_KEY_NAME="tls.key" \
  -e TZ=Etc/UTC \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:latest-armv7l
```

And here's the log, seems mogo is running out of memory: `ERROR: mmap private failed with out of memory. You are using a 32-bit build and probably need to upgrade to 64`

```
INFO: Time zone set to 'Etc/UTC'
INFO: Starting Omada Controller as user omada
tail: '/opt/tplink/EAPController/logs/server.log' has become inaccessible: No such file or directory
tail: '/opt/tplink/EAPController/logs/server.log' has appeared;  following new file
2021-07-09 14:07:45 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-07-09 14:07:46 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2021-07-09T14:07:47.011+0000 ***** SERVER RESTARTED *****
2021-07-09T14:07:47.046+0000 [initandlisten] MongoDB starting : pid=38 port=27217 dbpath=../data/db 32-bit host=pi-hole
2021-07-09T14:07:47.046+0000 [initandlisten]
2021-07-09T14:07:47.046+0000 [initandlisten] ** NOTE: This is a 32 bit MongoDB binary.
2021-07-09T14:07:47.046+0000 [initandlisten] **       32 bit builds are limited to less than 2GB of data (or less with --journal).
2021-07-09T14:07:47.047+0000 [initandlisten] **       See http://dochub.mongodb.org/core/32bit
2021-07-09T14:07:47.047+0000 [initandlisten]
2021-07-09T14:07:47.047+0000 [initandlisten] db version v2.6.10
2021-07-09T14:07:47.047+0000 [initandlisten] git version: nogitversion
2021-07-09T14:07:47.047+0000 [initandlisten] OpenSSL version: OpenSSL 1.0.2g  1 Mar 2016
2021-07-09T14:07:47.047+0000 [initandlisten] build info: Linux kishi10 3.2.0-84-highbank #121-Ubuntu SMP PREEMPT Tue May 5 19:44:15 UTC 2015 armv7l BOOST_LIB_VERSION=1_58
2021-07-09T14:07:47.047+0000 [initandlisten] allocator: tcmalloc
2021-07-09T14:07:47.048+0000 [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db", journal: { enabled: true } }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2021-07-09T14:07:47.066+0000 [initandlisten] journal dir=../data/db/journal
2021-07-09T14:07:47.067+0000 [initandlisten] recover : no journal files present, no recovery needed
2021-07-09T14:07:47.312+0000 [initandlisten] ERROR: mmap private failed with out of memory. You are using a 32-bit build and probably need to upgrade to 64
2021-07-09T14:07:47.315+0000 [initandlisten] Assertion: 13636:file /opt/tplink/EAPController/lib/../data/db/omada_data.3 open/create failed in createPrivateMap (look in log for more information)

```

eaconner, 2021-08-25:
I've been having the same issue going from 4.4.3 to 4.4.4 and clearing the "/data/logs" folder solved the issue for me.