mbentley/docker-omada-controller #101: Unable to create container @3.2 ?

proap, 2021-04-20:
**Describe the bug**
Unable to create container for the 3.2 version with arm64.
With the 32bits version, there is no problem.

**To Reproduce**
Raspberry pi 4 8gb
Ubuntu server 20.04.2 (64bits)
Docker version 19.03.8, build afacb8b7f0

**Expected behavior**

**How you're launching your container**
with docker-compose:
```
#  omada:
#    image: mbentley/omada-controller:3.2-arm64
#    container_name: omada
#    restart: unless-stopped
#    volumes:
#      - /home/andrep/docker/omada/omada-data:/opt/tplink/EAPController/data
#      - /home/andrep/docker/omada/omada-work:/opt/tplink/EAPController/work
#      - /home/andrep/docker/omada/omada-logs:/opt/tplink/EAPController/logs
#      - omada-data:/opt/tplink/EAPController/data
#      - omada-work:/opt/tplink/EAPController/work
#      - omada-logs:/opt/tplink/EAPController/logs
#    environment:
#      - SMALL_FILES=true
#      - TZ=Europe/Paris
#      - MANAGE_HTTP_PORT=8088
#      - MANAGE_HTTPS_PORT=8043
#      - PORTAL_HTTP_PORT=8088
#      - PORTAL_HTTPS_PORT=8843
#      - SHOW_SERVER_LOGS=true
#      - SHOW_MONGODB_LOGS=false
#      - SSL_CERT_NAME="tls.crt"
#      - SSL_KEY_NAME="tls.key"
#    ports:
#      - 8088:8088
#      - 8043:8043
#      - 8843:8843
#      - 29810:29810
#      - 29810:29810/udp
#      - 29811:29811
#      - 29811:29811/udp
#      - 29812:29812
#      - 29812:29812/udp
#      - 29813:29813
#      - 29813:29813/udp
#    networks:
#      - omada
```

or with docker run (which used to work a few months ago)
```
docker run -d \
  --name omada \
  --network omada \
  --restart unless-stopped \
  -e TZ=Etc/Paris \
  -e SMALL_FILES=true \
  -p 8088:8088 \
  -p 8043:8043 \
  -p 27001:27001/udp \
  -p 27002:27002 \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29812:29812 \
  -p 29813:29813 \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:3.2-arm64
```
**Container Logs**
```
ERROR: for omada  Cannot start service omada: OCI runtime create failed: container_linux.go:349: starting container process caused "exec: \"/entrypoint.sh\": stat /entrypoint.sh: no such file or directory": unknown
```

mbentley, 2021-04-20:
What does `uname -a` show?

proap, 2021-04-20:
Linux pi3 5.4.0-1034-raspi #37-Ubuntu SMP PREEMPT Mon Apr 12 23:14:49 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

mbentley, 2021-04-20:
OK, very strange.  The entrypoint script is missing.  Investigating.

mbentley, 2021-04-20:
Try to pull the image again and re-run it and see if it is good now.

proap, 2021-04-20:
That was it!

Thanks for fixing it so fast :+1:

closed 2021-04-20 by proap

mbentley, 2021-04-20:
Good deal.  I am not sure what went wrong exactly.  Seems like something screwed up the manifest for the image as I literally changed nothing.  I've been experimenting with `docker buildx` caching to optimize the builds as doing cross platform builds on amd64 based hardware and the builds used take like 5 minutes each but with buildx caching, I can make the builds happen in 3 seconds as long as nothing has changed (which is most of the time).  Just retriggering the build which re-writes the manifests to Docker Hub is all that changed.