mbentley/docker-omada-controller #104: SMTP Issue with gmail being unable to connect to host

aaronnad, 2021-05-08:
**Describe the bug**
Hi,

I'm trying to configure the SMTP settings for the controller - however i get failure to connect to SMTP Host.

**To Reproduce**
Steps to reproduce the behavior:

1. Configure settings as seen here
[image]

**Expected behavior**
I expect when pressing test - to receive an email on the configured email

**How you're launching your container**
Docker-compose.yml attached (had to use .txt format)
[docker-compose.txt](https://github.com/mbentley/docker-omada-controller/files/6446680/docker-compose.txt)

**Container Logs**

```
{"log":"DEBUG: setDebug: JavaMail version 1.4.7\n","stream":"stdout","time":"2021-05-08T22:44:26.243074071Z"}
{"log":"DEBUG: getProvider() returning javax.mail.Provider[TRANSPORT,smtps,com.sun.mail.smtp.SMTPSSLTransport,Oracle]\n","stream":"stdout","time":"2021-05-08T22:44:26.243371819Z"}
{"log":"DEBUG SMTP: useEhlo true, useAuth true\n","stream":"stdout","time":"2021-05-08T22:44:26.244022067Z"}
{"log":"DEBUG SMTP: trying to connect to host \"smtp.gmail.com\", port 465, isSSL true\n","stream":"stdout","time":"2021-05-08T22:44:26.244077161Z"}
{"log":"2021-05-08 23:44:26 [email-task-3] [INFO]-[SourceFile:101] - Send mail exception Could not connect to SMTP host: smtp.gmail.com, port: 465\n","stream":"stdout","time":"2021-05-08T22:44:27.200181693Z"}
```

```
2021-05-08 23:41:43 [email-task-2] [INFO]-[SourceFile:101] - Send mail exception Could not connect to SMTP host: smtp.gmail.com, port: 465
```

**Additional context**
I'm running this container using portainer - but i havent had any issues with connectivity from other containers.

I have an app password setup and this is currently reachable on my old omada instance using the exact same settings as the new SDN config.

aaronnad, 2021-05-08:
My portainer instance is able to connect to the SMTP host 
Connection to smtp.gmail.com 465 port [tcp/submissions] succeeded!

aaronnad, 2021-05-08:
From within my container i see

```
root@e241380c6f70:/opt/tplink/EAPController/lib# nc -v smtp.gmail.com 465
DNS fwd/rev mismatch: smtp.gmail.com != wq-in-f108.1e100.net
smtp.gmail.com [74.125.140.108] 465 (?) open
```

I can ping the destination too

```
root@e241380c6f70:/opt/tplink/EAPController/lib# ping smtp.gmail.com
PING smtp.gmail.com (74.125.140.108) 56(84) bytes of data.
64 bytes from wq-in-f108.1e100.net (74.125.140.108): icmp_seq=1 ttl=104 time=18.3 ms
64 bytes from wq-in-f108.1e100.net (74.125.140.108): icmp_seq=2 ttl=104 time=18.1 ms
64 bytes from wq-in-f108.1e100.net (74.125.140.108): icmp_seq=3 ttl=104 time=18.3 ms
64 bytes from wq-in-f108.1e100.net (74.125.140.108): icmp_seq=4 ttl=104 time=21.0 ms
64 bytes from wq-in-f108.1e100.net (74.125.140.108): icmp_seq=5 ttl=104 time=18.4 ms
64 bytes from wq-in-f108.1e100.net (74.125.140.108): icmp_seq=6 ttl=104 time=16.5 ms
```

mbentley, 2021-05-10:
Someone else had problems in #82.  It's probably something with the settings as I find them to be a bit odd in the UI.

I would suggest trying using StartTLS by unchecking the box for SSL and using 587 to see if that helps ([gmail outgoing settings doc](https://support.google.com/mail/answer/7126229?hl=en)).  If it still doesn't work, I would suggest taking it up on the [forums](https://community.tp-link.com/en/business/forum/582?page=1) as this is not likely to be an issue with the container image.

aaronnad, 2021-06-18:
Hi @mbentley, I followed through and asked if on the forums. It looks to be an issue with the new Java8 instance - see here https://community.tp-link.com/en/business/forum/topic/267794?replyId=606514

Basically Java8 disabled TLS1.0 and TLS1.1 which is what the controller uses to connect out on for the SMTP access.

They’ve given a workaround and are looking to resolve it in later a later version.

Cheers

mbentley, 2021-06-18:
Ah excellent.  Thanks for the follow up on the workaround.  Would it be helpful if I were to create an environment variable to provide the workaround so that hackery didn't need to be done on the image?

aaronnad, 2021-06-18:
Oh wow. That would be excellent if you could do. I was about to edit the image manually, but if you can do a variable that’d be excellent! Would make it easier down the line.

Thank you

mbentley, 2021-06-18:
Once the builds complete, there should now be an `-e TLS_1_11_ENABLED=true` option you can use.  It will output a notice that it is re-enabling those TLS versions.

As you can see, here is the updated file when it is running:

```
# grep ^jdk.tls.disabledAlgorithms /etc/java-8-openjdk/security/java.security
jdk.tls.disabledAlgorithms=SSLv3, RC4, DES, MD5withRSA, \
```

I do not have a quick way to test to verify it is definitely working so if you could, test it when you get a chance and let me know.  Thanks!

mbentley, 2021-06-18:
The image should be on Docker Hub now.

closed 2021-06-18 by mbentley

aaronnad, 2021-06-18:
Just got round to testing this and it does indeed work now
[image]

kipusoep, 2021-07-30:
I noticed today that I had the same issue, adding the env var helped :-)

commit d5e6d818 (2021-06-18): Update entrypoints; added TLS 1/1.1 option; fixes #104
commit d5637e3f (2021-06-18): Merge pull request #113 from mbentley/add-ssl-override

Update entrypoints; added TLS 1/1.1 option; fixes #104