mbentley/docker-omada-controller #105: I'm not a programmer can you help

feq451, 2021-05-21:
I currently run omada on a raspberry pi at version 3.2.1.

I want to upgrade to the newest controller and all the faqs I've looked at tell me to use docker and point towards your github.

I can easily figure out how to install docker on my raspberry pi but I'm kind of unclear on what code I actually need to run to install the controller.

 I'm just looking for a straightforward unchanged basic install. 

Any chance you could provide a simple step by step

Thanks
Adam

mbentley, 2021-05-22:
Have you checked out the instructions found here: https://github.com/mbentley/docker-omada-controller#upgrading-to-41

If so, what questions do you have?

feq451, 2021-05-22:
Thanks. I saw I guess I don't understand docker. Do I just need to install docker and then run

docker pull mbentley/omada-controller

OranguTech, 2021-05-24:
@feq451  - You'll need to get familiar w/ basics of docker.

The pull command will grab the latest version of the image, you may want to specify the version.

See https://github.com/mbentley/docker-omada-controller#example-usage

feq451, 2021-05-31:
so i got the pull command and updated to the latest but then I get this screen shot
[image]

I am stuck

mbentley, 2021-06-01:
You still have a container with the old name.  `docker ps -a` will show that.  You either need to remove the previous container (`docker rm omada-controller`) or rename it (`docker rename omada-controller omada-controller-old`).  If you want to be able to switch back to the previous one, I would suggest renaming and then removing when you're sure you no longer need it.  There shouldn't be anything from the old container that is needing to be kept as all data will persist outside of the container in the Docker managed named volumes.

feq451, 2021-06-01:
Thank you. I don't know exactly what I did or didn't do correctly.  Ended up using Portainer also/instead but finally got it working.  I'm still honestly not sure what I did but it works

OranguTech, 2021-06-07:
You probably renamed or deleted/overwrote your older existing container,
which again is mostly fine, because the saved data/config you care about is
stored in a different,  decoupled volume/container/directory/thing.

closed 2021-10-16 by mbentley