mbentley/docker-omada-controller #106: Question/documentation on /opt/tplink/EAPController/work

qdm12, 2021-05-22:
Hi there,

First of all thank you for this Docker image.

Sorry to pollute a bit the issues, this is not a bug, but more of a question and documentation thing.

I'm wondering why we need to bind mount or persist `/opt/tplink/EAPController/work`, especially if the database is in `/opt/tplink/EAPController/data` 🤔? Maybe adding a line to the readme describing why would be nice 😉

It also seems the container removes everything from that work directory on shutdown, so why not just leave it not bind mounted and not persisted? 

Thanks!

mbentley, 2021-05-23:
It's more of a personal preference thing. If a container is writing data to a path, it is more performant to use some sort of volume instead of the copy on wrote file system.  Most machines wouldn't see much of a difference but something like a raspberry pi could.

qdm12, 2021-05-23:
Ah interesting, thanks ! Maybe have it as a Docker volume `omada_work:/opt/.../work` instead? Or you could declare it as a volume in the Dockerfile to have it as an anonymous volume by default?

mbentley, 2021-05-24:
Yes, that is certainly possible to do with a Docker managed volume or anonymous to achieve the same without needing a path.  Either way would work.  With the path already being [declared as a volume](https://github.com/mbentley/docker-omada-controller/blob/master/Dockerfile.v4.1.x#L43), it should be an anonymous volume by default anyway.

qdm12, 2021-05-25:
Nice! Maybe removing it from the Docker run commands but precising you can bind mount it if you want would be better?

As in, it just confused me for a few minutes, but maybe that's just my odd brain too! Let's close the issue for now thanks!

closed 2021-05-25 by qdm12

qdm12, 2021-05-25:
Great thanks! 👍

commit 2c8caa32 (2021-05-25): Updated readme; fixes #106
commit fce413dd (2021-05-25): Merge pull request #107 from mbentley/update-readme

Updated readme; fixes #106