mbentley/docker-omada-controller #109: Clamp container mem 512m/1g [arm7]

mlarivie, 2021-05-26:
**Describe the bug**
When the default container image is used, the memory allocation is 16g. This should be set in the dockerfile (I think, I'm still picking things up)

What I did is use the podman run -m 1g directive to workaround this and the container has stopped rebooting

We also need to modify the java xms heap to 256m rather than 1g. We're not running vCenter, just a WLC.

**To Reproduce**
This is my build-launch file:
```
[root@podman launch]# cat /opt/launch/omada-wlc.sh
#/bin/bash
systemctl disable --now omada-wlc-container.service
logger -t omada-wlc-container "Container service disabled for container rebuild. [1/4]"
podman rm omada-wlc
logger -t omada-wlc-container "Pulling latest container image from docker.io/mbentley .. [2/4]"
podman pull mbentley/omada-controller:latest-armv7l
## We need to clamp memory to 512m or 1g - Since nothing else is running lets go with 1g
podman run -d --name omada-wlc --restart on-failure:10 --net host -m 1g -e TZ=America/New_York -v /opt/omada-wlc/data:/opt/tplink/EAPController/data -v /opt/omada-wlc/work:/opt/tplink/EAPController/work -v /opt/omada-wlc/logs:/opt/tplink/EAPController/logs mbentley/omada-controller:latest-armv7l
podman kill omada-wlc
sleep 6
logger -t "Container rebuilt and updated [3/4]"
systemctl enable --now omada-wlc-container.service
logger -t omada-wlc-container "Container service re-enabled and launched. [1/4]"
```
This is my systemd container launch file:
```
[root@podman launch]# cat /etc/systemd/system/omada-wlc-container.service
[Unit]
Description=TP-Link Omada Controller
Requires=NetworkManager-wait-online.service
#Wants=podman.service
After=NetworkManager-wait-online.service

[Service]
Restart=always
ExecStart=/usr/bin/podman start -a omada-wlc
ExecStop=/usr/bin/podman stop -t 90 omada-wlc
[Install]
WantedBy=multi-user.target
```

**Expected behavior**

Somethings still fishy... output of `podman stats`
```
ID            NAME       CPU %   MEM USAGE / LIMIT  MEM %      NET IO   BLOCK IO         PIDS
1887db544b5e  omada-wlc  19.47%  457.7MB / 508.8kB  89955.45%  -- / --  12.71MB / 287MB  126
```
Cockpit, however shows that mem is clamped to 1g. 

Sounds like resource definitions need some tweaking but I've not done this before.

**How you're launching your container**
not applicable

**Container Logs**
not applicable

**Additional context**
not applicable

mlarivie, 2021-05-26:
Hardware platform is odroid-hc1 on fedora-33
`podman.lariv.lan 5.12.6-200.fc33.armv7hl #1 SMP Sat May 22 21:29:19 UTC 2021 armv7l armv7l armv7l GNU/Linux`

mbentley, 2021-05-27:
There isn't a way to set the default memory reservation for the container through the Dockerfile.  That's up to the runtime to set; at least when using a Dockerfile and the Docker runtime.

From my own experience, when it comes to running Java applications in containers with memory limits, it is best to set the memory limit to about 1.3x the xmx argument to give the app overhead.

I have not used podman before but with Docker, you can override the [CMD](https://github.com/mbentley/docker-omada-controller/blob/master/Dockerfile.v4.3.x#L23) defined in the container by passing your own arguments after the Docker image where it would be something like this:

```podman run -d \
  --name omada-wlc \
  --restart on-failure:10 \
  --net host \
  -m 1g \
  -e TZ=America/New_York \
  -v /opt/omada-wlc/data:/opt/tplink/EAPController/data \
  -v /opt/omada-wlc/work:/opt/tplink/EAPController/work \
  -v /opt/omada-wlc/logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:latest-armv7l \
  /usr/bin/java -server -Xms128m -Xmx512mm -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -cp /opt/tplink/EAPController/lib/*: com.tplink.omada.start.OmadaLinuxMain
```

It seems like the stats on podman for armv7l are not being collected properly from the kernel according to the stats.  I know that memory metrics are different between amd64 and armv7l from my own usage.

alvarlagerlof, 2021-11-01:
Since this is java in a container, setting memory related argumens on java at all is unnescary since JDK 10.

mbentley, 2021-11-02:
The `armv7l` image uses the `openjdk-8-jre-headless` package as it is stuck on Ubuntu 16.04 due to the lack of a pre-packaged `mongodb` in Ubuntu for `armv7l` in newer versions.

closed 2021-11-02 by mbentley