mbentley/docker-omada-controller #110: Crazy CPU usage at over 300%

easaw, 2021-05-29:
**Describe the bug**
When running Omada Controller (I assume it is something to do with docker) on RPi 3 A+ RPi OS arm64, Java runs around 250%-350% and eventually freezes Pi.

**To Reproduce**
Steps to reproduce the behavior:

1. install _only_ docker & docker compose on RPi 3 A+ running Raspbian arm64 (also tried Ubuntu 20 arm64)
2. Setup WiFi only (as the A+ has no ethernet)
2. install the omada controller using: https://github.com/mbentley/docker-omada-controller#using-nethost-2
3. reboot
4. Java runs at 300%+ before freezing the Pi and CLI no longer works.

**Expected behavior**
Run normally at around 10% or less. But I can't even run the Omada Controller in a browser, or start an SSH. It seems the Pi os overloaded, even though it has the same specs as the B+??

**How you're launching your container**
Using net host for arm64 (4.3-arm64)
https://github.com/mbentley/docker-omada-controller#using-nethost-2
then ``sudo docker start omada-controller``

**Container Logs**
Can't run logs due to overload. Can't even SSH locally.

**Additional context**
I have disabled HDMI and also tried overclocking.
I have ordered cooling and a fan but I doubt this will improve things.

Image below is from `top` on a **RPi 3 B+** - not the A+ I am having touble with, but the CPU is still high. Is this possibly a docker issue?
[image]

mbentley, 2021-05-29:
I don't have a pi to check this on right now. Maybe try to put a `--cpu <num>` limit on the container and see if you see similar behavior. Maybe just start with `--cpu 1` for now.

easaw, 2021-05-29:
> I don't have a pi to check this on right now. Maybe try to put a `--cpu <num>` limit on the container and see if you see similar behavior. Maybe just start with `--cpu 1` for now.

Thank you for the quick reply, I'll try that.

Just to confirm you mean something like this?

```
docker run -d \
  --cpu 1 \
  --name omada-controller \
  --restart unless-stopped \
```

mbentley, 2021-05-29:
Exactly.

easaw, 2021-05-29:
> Exactly.
> 
> Actually, it might be `--cpus`

Will report back soon, thank you.

easaw, 2021-05-30:
The CPU doesn't go over about 100.5%, but the controller still doesn't load in browser. Oddly it does when the same SD card is used in a Pi 3 B+, rather than the Pi 3 A+.

I assume that's a different issue so for this one I'll say the `--cpus 1 \` was the right way to go, thank you! :)

easaw, 2021-05-30:
Tehnically this issue is answered, my CPU is not running at high numbers, but the overall problem of the Pi 3 A+ freezing (green LED constant) is still a problem (can't even login via ssh).

closed 2021-05-30 by easaw

easaw, 2021-05-30:
@mbentley, sorry to bother you but is there a way to use your omada-controller version without docker?

mbentley, 2021-05-30:
Sure. You'd have to check out the installer that comes with the app. There might be better instructions to do so on the tp-link forums. The link is in the readme (sorry on mobile).

easaw, 2021-05-30:
I just realised that the RPi 3 A+ only has 512Mb RAM, Vs the RPi 3 B+ 1GB... D'oh!

Can I change the `-Xmx1024m` in the 4.x Dockerfile, or will it not run well on that?

easaw, 2021-05-30:
Is there a way I can change the 

``CMD ["/usr/bin/java","-server","-Xms128m","-Xmx1024m","-XX:MaxHeapFreeRatio=60","-XX:MinHeapFreeRatio=30","-XX:+HeapDumpOnOutOfMemoryError","-cp","/opt/tplink/EAPController/lib/*:","com.tplink.omada.start.OmadaLinuxMain"]``
to `-Xmx512m`
In the Dockerfile, rather than like this?

``
docker run -d \
  --cpus 1 \
  --m 512mb \  
``

mbentley, 2021-06-01:
Yes, you can adjust those arguments by passing the CMD you wish you execute after the image name.  That will override the default CMD.  Just an example using the default run command:

```
docker run -d \
  --name omada-controller \
  --restart unless-stopped \
  -p 8088:8088 \
  -p 8043:8043 \
  -p 8843:8843 \
  -p 29810:29810 \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29811:29811/udp \
  -p 29812:29812 \
  -p 29812:29812/udp \
  -p 29813:29813 \
  -p 29813:29813/udp \
  -e MANAGE_HTTP_PORT=8088 \
  -e MANAGE_HTTPS_PORT=8043 \
  -e PORTAL_HTTP_PORT=8088 \
  -e PORTAL_HTTPS_PORT=8843 \
  -e SHOW_SERVER_LOGS=true \
  -e SHOW_MONGODB_LOGS=false \
  -e SSL_CERT_NAME="tls.crt" \
  -e SSL_KEY_NAME="tls.key" \
  -e TZ=Etc/UTC \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:4.3-armv7l \
  /usr/bin/java -server -Xms128m -Xmx512m -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -cp /opt/tplink/EAPController/lib/*: com.tplink.omada.start.OmadaLinuxMain```

I am not sure how well it will run on a pi with that little memory as it also would need to run mongodb and I do not believe that process is taken into account memory wise.