mbentley/docker-omada-controller #112: Latest docker image 3.2 seems broken

dsteinkopf, 2021-06-10:
**Describe the bug**
Pulling and running `mbentley/omada-controller:3.2` has been working fine for months (or years). Since today I keep getting an error: `ERROR: for omada_omada_1  Cannot start service omada: OCI runtime create failed: container_linux.go:380: starting container process caused: exec: "/entrypoint.sh": stat /entrypoint.sh: no such file or directory: unknown`

Cloneing this repo and building the image myself works:
```
git clone https://github.com/mbentley/docker-omada-controller
docker build -f Dockerfile.v3.2.x -t mbentley/omada-controller:3.2dirk .
```

**To Reproduce**
Steps to reproduce the behavior:

I am using a compose file but the problem should exist in any way you try to start the container.

**Expected behavior**
COontainer starts normally.

**How you're launching your container**
```
version: '2.1'
services:
  omada:
    build: .
    environment:
      - TZ=Europe/Berlin
      - SMALL_FILES=true
    ports:
      - ${BIND_IP}:8088:8088
      - ${BIND_IP}:8043:8043
      - ${BIND_IP}:27001:27001/udp
      - ${BIND_IP}:27002:27002
      - ${BIND_IP}:29810:29810/udp
      - ${BIND_IP}:29811:29811
      - ${BIND_IP}:29812:29812
      - ${BIND_IP}:29813:29813
    restart: unless-stopped
    volumes:
      - ${OMADA_DATA_DIR:?err}/data:/opt/tplink/EAPController/data
      - ${OMADA_DATA_DIR:?err}/work:/opt/tplink/EAPController/work
      - ${OMADA_DATA_DIR:?err}/logs:/opt/tplink/EAPController/logs
      - ${OMADA_DATA_DIR:?err}/backup:/opt/tplink/EAPController/backup
      - /etc/localtime:/etc/localtime:ro
      - /etc/timezone:/etc/timezone:ro
```

Dockerfile:
```
FROM mbentley/omada-controller:3.2dirk
# upgrade to 4.1 will make EAP110 V1.0/2.0 incompatible
# add mongo client for backup
RUN \
    apt-get update && \
    apt-get install -y \
        mongodb-clients
```

**Container Logs**
Starting the container fails with this error:
```
ERROR: for omada_omada_1  Cannot start service omada: OCI runtime create failed: container_linux.go:380: starting container process caused: exec: "/entrypoint.sh": stat /entrypoint.sh: no such file or directory: unknown
```

**Additional context**
-

mbentley, 2021-06-10:
Whoops, I must have broken something.  I recently had to move from the Docker Hub autobuilds to another solution since they're discontinuing autobuilds for the free tier.  Will check it out.

dsteinkopf, 2021-06-10:
Interesting. I am dealing with the same problem for my own projects. May I ask for your solution, maybe it would help me, too?

mbentley, 2021-06-10:
Could you check and see if it is working now for you?

Unfortunately my solution isn't great and involves a lot of manual work.  I'm building my images via Jenkins at my house and pushing them.  I had already setup the build infrastructure for the non-amd64 images so it was just updating my Jenkins jobs to include the other builds.

To optimize the builds, I am doing a lot with `buildx` and caching the data to a local Docker registry.  Here is an example using the 3.2 build:
```
docker buildx build --build-arg http_proxy=http://acng.casa.mbentley.net:3142 --builder builder1 --pull --push --progress plain --platform linux/amd64 -t mbentley/omada-controller:3.2 -f Dockerfile.v3.2.x --cache-from=type=registry,ref=registry.casa.mbentley.net/mbentley/omada-controller:3.2-cache --cache-to=type=registry,ref=registry.casa.mbentley.net/mbentley/omada-controller:3.2-cache,mode=max .
```

I am running apt-cacher-ng to cache .deb packages with the image caching.

dsteinkopf, 2021-06-10:
It's working again now. Thank you very much for your quick solution.

Ok. My situation (and requirements) is a bit different. Good luck for you and thank you again!

closed 2021-06-10 by dsteinkopf