mbentley/docker-omada-controller #114: OCI Runtime create failed

lignicolos, 2021-06-19:
Running the controller in docker on ubuntu using docker-compose. amd64

did a "docker-compose pull" which updated my omada docker image and then "docker-compose up -d" this morning. all of my other containers started right up except omada. I got the message below.

"ERROR: for omada-controller  Cannot start service omada-controller: OCI runtime create failed: container_linux.go:380: starting container process caused: exec: "/entrypoint.sh": stat /entrypoint.sh: no such file or directory: unknown
ERROR: Encountered errors while bringing up the project."

also pruned existing stuff and started fresh, but got the same message above.

below is my docker-compose.

  omada-controller:
    image: mbentley/omada-controller:latest
    container_name: omada-controller
    network_mode: host
    volumes:
      - /home/josh/.config/omada-controller/data:/opt/tplink/EAPController/data
      - /home/josh/.config/omada-controller/work:/opt/tplink/EAPController/work
      - /home/josh/.config/omada-controller/logs:/opt/tplink/EAPController/logs
    environment:
      - TZ=America/New_York
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
    restart: unless-stopped

mbentley, 2021-06-19:
I'm either having a problem with buildx or something with a Jenkins job.  Let me take a look...

mbentley, 2021-06-19:
OK, it should be fixed now.  It seems like there is a problem with buildx and writing manifests as this is the 2nd image I've seen this happen to but I am not exactly sure why it happens as just re-writing the manifest fixes it.

lignicolos, 2021-06-19:
and I'm in. Thank you!

closed 2021-06-19 by lignicolos