mbentley/docker-omada-controller #116: latest docker image broken due to invalid entrypoint

idyedov, 2021-06-24:
After updating to latest docker image my container is not able to start. tried removing and re-creating the container and got the same result.

docker command:

```$ docker run --name=omada-controller --env=TZ=America/Los_Angeles --env=MANAGE_HTTP_PORT=8088 --env=PORTAL_HTTPS_PORT=8843 --env=SHOW_SERVER_LOGS=true --env=SSL_CERT_NAME=tls.crt --env=SSL_KEY_NAME=tls.key --env=MANAGE_HTTPS_PORT=8043 --env=PORTAL_HTTP_PORT=8088 --env=SHOW_MONGODB_LOGS=true --env=PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin --volume=omada-work:/opt/tplink/EAPController/work --volume=omada-logs:/opt/tplink/EAPController/logs --volume=omada-data:/opt/tplink/EAPController/data --volume=/opt/tplink/EAPController/work --volume=/opt/tplink/EAPController/logs --volume=/opt/tplink/EAPController/data --network=host --workdir=/opt/tplink/EAPController/lib --expose=29811/tcp --expose=27002/tcp --expose=29810/udp --expose=8088/tcp --expose=8843/tcp --expose=29813/tcp --expose=29812/tcp --expose=27001/udp --expose=8043/tcp --restart=unless-stopped --detach=true mbentley/omada-controller:4.3-armv7l```

yields an error:

```docker: Error response from daemon: OCI runtime create failed: container_linux.go:367: starting container process caused: exec: "/entrypoint.sh": stat /entrypoint.sh: no such file or directory: unknown.```

I see there were some recent changes to entrypoint based on recent commits, perhaps it's related

silentkiller88, 2021-06-24:
I have the same issue:
ERROR: for omada-controller  Cannot start service omada-controller: OCI runtime create failed: container_linux.                                           go:367: starting container process caused: exec: "/entrypoint.sh": stat /entrypoint.sh: no such file or directo                                           ry: unknown

Image used: mbentley/omada-controller:4.3-armv7l

mbentley, 2021-06-24:
Sorry, I'm trying to get someone at Docker to check this issue out (https://github.com/docker/buildx/issues/637) as there is something wrong with buildkit.  It should be fixed now.

silentkiller88, 2021-06-24:
Thanks for the zwift response! Working again indeed

closed 2021-06-24 by mbentley

kipusoep, 2021-06-24:
Still had this issue with `latest-armv7l`, switched to `4.3-armv7l` to fix it.

idyedov, 2021-06-25:
thanks for the quick fix