mbentley/docker-omada-controller #117: Router's WAN down - log [INFO] javax.net.ssl.SSLException: Socket closed

Shurov, 2021-06-25:
**Describe the bug**
I think there is no clear dependencies to be able to reproduce. Several times a day my WAN connection is dropped. I do understand that it might be an ISP issue, however, there were no such issues on another router (non Omada - both consumer-grade Netgear and pfSense). I'm also not sure whether it is a container issue, or the controller app itself...

**How you're launching your container**
docker-compose:
```java
version: '3.3'
services:
  omada-controller:
    image: 'mbentley/omada-controller:latest'
    container_name: omada-controller
    restart: unless-stopped
    network_mode: host
    volumes:
      - ./data:/opt/tplink/EAPController/data
      - ./work:/opt/tplink/EAPController/work
      - ./logs:/opt/tplink/EAPController/logs
    environment:
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
```
Actually, I don't have any cert attached\used, so not sure last 2 lines are required. However, controller is accessed via https.

**Container Logs**

```
2021-06-21 00:27:58 [Thread-28] [INFO]-[SourceFile:146] - Close connection to service server.
2021-06-21 00:27:58 [Thread-39] [INFO]-[SourceFile:921] - Thread 'heartBeatThread' is stopped
2021-06-21 00:27:58 [Thread-38] [INFO]-[SourceFile:792] - expiredRequestCleanThread is interrupted.
2021-06-21 00:27:58 [Thread-38] [INFO]-[SourceFile:796] - Thread 'expiredRequestCleanThread' is stopped
2021-06-21 00:27:58 [Thread-37] [INFO]-[SourceFile:837] - javax.net.ssl.SSLException: Socket closed
javax.net.ssl.SSLException: Socket closed
	at sun.security.ssl.Alert.createSSLException(Alert.java:127) ~[?:1.8.0_292]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:324) ~[?:1.8.0_292]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:267) ~[?:1.8.0_292]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:262) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl.handleException(SSLSocketImpl.java:1554) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl.access$400(SSLSocketImpl.java:73) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:964) ~[?:1.8.0_292]
	at com.tplink.eap.cloudsdk.client.p.a(SourceFile:120) ~[cloudsdk-1.0.5.jar:?]
	at com.tplink.eap.cloudsdk.client.s.run(SourceFile:830) [cloudsdk-1.0.5.jar:?]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_292]
Caused by: java.net.SocketException: Socket closed
	at java.net.SocketInputStream.socketRead0(Native Method) ~[?:1.8.0_292]
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116) ~[?:1.8.0_292]
	at java.net.SocketInputStream.read(SocketInputStream.java:171) ~[?:1.8.0_292]
	at java.net.SocketInputStream.read(SocketInputStream.java:141) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketInputRecord.read(SSLSocketInputRecord.java:457) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketInputRecord.bytesInCompletePacket(SSLSocketInputRecord.java:68) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl.readApplicationRecord(SSLSocketImpl.java:1332) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl.access$300(SSLSocketImpl.java:73) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:948) ~[?:1.8.0_292]
	... 3 more
2021-06-21 00:27:58 [Thread-37] [INFO]-[SourceFile:839] - Thread 'recvThread' is stopped
2021-06-21 00:27:58 [Thread-28] [INFO]-[SourceFile:127] - Connect service server automatically, ConnectionType is PERSISTENT_CONNECTION.
2021-06-21 00:27:58 [Thread-28] [INFO]-[SourceFile:366] - The result of connection is false.
2021-06-21 00:27:58 [Thread-28] [INFO]-[SourceFile:146] - Close connection to service server.
2021-06-21 00:27:58 [Thread-28] [INFO]-[SourceFile:110] - Connect SEF server automatically.
2021-06-21 00:29:03 [Thread-28] [INFO]-[SourceFile:194] - The result of connection is true, service host is n-devs-smb.tplinkcloud.com, port is 443.
2021-06-21 00:29:03 [Thread-28] [INFO]-[SourceFile:114] - Close connection to SEF server.
2021-06-21 00:29:03 [Thread-1702] [INFO]-[SourceFile:403] - javax.net.ssl.SSLException: Socket closed
javax.net.ssl.SSLException: Socket closed
	at sun.security.ssl.Alert.createSSLException(Alert.java:127) ~[?:1.8.0_292]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:324) ~[?:1.8.0_292]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:267) ~[?:1.8.0_292]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:262) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl.handleException(SSLSocketImpl.java:1554) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl.access$400(SSLSocketImpl.java:73) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:964) ~[?:1.8.0_292]
	at com.tplink.eap.cloudsdk.client.p.a(SourceFile:120) ~[cloudsdk-1.0.5.jar:?]
	at com.tplink.eap.cloudsdk.client.o.run(SourceFile:397) [cloudsdk-1.0.5.jar:?]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_292]
Caused by: java.net.SocketException: Socket closed
	at java.net.SocketInputStream.socketRead0(Native Method) ~[?:1.8.0_292]
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116) ~[?:1.8.0_292]
	at java.net.SocketInputStream.read(SocketInputStream.java:171) ~[?:1.8.0_292]
	at java.net.SocketInputStream.read(SocketInputStream.java:141) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketInputRecord.read(SSLSocketInputRecord.java:457) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketInputRecord.bytesInCompletePacket(SSLSocketInputRecord.java:68) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl.readApplicationRecord(SSLSocketImpl.java:1332) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl.access$300(SSLSocketImpl.java:73) ~[?:1.8.0_292]
	at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:948) ~[?:1.8.0_292]
	... 3 more
2021-06-21 00:29:03 [Thread-1702] [INFO]-[SourceFile:407] - ssl recv thread is stopped.
2021-06-21 00:29:03 [Thread-28] [INFO]-[SourceFile:127] - Connect service server automatically, ConnectionType is PERSISTENT_CONNECTION.
2021-06-21 00:29:23 [Thread-28] [INFO]-[SourceFile:320] - Exception occurs when connecting service server:java.net.UnknownHostException: n-devs-smb.tplinkcloud.com: Temporary failure in name resolution
2021-06-21 00:29:23 [pool-32-thread-1] [WARN]-[SourceFile:577] - check device bind status error: -7119, Operation failed. Network connection error.
2021-06-21 00:30:48 [Thread-28] [INFO]-[SourceFile:366] - The result of connection is true.
2021-06-21 04:47:48 [scheduled-pool-8] [INFO]-[SourceFile:61] - Roaming status unstable, roaming interval changed to 20s
2021-06-21 04:57:48 [scheduled-pool-4] [INFO]-[SourceFile:54] - Roaming status stable, roaming interval changed to 8h
2021-06-21 08:36:11 [topology-task-0] [INFO]-[SourceFile:221] - topology of switch 00-5F-67-00-00-00 changed
2021-06-21 08:39:41 [topology-task-0] [INFO]-[SourceFile:221] - topology of switch 00-5F-67-00-00-00 changed
```
Apparently, my switch hasn't been moved\replugged\etc (referring to log line regarding topology of switch changed.

Could you advise, is it a docker's problem?

mbentley, 2021-06-25:
It is hard to say.  You might try to ping a known IP address, maybe your router, maybe even an internet address, from inside the container that the omada controller is running and then from another host on your network and see if you can narrow down where the problem is occurring or at least determine if it is a container only issue.

Shurov, 2021-06-26:
Hey @mbentley, thanks for a prompt reply.

The thing I forgot to mention initially (sorry for that) - is that controller loses all connected devices during this WAN down:
[image]
Actually, this was the issue I noticed initially and that bothered me, since all clients are losing WiFi connection from EAPs at this moment. Only afterwards I noticed in logs that the WAN is down at the same time. However, if it would be the ISP thing - my LAN should be still working, right?
So I have a gut feeling that it is controller (docker/app side) that loosing all the devices, including the router, so that WAN is being dropped by it for a while.

I will try to follow your advice. Will be tricky, since will need to catch that moment and act quickly until it is auto-restored after some time.
I'll definitely won't be able to ping wirelessly (most of my devices with cmd/terminal are connected via WiFi. Will try to hardwire quickly during such outage and try to ping, however, not sure that the switch is continuing to be working properly during this outage

mbentley, 2021-06-26:
I personally only have access points so I am not totally sure about the behavior but I do know that when my container gets restarted, the controller basically re-provisions my access points but I am not sure about a wan down situation.

Nothing scheduled that would do anything on the host that the container is running that could trigger and event?  And is it regularly at the same time or is it random?

mbentley, 2021-06-26:
Also, I would suggest posting a thread over at https://community.tp-link.com/en/business/forum/582 as a dropping WAN connection or connection to the controller really shouldn't cause this sort of interruption.  They have access to source code to see what is going on.

Shurov, 2021-06-27:
As advised, [posted to their forums](https://community.tp-link.com/en/business/forum/topic/268800). I think this issue may be moved to Stale. I'll post an update here when something new will be known.
Thanks

Shurov, 2021-07-02:
They've found a root cause. The controller seems to be restarted because container is updated automatically by the Watchtower (facepalming how could I miss this initially). However, everything is being re-adopted is a bug in the controller indeed:
> When the Controller get started, it will provision the DST settings to the Devices ahead of the time when the Devices are successfully adopted, leading to the version of configuration inconsistency, which results in that the Controller will provision the full configuration to the Devices once they are successfully adopted.
> As you may know, the Devices will be reconfigured during the provisioning (you will see the WAN is down and up later), then everything returns back to normal after the configuration is completed.
They'll fix it in some next update. From now on I'll try to match the root cause of the controller restarts - whether it is solely auto-updates, or something else.

closed 2021-07-02 by Shurov

mbentley, 2021-07-02:
Good deal.  I just noticed that 4.4.3 is now out and I just added support for it.  Hopefully that fixes it in any event.

Shurov, 2021-07-02:
that's when I realized the root cause of the restart 30 mins ago - new container version :D