mbentley/docker-omada-controller #121: Controller is restarted daily by itself

Shurov, 2021-07-06:
**Describe the bug**

I have mentioned restarts in my [prev issue](https://github.com/mbentley/docker-omada-controller/issues/117). I found that it actually happens not only when the container is auto-updated by Watchtower. Everyday at the same time (9:15 AM) it's restarted by itself.

**To Reproduce**
Steps to reproduce the behavior:
1. Wait till 09:15 AM
2. Restart happens

**Expected behavior**

Container is not restarted by itself until I do it myself

**How you're launching your container**
Include your `docker run` or compose file to make analysis easier.
docker-compose:
```
version: '3.3'
services:
  omada-controller:
    image: 'mbentley/omada-controller:latest'
    container_name: omada-controller
    restart: unless-stopped
    network_mode: host
    volumes:
      - ./data:/opt/tplink/EAPController/data
      - ./work:/opt/tplink/EAPController/work
      - ./logs:/opt/tplink/EAPController/logs
    environment:
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
```

**Container Logs**

```
2021-07-06 09:15:08 [comm-pool-0] [WARN]-[SourceFile:142] - get InterruptedException while saving log.
2021-07-06 09:15:08 [Thread-1] [INFO]-[SourceFile:219] - Going to stop mongod which pid is 46
2021-07-06 09:15:09 [Thread-2] [WARN]-[SLF4JLogger.java:86] - Got socket exception on connection [connectionId{localValue:8, serverValue:8}] to 127.0.0.1:27217. All connections to 127.0.0.1:27217 will be closed.
2021-07-06 09:15:09 [nioEventLoopGroup-8-10] [INFO]-[SourceFile:50] - Omada Controller isn't prepared to handle event
2021-07-06 09:15:09 [nioEventLoopGroup-6-1] [INFO]-[SourceFile:50] - Omada Controller isn't prepared to handle event
2021-07-06 09:15:13 [nioEventLoopGroup-8-8] [INFO]-[SourceFile:50] - Omada Controller isn't prepared to handle event
2021-07-06 09:15:14 [Thread-1] [WARN]-[SourceFile:183] - Failed to shutdown mongodb database
2021-07-06 09:15:14 [Thread-1] [INFO]-[SourceFile:186] - Omada Controller exited
2021-07-06 09:15:16 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2021-07-06 09:15:16 [main] [INFO]-[SourceFile:91] - going to start local mongod.
2021-07-06 09:15:17 [main] [INFO]-[SourceFile:167] - mongodb process id is 46
2021-07-06 09:15:17 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2021-07-06 09:15:22 [main] [INFO]-[SourceFile:193] - User do not configure max.device, use defaultValue 1500.
2021-07-06 09:15:24 [main] [INFO]-[SourceFile:38] - monitor context initialing...
2021-07-06 09:15:25 [main] [INFO]-[SourceFile:193] - User do not configure max.device, use defaultValue 1500.
2021-07-06 09:15:26 [main] [INFO]-[SourceFile:463] - restart all customize tasks.
2021-07-06 09:15:26 [main] [INFO]-[SourceFile:107] - Success sending config to devices.
2021-07-06 09:15:26 [main] [INFO]-[SourceFile:73] - DisconnectRequestServer start
2021-07-06 09:15:26 [main] [INFO]-[SourceFile:152] - no need to compatible db.
2021-07-06 09:15:26 [main] [INFO]-[SourceFile:20] - Omada Controller started
```

**Additional context**

As far as I understand rolling logs are being saved and compressed daily at this time. So for some reason it fails for the MongoDb logs:
[image]
archived server_logs are being stashed daily. However, there is no archived history for the mongod.log. I'm not sure what differs between them, since the ownership is similar.

mbentley, 2021-07-06:
It would seem like something is instructing the container to stop then.  I would check the Docker daemon logs to see if there is anything to indicate what is going on - maybe Docker itself is being restarted.  I had not seen this behavior before and I know I do not have the same issue.  My best guess is that there is some external process doing something to cause the container to be gracefully stopped at that time and because of the `unless-stopped` restart policy, it doesn't come back up on it's own since it seems to be gracefully shutting down.

Shurov, 2021-07-06:
Could it be due to `mongod.log` size and free space available?
[image]
It looks like it should suffice, but not sure.

mbentley, 2021-07-06:
Is there anything in the `mongod.log` at the end that might indicate what is going on with MongoDB at all when it is shutting down?  Very odd that it happens at the same time.  A clean stop for me looks like this:

```
...
2021-07-06 13:27:40 [comm-pool-0] [WARN]-[SourceFile:142] - get InterruptedException while saving log.
ShutdownHook: service stopped
```

So there may be something odd going on there.

Shurov, 2021-07-06:
the only line from docker daemon log at the restart time is
```
Jul 06 09:15:15 shurov dockerd[1066]: time="2021-07-06T09:15:15.355321173+02:00" level=info msg="ignoring event" container=736f2db83206e0e5692e4ac3ff42d69f451f68f19227a1befddc736b6efd574b module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
```

mongod.log before and at restart time:
```
2021-07-06T07:20:28.145+0200 I NETWORK  [listener] connection accepted from 127.0.0.1:33184 #9 (5 connections now open)
2021-07-06T**07:20:28**.150+0200 I NETWORK  [conn9] received client metadata from 127.0.0.1:33184 conn: { driver: { name: "mongo-java-driver", version: "3.6.4" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.4.114-1-pve" }, platform: "Java/Private Build/1.8.0_292-8u292-b10-0ubuntu1~18.04-b10" }
2021-07-06T**09:15:08**.919+0200 I CONTROL  [signalProcessingThread] got signal 15 (Terminated), will terminate after current cmd ends
2021-07-06T09:15:08.920+0200 I NETWORK  [signalProcessingThread] shutdown: going to close listening sockets...
2021-07-06T09:15:08.920+0200 I NETWORK  [signalProcessingThread] removing socket file: /tmp/mongodb-27217.sock
2021-07-06T09:15:08.926+0200 I FTDC     [signalProcessingThread] Shutting down full-time diagnostic data capture
2021-07-06T09:15:08.930+0200 I STORAGE  [signalProcessingThread] WiredTigerKVEngine shutting down
2021-07-06T09:15:09.158+0200 I STORAGE  [signalProcessingThread] shutdown: removing fs lock...
2021-07-06T09:15:09.159+0200 I CONTROL  [signalProcessingThread] now exiting
2021-07-06T09:15:09.159+0200 I CONTROL  [signalProcessingThread] shutting down with code:0
```

mbentley, 2021-07-06:
OK, that looks like a normal shutdown as far as I can tell.  I don't think there would be anything specifically related to the amount of disk space available even though that is getting low.  Not sure how much reserved space there might be for the root fs but my money would still be on an external process stopping the container due to how it is always happening at the same time.  Any other tasks/crons that could be running and performing a command at that time?

Shurov, 2021-07-06:
nothing I can think of...
I have Portainer as a web management, but it's a default setup showing docker-compose stacks. Anyway, will try stopping it and see tomorrow morning how the Omada container will behave. 
Afterwards, if this won't change anything, I'll try increasing the disk volume for the root (or just removing the huge mongod log).
Will get in couple of days with results. Thanks for your prompt replies and a help!

mbentley, 2021-07-06:
Yeah, it wouldn't cause any harm to prune the logs for mongodb.  I would probably also check my system logs like `/var/log/syslog` at 9:15 to see if there is something running at 9:15, unless you've already done that.

Shurov, 2021-07-28:
After cleaning old logs and regaining some space I didn't have those restarts. 
I think most (if not all) of my restarts are due to the automated image upgrades (by [Watchtower](https://containrrr.dev/watchtower/)) onto the new builds. 

However, it's a mystery for me, why would dockerhub have a new build (e.g. today), when the last commit to the repo was 13 days ago, so no real changes to the image?

mbentley, 2021-07-28:
I trigger re-builds daily but I use the buildkit integration in Docker to cache images.  On some occasions, buildx w/buildkit will have a cache miss for no reason and rebuild.  The other reasons it would rebuild would be is if there is a base image update.  In the case of yesterday, it was a random cache being busted issue.  Unfortunately, I have to run an older version of buildkit due to an issue that causes images to have missing layers which means that the cache detection isn't as good as newer versions.

I personally have a job that runs daily to do a `docker image prune -f` to just remove the older versions of images daily to ensure I don't keep around old images that no longer have tags.

Shurov, 2021-07-28:
Thanks for explaining!

closed 2021-07-28 by Shurov