mbentley/docker-omada-controller #122: Feature request: build multi-arch images

e3b0c442, 2021-07-13:
**Is your feature request related to a problem? Please describe.**
Running omada-controller in a heterogenous Kubernetes cluster currently requires the use of `nodeSelector` because the docker image tags are architecture-specific.

**Describe the solution you'd like**
Utilize the modern docker buildx (or legacy manifest builds) to publish multi-architecture tags.

**Additional context**
https://docs.docker.com/desktop/multi-arch/
https://www.docker.com/blog/multi-arch-build-and-images-the-simple-way/

mbentley, 2021-07-13:
Thanks for filing an issue to track this!  It's something I had been thinking about doing for some time but in the past, I was a bit limited with what I could do, mostly because I was using Docker Hub Autobuilds to build the amd64 images and then building the arm based armv7l and arm64 images in my Jenkins build environment and pushing them up.  Now that I have moved everything from Docker Hub to Jenkins, I actually have control of all of the image builds so with some minor reworking, I should be able to go the `docker manifest` route.

mbentley, 2021-07-13:
I just added mult-arch images and updated the README.  Let me know if anything isn't clear or could use some rewording.

commit 23129862 (2021-07-13): Merge pull request #123 from mbentley/add-multi-arch-readme

Updated README for multi-arch; fixes #122
commit ad81b635 (2021-07-13): Updated README for multi-arch; fixes #122
closed 2021-07-13 by mbentley