mbentley/docker-omada-controller #124: Only push new containers to docker hub when new version is released, instead of daily

codegeek, 2021-08-13:
**Is your feature request related to a problem? Please describe.**

There is a problem in the Omada Controller Software (up to 4.4.3, supposedly to be fixed in 4.4.4) where a restart of the controller will cause a re-adoption of all devices, thus causing network disruption. I have the docker container running in  a system which checks every day for updated versions of containers (Unraid). Since there is a daily build of the `docker-omada-controller` that, even though I don't see any changes in content, seem to have a different hash, thus causing a container update, thus causing a daily reset of the network.

**Describe the solution you'd like**

I would like for at least the version specific tags (4.4.x, for example) o not have a daily build that makes my system think a new version has been released when it hasn't. 

**Additional context**

One thing I noticed that might or might not be related to the problem, in the `install.sh` script all `cp` commands are using `-r`. I wonder if changing this to `-a` would help in creating a container that has the same checksum, thus not triggering an update.

mbentley, 2021-08-13:
I am not sure why it would trigger a restart of the controller.  The only thing that is happening is that the image manifests are re-written when there are no changes - none of the content actually changes in that case.  You can see here in the builds that they are not actually rebuilding:

[image]

What's the system that is doing the checking?  Watchtower?  It seems like it might be detecting that the manifest has been pushed so it re-deploys when really it should be checking the manifest for the actually image that needs to (or doesn't need to) be updated.  I can't not do daily builds as that is how I ensure the software is always up to date and there are not vulnerabilities.  It's a part of my update process for all of my images and it takes like no time at all if there is no change which is a majority of my images.

codegeek, 2021-08-13:
The docker images are being pushed to the hub [daily](https://hub.docker.com/r/mbentley/omada-controller/tags?page=1&ordering=last_updated). Notice that the "last pushed" time is just 11 hours (at the time of writing). Because the script that checks daily for container updates sees that a new container was pushed, it will treat it as an update.

mbentley, 2021-08-13:
Right.  Every single day a manifest is pushed but that doesn't mean that the image has changed.  The script/tool that checks for a container update is doing so incorrectly and they should fix that.

First pull:
```
$ docker pull mbentley/curl
Using default tag: latest
latest: Pulling from mbentley/curl
29291e31a76a: Already exists
ea860d19b0a2: Pull complete
Digest: sha256:5cf381f30af6833d74d975f5b39e635c2b8f691fcc26788e796448533168798e
Status: Downloaded newer image for mbentley/curl:latest
docker.io/mbentley/curl:latest
```

Second pull (nothing has been changed on Hub):

```
$ docker pull mbentley/curl
Using default tag: latest
latest: Pulling from mbentley/curl
Digest: sha256:5cf381f30af6833d74d975f5b39e635c2b8f691fcc26788e796448533168798e
Status: Image is up to date for mbentley/curl:latest
docker.io/mbentley/curl:latest
```

Third pull (after the build job has been re-triggered; in this case with no changes, all images showed as cached and the manifest re-written):

```
$ docker pull mbentley/curl
Using default tag: latest
latest: Pulling from mbentley/curl
Digest: sha256:5cf381f30af6833d74d975f5b39e635c2b8f691fcc26788e796448533168798e
Status: Image is up to date for mbentley/curl:latest
docker.io/mbentley/curl:latest
```

Looking at the manifests between daily builds:

```
$ docker manifest inspect mbentley/curl
{
   "schemaVersion": 2,
   "mediaType": "application/vnd.docker.distribution.manifest.list.v2+json",
   "manifests": [
      {
         "mediaType": "application/vnd.docker.distribution.manifest.v2+json",
         "size": 739,
         "digest": "sha256:e27eacabbb4092e0d207b981db5688203bb957de62ba4dbf44818a2e91c52a23",
         "platform": {
            "architecture": "amd64",
            "os": "linux"
         }
      },
      {
         "mediaType": "application/vnd.docker.distribution.manifest.v2+json",
         "size": 739,
         "digest": "sha256:c203037f57cfdcfd0c2c13861524f95287541c71b98d3ae616542c98d98358bf",
         "platform": {
            "architecture": "arm64",
            "os": "linux"
         }
      }
   ]
}
```

Another pull of the manifest after a re-build:

```
$ docker manifest inspect mbentley/curl
{
   "schemaVersion": 2,
   "mediaType": "application/vnd.docker.distribution.manifest.list.v2+json",
   "manifests": [
      {
         "mediaType": "application/vnd.docker.distribution.manifest.v2+json",
         "size": 739,
         "digest": "sha256:e27eacabbb4092e0d207b981db5688203bb957de62ba4dbf44818a2e91c52a23",
         "platform": {
            "architecture": "amd64",
            "os": "linux"
         }
      },
      {
         "mediaType": "application/vnd.docker.distribution.manifest.v2+json",
         "size": 739,
         "digest": "sha256:c203037f57cfdcfd0c2c13861524f95287541c71b98d3ae616542c98d98358bf",
         "platform": {
            "architecture": "arm64",
            "os": "linux"
         }
      }
   ]
}
```

As you can see, the manifests have not changed so the tools should be examining the manifests to make sure they are not different, not that something has been updated on Docker Hub.  There is nothing I can do about another tool that is looking for the wrong thing.

codegeek, 2021-08-15:
Well, that is weird, I have checked a couple of days, and the digest has stayed the same, even though Docker Hub does show that it was pushed. And the weirdest part is that now Unraid is also not detecting any change. It must've been some misconfiguration on my part.

closed 2021-08-15 by codegeek

Shurov, 2021-08-17:
> I am not sure why it would trigger a restart of the controller

This is the [same problem](https://github.com/mbentley/docker-omada-controller/issues/117#issuecomment-872948019) I had. Indeed, fixed in Controller 4.4.4 (+ I think router R605 FW v1.1.1)