mbentley/docker-omada-controller #129: latest docker image broken due to invalid entrypoint #116

thierrybla, 2021-08-16:
After updating to latest docker image my container is not able to start. tried removing and re-creating the container and got the same result.

docker command:

$ docker run -d \
--name omada-controller \
--restart unless-stopped \
--net host \
--no-healthcheck \
-e MANAGE_HTTP_PORT=8088 \
-e MANAGE_HTTPS_PORT=8043 \
-e PORTAL_HTTP_PORT=8088 \
-e PORTAL_HTTPS_PORT=8843 \
-e SHOW_SERVER_LOGS=true \
-e SHOW_MONGODB_LOGS=false \
-e SSL_CERT_NAME="tls.crt" \
-e SSL_KEY_NAME="tls.key" \
-e TZ=Europe/Amsterdam \
-v /docker/omada-controller/data:/opt/tplink/EAPController/data \
-v /docker/omada-controller/work:/opt/tplink/EAPController/work \
-v /docker/omada-controller/logs:/opt/tplink/EAPController/logs \
mbentley/omada-controller:latest

yields an error:

docker: Error response from daemon: OCI runtime create failed: container_linux.go:367: starting container process caused: exec: "/entrypoint.sh": stat /entrypoint.sh: no such file or directory: unknown.

mbentley, 2021-08-16:
Same issue as https://github.com/mbentley/docker-omada-controller/issues/116 with buildkit.  Re-pull the image and it should be fine now.

thierrybla, 2021-08-16:
I did Re-pull and it didn't work, Thanks for the help in advance!
[image]

mbentley, 2021-08-16:
I just manually cleared the build cache and rebuilt the image so it should be good now.

thierrybla, 2021-08-16:
Is there a delay on this ? I just tried it again but run in the same error.

mbentley, 2021-08-16:
No, there shouldn't be a delay.  I just pulled the latest 4.4 image `mbentley/omada-controller:4.4@sha256:f97072e2dc553c8b70341af43c7e66caab4c362554071aa60f8cde9916f31b80` and it is working again.

mbentley, 2021-08-16:
Oh hmm...still not working with latest.  Checking.

mbentley, 2021-08-16:
OK, I just rebuilt the manifest again.  Not sure why I failed - I probably fat fingered something the first time.  Should really be good now - I just tested it.

thierrybla, 2021-08-16:
Yes it works now, Thanks for the quick help.

closed 2021-08-16 by thierrybla

Gamel53, 2021-09-07:
Hello,
First, thank you for sharing this container. Very useful for me.
I figured out that I have this same issue.
It is quite surprising, I am using watchtower that is automatically updating my containers. It seems that my omada server is unable to start since the last update, on August 16th (presumably the last build you did).
I have the same type of message when I start to start the docker image since then:
"OCI runtime create failed: container_linux.go:367: starting container process caused: exec: \"/entrypoint.sh\": stat /entrypoint.sh: no such file or directory: unknown".

mbentley, 2021-09-09:
Re-pull the image and it should be fine if you haven't yet.

Gamel53, 2021-09-12:
Hello,
I finally got time to go back on this issue and I confirm that it is working now. Thank you very much for your assistance.
Cheers.
Gaetan